jneilliii/OctoPrint-TPLinkSmartplug #177: [BUG]: Leading spaces in IP entry form prevent successful connections

0xz00n, 2020-05-29:
**Describe the bug**
When adding a smart plug, the IP entry form allows leading spaces, preventing the plugin from reaching the target.

**Solution**
Use something like `strip()` to remove leading/trailing spaces from that variable

**To Reproduce**
Steps to reproduce the behavior:
1. Add a new smart plug or modify an existing one
2. Enter a leading space before the IP address
3. Attempt to send on/off commands to the smart plug

**Desktop (please complete the following information):**
 - OS: Linux
 - Browser: Firefox
 - OctoPrint Version: 1.4.0
 - Plugin Version: 0.9.20

jneilliii, 2020-05-29:
How about a visual indicator in the input field if it being wrong?

0xz00n, 2020-05-29:
That works just as well!

jneilliii, 2020-05-31:
I've just pushed an update to my latest dev branch that adds the error highlighting of both the input field and the row in the list. Hopefully that will help other users avoid this issue.

PR #178 (2020-06-19): 0.9.21
* use threading.timer for print progress status checks and switched to queuing gcode hook and other gcode processing improvements, attempting to resolve #164 
* added error classes for ip address entry errors, resolves #177
closed 2020-06-19 by jneilliii