jneilliii/OctoPrint-TPLinkSmartplug #179: [Feature Request]: Detect actions and automatically cancel auto-poweroff

thelewiss1, 2020-06-01:
**Is your feature request related to a problem? Please describe.**
Sometimes, I'll use Octopod for troubleshooting, etc, and not realise the autopoweroff timer is counting, and while i can control the power of the printer via the app, it does not give any warning to cancel the power off.

**Describe the solution you'd like**
TPLinkSmartplug to detect actions, like heating, movement, command sent and cancel autopoweroff, resetting the timer.
The author of Octopod would have to implement the autopower off cancel and alert otherwise.

**Describe alternatives you've considered**
Only real option for mobile is to use the web interface, which is not preffered on a phone. 

**Additional context**
N/A

jneilliii, 2020-06-01:
I'm pretty sure the timer gets reset when gcode commands are sent. Just not sure if that is from a file only or not, so things like heating bed in theory should be doing that already. Sorry I don't know more about octopod since I don't have apple products. The timer code was merged in from another project and I'll need to reverse engineer the logic again to see what might be possible.

jneilliii, 2020-09-03:
I was catching up on the backlog, and thought of something else that might help you out. The idle timer has a temp threshold, so if you set the temperature of the bed for example it shouldn't time out.

thelewiss1, 2020-09-03:
I do already have that set up, and you’re correct. If I’m working on the printer, maybe I’ll just set the bed temp to slightly above the threshold I have set. 

Thanks.

jneilliii, 2020-09-03:
Actually, I just double-checked the code, and the idle timer will actually set the temps to 0, so that won't work either for you. I'll look into adding an additional gcode/@command to allow for disabling the idle timer. Do you have the option to send any command via OctoPod?

thelewiss1, 2020-09-03:
Yes, I can set that up as a Custom Commands Button.
 Im not sure if you could, but maybe it could be made so it pauses auto power off timers until an event, like next print start or end?

Thanks!

thelewiss1, 2020-09-03:
>Im not sure if you could, but maybe it could be made so it pauses auto power off timers until an event, like next print start or end?

On further reflection, If there was @EnableAutoPowerOff and @DisableAutoPowerOff, I could instead use "Before print" / "After Print" Gcode scripts.

jneilliii, 2020-09-03:
Yeah, it would be a little more flexible I think, but if you were to have that in the before print/after print scripts it would break the logic of the idle timeout altogether because on print complete it would disable the idle timer.

thelewiss1, 2020-09-03:
> Yeah, it would be a little more flexible I think, but if you were to have that in the before print/after print scripts it would break the logic of the idle timeout altogether because on print complete it would disable the idle timer.

I think I'd only need to add:

**Print _Start_**
`@EnableAutoPowerOff`

since that would enable the poweroff before any of that logic ready for when I'd actually want it to start the timeout anyway.

So I'd instead

1. Run `@DisableAutoPowerOff` via app
2. _**Do whatever**_
3. Start Print, _it would enable the auto power off_
4. It would finish, and would time out the plug with the temperatures or idle time etc.

That'd stop the logic from breaking I'd assume

jneilliii, 2020-09-03:
Perfect, yeah that would definitely work. Will probably have this new feature available by the end of the weekend, if not sooner.

thelewiss1, 2020-09-03:
My printer is currently in a cupboard, but If you need me to try it, let me know! I’m more than happy to.

jneilliii, 2020-09-04:
feature implemented by above commit and will be included in the next release. I added `@TPLINKIDLEON` and `@TPLINKIDLEOFF` at commands so allow for controlling the idle timer. No testing necessary, but if you want to try it out you can install by copying/pasting the URL below into Plugin Manager > Get More > ...from URL and clicking install. The version will still show 0.9.24 until official release so you get any future updates to this dev version.

```
https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/archive/0.9.25.zip
```

thelewiss1, 2020-09-04:
That's awesome! Thanks for your quick work! If you do not believe it needs to be tested, I'll close this. If I have any problems, I'll reopen.

commit 91d8d6b7 (2020-09-04): added TPLINKIDLEON and TPLINKIDLEOFF at commands for controlling idle timer, resolves #179
closed 2020-09-04 by thelewiss1