jneilliii/OctoPrint-TPLinkSmartplug #180: [BUG]:

Sonnenwerk, 2020-06-04:
**Describe the bug**
Can not switch on/off TPLink HS100 with OctoPi / TP Plugin.

**To Reproduce**
Steps to reproduce the behavior:
1. Setup TP Plug --> Works with Kasa app, Go to octo webinterface and configure with same IP
2. Click on smartplug icon 
4. Icon rotates about 5 times, no warning, no error message, but plug is still on. 

**Expected behavior**
Switch plug off

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [Win/Raspi)
 - Browser chrome (most current)
 - OctoPrint Version 1.4 (most current)
 - Plugin 0.9.20
 - HS 100 most current firmware 1.1.1

**Additional context**
Thats what i got out of octos log..not sure, if that helps or what loginfo is needed.

2020-06-04 09:52:42,365 - octoprint.plugins.tplinksmartplug - INFO - TPLinkSmartplug loaded!
2020-06-04 09:52:42,648 - octoprint.plugins.tracking - INFO - Sent tracking event ping, payload: {'octoprint_uptime': 12}
2020-06-04 09:52:42,672 - octoprint.server.preemptive_cache - INFO - Preemptively caching / (ui _default) for {'query_string': 'l10n=de', 'path': '/', 'base_url': 'http://192.168.2.109/'}
2020-06-04 09:52:42,847 - octoprint.plugins.tracking - INFO - Sent tracking event pong, payload: {'plugins': u'displaylayerprogress:1.21.0,active_filters:0.0.1,opitemp:1.0.8,iponconnect:0.2.3,cancelobject:0.4.1,resource_monitor:0.2.3,abl_expert:0.5,themeify:1.2.2,bedlevelvisualizer:0.1.13,tplinksmartplug:0.9.20,navbartemp:0.14,excluderegion:0.2.0,dashboard:1.13.0,eeprom_marlin:1.2.1,active_filters_extended:0.1.0,octolapse:0.3.4'}
2020-06-04 09:52:43,069 - octoprint.plugins.tracking - INFO - Sent tracking event startup, payload: {'python': '2.7.16', 'ram': 917016576, u'octopi_version': u'0.17.0', 'version': u'1.4.0', 'pip': '19.3.1', 'cores': 4, 'freq': 1400.0, u'pi_model': u'Raspberry Pi 3 Model B Plus Rev 1.3', 'os': 'linux'}
2020-06-04 09:52:46,330 - octoprint.plugin - ERROR - Error while calling plugin tplinksmartplug
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/plugin/__init__.py", line 224, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_tplinksmartplug/__init__.py", line 516, in on_event
    if response["currentState"] == "off":
TypeError: 'NoneType' object has no attribute '__getitem__'
2020-06-04 09:52:49,175 - tornado.access - WARNING - 409 GET /api/printer (::ffff:192.168.2.108) 28.27ms

[image]
[image]

any idea how to progress with finding a solution ? 
cu
Thomas

jneilliii, 2020-06-05:
It's not finding a response from the plug for some reason. Can you upload your `plugin_tplinksmartplug_debug.log` file form the logging section of OctoPrint please?

jneilliii, 2020-06-05:
I notice you mentioned most current firmware as well. A recent HS103 plug user reported an issue withe most current version of its firmware in #163 that may be related.

Sonnenwerk, 2020-06-06:
Hmm..yesterday the powersupply went defect, i ordered a new one. So curently i can not verify.

snecklifter, 2020-06-08:
This works fine for me however I'm in UK on firmware 1.0.3 with no update available?

Sonnenwerk, 2020-06-20:
Hi good afternoon !
I did not get my spare powersupply yet..still having issues to communicate with that company..well Corona does not help either..anyway, i managed to install another powersupply. So i now can access the log and upload it.
First input from my router, that the ip is the right one..
---
Gerätename PrintPlug
MAC-Adresse 74-DA-88-FD-23-22
IPv4-Adresse 192.168.2.115
---
[2020-06-20 17:50:11,396] DEBUG: {'emeter': {'err_code': True}, 'system': {'get_sysinfo': {'relay_state': 3}}}
[2020-06-20 17:50:11,519] DEBUG: ImmutableMultiDict([('checkStatus', u'192.168.2.115')])
[2020-06-20 17:50:11,520] DEBUG: Checking status of 192.168.2.115.
[2020-06-20 17:50:11,521] DEBUG: {'system': {'get_sysinfo': {}}}
[2020-06-20 17:50:11,522] DEBUG: IP 192.168.2.115 is valid.
[2020-06-20 17:50:11,522] DEBUG: Sending command {'system': {'get_sysinfo': {}}} to 192.168.2.115
[2020-06-20 17:50:14,506] DEBUG: Could not connect to 192.168.2.115.
[2020-06-20 17:50:14,507] DEBUG: 
[2020-06-20 17:50:14,508] DEBUG: {'emeter': {'err_code': True}, 'system': {'get_sysinfo': {'relay_state': 3}}}
[2020-06-20 17:50:14,590] DEBUG: ImmutableMultiDict([('checkStatus', u'192.168.2.115')])
[2020-06-20 17:50:14,592] DEBUG: Checking status of 192.168.2.115.
[2020-06-20 17:50:14,592] DEBUG: {'system': {'get_sysinfo': {}}}
[2020-06-20 17:50:14,593] DEBUG: IP 192.168.2.115 is valid.
[2020-06-20 17:50:14,594] DEBUG: Sending command {'system': {'get_sysinfo': {}}} to 192.168.2.115
[2020-06-20 17:50:17,626] DEBUG: Could not connect to 192.168.2.115.
[2020-06-20 17:50:17,627] DEBUG: 
[2020-06-20 17:50:17,628] DEBUG: {'emeter': {'err_code': True}, 'system': {'get_sysinfo': {'relay_state': 3}}}
[2020-06-20 17:50:25,311] DEBUG: Checking status of 192.168.2.115.
[2020-06-20 17:50:25,312] DEBUG: {'system': {'get_sysinfo': {}}}
[2020-06-20 17:50:25,313] DEBUG: IP 192.168.2.115 is valid.
[2020-06-20 17:50:25,314] DEBUG: Sending command {'system': {'get_sysinfo': {}}} to 192.168.2.115
[2020-06-20 17:50:28,426] DEBUG: Could not connect to 192.168.2.115.
[2020-06-20 17:50:28,427] DEBUG: 
[2020-06-20 17:50:28,428] DEBUG: {'emeter': {'err_code': True}, 'system': {'get_sysinfo': {'relay_state': 3}}}

If i use the Kasa App i can switch on / off in the same network.

jneilliii, 2020-06-20:
So you might be impacted by a recent firmware update. I have received reports that TPLink may have broken communication, which there is nothing I can do about until someone else figures out a way to reverse engineer the communication again. For example, my working HS100 is a HW version 1.0, Firmware version 1.2.5 and I'm still able to control my device. All my plugs are in Local Only mode though, so not sure if that has an impact on your use case or not. You could try turning off Remote Control in the Device Settings (Kasa app) and then unplug the unit from the wall and plug it back in to see if that makes a difference.

jneilliii, 2020-06-23:
There have been reports from other users in the forum and on discord that performing a factory reset on their plug was able to get it working again, but I'm not going to guarantee that it fixes it for you.

DamaxusAwe, 2020-06-24:
FYI, I just set up 2 HS100 devices, Hardware v 2.0, Firmware v 1.5.6, both work fine.

Sonnenwerk, 2020-06-25:
Hi,
i did a factory reset today. Problem stays. I also contacted support and referenced to this bugfiling. Maybe they will read it.
cu

jneilliii, 2020-06-25:
I doubt they will care what people are doing outside of their own app. Especially since it was their protocol that was reverse engineered and technically could be considered a security risk.

PhriZ, 2020-07-16:
I am seeing the exact same behaviour.
I use a HS110 HW Version 4.0 and FW Versuib 1.0.4

Damn. I just bought it for this purpose.

PhriZ, 2020-07-16:
I just got the plug so I am not sure If I am doing sth else wrong. 
This shows up in the log:

> [2020-07-16 23:10:32,481] DEBUG: ImmutableMultiDict([('checkStatus', u'192.168.1.102')])
> [2020-07-16 23:10:32,482] DEBUG: Checking status of 192.168.1.102.
> [2020-07-16 23:10:32,482] DEBUG: {'system': {'get_sysinfo': {}}}
> [2020-07-16 23:10:32,483] DEBUG: IP 192.168.1.102 is valid.
> [2020-07-16 23:10:32,483] DEBUG: Sending command {'system': {'get_sysinfo': {}}} to 192.168.1.102
> [2020-07-16 23:10:35,611] DEBUG: Could not connect to 192.168.1.102.
> [2020-07-16 23:10:35,612] DEBUG: 
> [2020-07-16 23:10:35,612] DEBUG: {'emeter': {'err_code': True}, 'system': {'get_sysinfo': {'relay_state': 3}}}

PhriZ, 2020-07-17:
Never mind me. Works fine now. 
My guess would be that my static IP mapping wasn't actually forwarded to the plug and thus the IP set in octoprint being wrong.

jneilliii, 2020-07-17:
@Sonnenwerk it seems that this may be resolvable for you following the steps linked in this comment, and now added to the warning in the readme.

https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/issues/163#issuecomment-660281143

fearlesschicken, 2020-08-01:
Mine stopped working today with pretty much the same errors in the logs.

[2020-08-01 13:16:50,618] DEBUG: Checking statuses during print progress (51).
[2020-08-01 13:16:50,627] DEBUG: Resetting idle timer during print progress (51)...
[2020-08-01 13:16:51,628] DEBUG: Checking status of 192.168.43.171.
[2020-08-01 13:16:51,629] DEBUG: {'system': {'get_sysinfo': {}}}
[2020-08-01 13:16:51,630] DEBUG: IP 192.168.43.171 is valid.
[2020-08-01 13:16:51,631] DEBUG: Sending command {'system': {'get_sysinfo': {}}} to 192.168.43.171
[2020-08-01 13:16:54,766] DEBUG: Could not connect to 192.168.43.171.
[2020-08-01 13:16:54,767] DEBUG:
[2020-08-01 13:16:54,767] DEBUG: {'emeter': {'err_code': True}, 'system': {'get_sysinfo': {'relay_state': 3}}}

The smartplug (HS100 on 1.5.6) still works on Kasa and on my Homebridge setup.  Not sure what happened overnight for this to suddenly stop working

jneilliii, 2020-08-01:
Interesting, an HS100 shouldn't have energy monitoring so shouldn't be attempting to check status during print progress. I'v seen cases where the plug had to be unplugged from the wall and reconnected. It could also be your firmware updated and if it was set to being able to be controlled from the cloud it locked out local mode connections, which is how this plugin communicates. More about that in the warning [here](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug#octoprint-tplinksmartplug).

fearlesschicken, 2020-08-02:
FWIW, I downloaded the latest tplink_smartplug.py this still works

python tplink_smartplug.py  -t 192.168.43.171 -c info
('Sent:     ', '{"system":{"get_sysinfo":{}}}')
('Received: ', '{"system":{"get_sysinfo":{"sw_ver":"1.5.6 Build 191118 Rel.140307","hw_ver":"2.0","type":"IOT.SMARTPLUGSWITCH","model":"HS100(US)","mac":"<redacted>","dev_name":"Smart Wi-Fi Plug","alias":"Ender 3 Pro","relay_state":0,"on_time":0,"active_mode":"none","feature":"TIM","updating":0,"icon_hash":"","rssi":-56,"led_off":0,"longitude_i":<redacted>,"latitude_i":<redacted>,"hwId":"0DC<redacted>277E","fwId":"000000000<redacted>000000000","deviceId":"80<redacted>3BC63946795C<redacted>F3FA1","oemId":"80CB8<redacted>A1","next_action":{"type":-1},"err_code":0}}}')

jneilliii, 2020-08-02:
Yeah, so my plugin should be working then, because it's the same guts. Did you run that command from your pi within the `~/oprint/bin` folder or from another machine/location? I'm totally confused at this point why it would just stop working. What else have your tried, ie unplugging from the wall, restarting the pi, etc.? Do you see any errors in octoprint.log or the browser's console on initial page load?

fearlesschicken, 2020-08-02:
I ran several the python script as well as hs100.sh (https://github.com/ggeorgovassilis/linuxscripts/blob/master/tp-link-hs100-smartplug/hs100.sh) from several different machines with success.   At this point, I'm not sure what I did to get this working but it did NOT involve setting the plug to local only.  I would say this problem is not with this plugin but something on the plug side that prevents the octopi from communicating. I constantly got tcp errors:

./hs100.sh 192.168.43.171 9999 off
args are 192.168.43.171 9999 off
nc: connect to 192.168.43.171 port 9999 (tcp) failed: No route to host

and with some combination of connecting to the device at the same time on another machine, my octopi finally connected
pi@octopi:~/oprint/bin $ ./hs100.sh 192.168.43.171 9999 off
args are 192.168.43.171 9999 off
Connection to 192.168.43.171 9999 port [tcp/*] succeeded!
^C

After this, I checked with the plugin on the octoprint interface and verified its working again.

closed 2020-08-23 by stale[bot]

Appa1990, 2023-01-03:
I updated my octoprint today (completly reinstalled on the sd card). Because i run on an old python version. I now get the same error... No clue what to do now. Kasa works fine.

[2023-01-03 14:51:24,699] DEBUG: ImmutableMultiDict([('checkStatus', '100.100.100.34')])
[2023-01-03 14:51:24,700] DEBUG: Checking status of 100.100.100.34.
[2023-01-03 14:51:24,706] DEBUG: {'system': {'get_sysinfo': {}}}
[2023-01-03 14:51:24,706] DEBUG: IP 100.100.100.34 is valid.
[2023-01-03 14:51:24,707] DEBUG: Sending command {'system': {'get_sysinfo': {}}} to 100.100.100.34
[2023-01-03 14:51:27,833] DEBUG: Could not connect to 100.100.100.34.
[2023-01-03 14:51:27,834] DEBUG: 
[2023-01-03 14:51:27,834] DEBUG: {'system': {'get_sysinfo': {'relay_state': 3}}, 'emeter': {'err_code': True}}
[2023-01-03 14:53:39,714] DEBUG: ImmutableMultiDict([('checkStatus', '100.100.100.34')])
[2023-01-03 14:53:39,715] DEBUG: Checking status of 100.100.100.34.
[2023-01-03 14:53:39,715] DEBUG: {'system': {'get_sysinfo': {}}}
[2023-01-03 14:53:39,715] DEBUG: IP 100.100.100.34 is valid.
[2023-01-03 14:53:39,715] DEBUG: Sending command {'system': {'get_sysinfo': {}}} to 100.100.100.34
[2023-01-03 14:53:42,872] DEBUG: Could not connect to 100.100.100.34.
[2023-01-03 14:53:42,874] DEBUG: 
[2023-01-03 14:53:42,874] DEBUG: {'system': {'get_sysinfo': {'relay_state': 3}}, 'emeter': {'err_code': True}}

jneilliii, 2023-01-03:
Any errors in octoprint.log? So the plug is seen but it's not responding to commands. Can you try unplugging it and plugging it back in, verify the IP didn't change from your router, adjust plugin settings accordingly and try again. It's possibly your firmware updated and is no longer locally controllable.