jneilliii/OctoPrint-TPLinkSmartplug #183: Plug not giving any data

Peasndcarrots, 2020-06-15:
My plug isn't providing any power data. Is there a setting I need to implement to get this info to show up?

[image]

[image]

jneilliii, 2020-06-15:
What model plug do you have? Not all support power data. Typically enabling polling and restarting octoprint will get out to show the data if your model is power enabled.

Peasndcarrots, 2020-06-15:
I have the HS100. I enabled polling and restarted, but unfortunately that didn't work.

jneilliii, 2020-06-16:
Yep, that's the problem. That model is not a power reporting version. You'd have to have an HS110 or other device that includes "power monitoring" as a feature.

closed 2020-06-16 by jneilliii