jneilliii/OctoPrint-TPLinkSmartplug #184: [BUG]: When I press the + button in OctoPrint to add the plug nothing happens.

reynolpe, 2020-06-19:
**Describe the bug**
When I press the + button in OctoPrint to add the plug nothing happens.  

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'TP-Link Smartplug Settings'
2. Click on '+ button'
3. Nothing happens.  The window does not pop up to put in the IP address of the plug.

**Expected behavior**
When I press the + button in OctoPrint to add the plug nothing happens. 

**Desktop (please complete the following information):**
 - OS: Whatever linux is used with the Octoprint image
 - Browser Edge Chromium and Chrome
 - OctoPrint Version 1.4.0
 - Plugin Version: 0.9.21

nmota1974, 2020-06-19:
I have the same problem.

jneilliii, 2020-06-19:
Sounds like settings corruption. Did you install an upgrade or from scratch? Might need to get you to check config.yaml to ser what is listed in the plugin's settings.

Tron4880, 2020-06-19:
Hi,  I have the same problem, I uninstalled and cleaned up from the last build and installed from scratch

jneilliii, 2020-06-19:
I'm double-checking things now. I think I might know what it is.

jneilliii, 2020-06-19:
Please test this version and let me know if it fixes the issue. Use the URL below in plugin manager.

https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/archive/0.9.22.zip

nmota1974, 2020-06-19:
This version works for me.

jneilliii, 2020-06-19:
Thanks, bumping version and releasing update.

closed 2020-06-19 by jneilliii

reynolpe, 2020-06-20:
So the new plug worked for me and let me add the TP Switch but I can't check any of the buttons and the power icon shows up but does not toggle the power.  Kasa app on my phone toggles the power so I know the switch is working.

jneilliii, 2020-06-20:
Enable debug logging in the plugin's settings, restart octoprint, try the process, if it doesn't work grab octoprint.log and plugin_tplinksmartplug_debug.log files from octoprint's logging section in settings and upload here. Include screenshot of your settings for this plugin. There have been reports of newer firmware blocking this communication, as noted on the readme.

reynolpe, 2020-06-20:
Here are the logs.

[plugin_tplinksmartplug_debug.log](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/files/4808395/plugin_tplinksmartplug_debug.log)
[octoprint.log](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/files/4808396/octoprint.log)

jneilliii, 2020-06-20:
```
[2020-06-20 13:31:40,130] DEBUG: IP 192.168.1.55 is valid.
[2020-06-20 13:31:40,131] DEBUG: Sending command {'system': {'get_sysinfo': {}}} to 192.168.1.55
[2020-06-20 13:31:40,132] DEBUG: Could not connect to 192.168.1.55.
```
Yep, looks like you might be having the issue with recent firmware releases. There's nothing I can do to fix that issue unfortunately until someone is able to reverse engineer the connection again.

jneilliii, 2020-06-20:
@reynolpe could you let me know what Model, HW version and Firmware version are shown for your device in the Kasa App?

reynolpe, 2020-06-20:
Have you heard of a way to roll back firmware on a TP-Link plug?

jneilliii, 2020-06-20:
The factory reset procedure is listed below, but there was a post on another repository that uses the same methods I use to communicate with the plugs that this didn't work for them. It may also be related to being in Remote Control (cloud connected), but switching to Local Only mode didn't appear to be promising either.

> Press and hold (about 5 seconds) until the Wi-Fi LED blinks amber and green alternately to initiate the smart-config process, without losing the configuration settings.
> Press and hold (about 10 seconds) until the WI-Fi LED blinks amber quickly to reset the plug to factory defaults.

reynolpe, 2020-06-20:
I worked with TP-Link and they had me factory reset also and it is working now.  Still has the new firmware on it though.

jneilliii, 2020-06-20:
Weird. I have seen occasionally just unplugging the device when it starts to stop working and plugging it back in fixes some issues, but assumed you already tried that. Was the procedure above the correct factory reset procedure?

jneilliii, 2020-06-20:
And did you keep it in Local Only mode or enable remote control?

reynolpe, 2020-06-20:
Screenshots of my settings. Not sure what local only means?  You mean open
across the Internet?

[image: image.png]

[image: image.png]

jneilliii, 2020-06-20:
screenshots don't come through on email replies unfortunately. Local Only mode means not enabling "Remote Control" option, ie not be able to control the plug outside of your local network but through their cloud service.

reynolpe, 2020-06-20:
I don’t have remote enabled.