jneilliii/OctoPrint-TPLinkSmartplug #186: [BUG]: GCODE M81 not working since 0.9.22 release

rominetb44, 2020-06-21:
Hello,

Since 0.9.22 update and after reboot the system (rapsberry pi 4), GCODE M81 command does not work anymore (ok for M80).

Terminal : 
Send: M80 192.168.0.44
Recv: ok
[...]
Send: M81 192.168.0.44
Recv: Deletion(bin) failed.
Recv: ok

octoprint.log : 
2020-06-21 14:00:22,182 - octoprint.plugins.tplinksmartplug - INFO - PowerOn: None
2020-06-21 14:00:29,617 - octoprint.plugins.tplinksmartplug - INFO - PowerOff: None

plugin_tplinksmartplug_debug.log : 
[2020-06-21 14:00:22,159] DEBUG: Received M80 command, attempting power on of 192.168.0.44.
[2020-06-21 14:00:22,161] DEBUG: {'autoConnect': True, 'ip': '192.168.0.44', 'btnColor': '#808080', 'autoDisconnectDelay': 0, 'sysCmdOff': False, 'gcodeEnabled': True, 'gcodeOnDelay': 0, 'emeter': {'get_realtime': {}}, 'countdownOnDelay': 1, 'label': 'TP-LINK', 'warnPrinting': False, 'thermal_runaway': False, 'gcodeOffDelay': 0, 'sysCmdOnDelay': 0, 'event_on_error': False, 'countdownOffDelay': 1, 'sysRunCmdOff': '', 'sysRunCmdOn': '', 'autoDisconnect': True, 'automaticShutdownEnabled': True, 'event_on_upload': False, 'useCountdownRules': False, 'icon': 'icon-bolt', 'displayWarning': True, 'autoConnectDelay': 10, 'event_on_disconnect': False, 'sysCmdOn': False, 'currentState': 'on', 'sysCmdOffDelay': 0}
[2020-06-21 14:00:22,162] DEBUG: Turning on 192.168.0.44.
[2020-06-21 14:00:22,171] DEBUG: {'autoConnect': True, 'ip': '192.168.0.44', 'btnColor': '#808080', 'autoDisconnectDelay': 0, 'sysCmdOff': False, 'gcodeEnabled': True, 'gcodeOnDelay': 0, 'emeter': {'get_realtime': {}}, 'countdownOnDelay': 1, 'label': 'TP-LINK', 'warnPrinting': False, 'thermal_runaway': False, 'gcodeOffDelay': 0, 'sysCmdOnDelay': 0, 'event_on_error': False, 'countdownOffDelay': 1, 'sysRunCmdOff': '', 'sysRunCmdOn': '', 'autoDisconnect': True, 'automaticShutdownEnabled': True, 'event_on_upload': False, 'useCountdownRules': False, 'icon': 'icon-bolt', 'displayWarning': True, 'autoConnectDelay': 10, 'event_on_disconnect': False, 'sysCmdOn': False, 'currentState': 'on', 'sysCmdOffDelay': 0}
[2020-06-21 14:00:22,173] DEBUG: IP 192.168.0.44 is valid.
[2020-06-21 14:00:22,173] DEBUG: Sending command {'system': {'set_relay_state': {'state': 1}}} to 192.168.0.44
[2020-06-21 14:00:22,338] DEBUG: «-ý"system":{"set_relay_state":{"err_code":0}}}
[2020-06-21 14:00:22,345] DEBUG: 0
[2020-06-21 14:00:22,346] DEBUG: Checking status of 192.168.0.44.
[2020-06-21 14:00:22,346] DEBUG: {'system': {'get_sysinfo': {}}}
[2020-06-21 14:00:22,347] DEBUG: IP 192.168.0.44 is valid.
[2020-06-21 14:00:22,347] DEBUG: Sending command {'system': {'get_sysinfo': {}}} to 192.168.0.44
[2020-06-21 14:00:22,545] DEBUG: «E�"system":{"get_sysinfo":{"sw_ver":"1.5.5 Build 191125 Rel.114303","hw_ver":"2.0","type":"IOT.SMARTPLUGSWITCH","model":"HS100(FR)","mac":"50:D4:F7:64:80:DF","dev_name":"Smart Wi-Fi Plug","alias":"Ender3","relay_state":1,"on_time":0,"active_mode":"none","feature":"TIM","updating":0,"icon_hash":"","rssi":-79,"led_off":0,"longitude_i":-21615,"latitude_i":472471,"hwId":"11D803E325E41BCC381AE00341B472B4","fwId":"00000000000000000000000000000000","deviceId":"8006A9F2C026A768F62D830113DA9E101BF00781","oemId":"1967B9D71C3666B0D196769EE4452D38","next_action":{"type":-1},"err_code":0}}}
[2020-06-21 14:00:22,588] DEBUG: TIM

Thanks for help.

Regards.

jneilliii, 2020-06-21:
I'm not familiar with that response `Recv: Deletion(bin) failed.` what is that? Shouldn't effect this situation though. Can you share your plugin's default settings as well as the plug's individual settings please?

rominetb44, 2020-06-21:
Same issue with 0.9.21 and works fine with 0.9.20.

This is my settings : 
[image]
[image]

Thanks.

rominetb44, 2020-06-21:
I re-install the 0.9.20 release and it works fine.
This this what I get in the terminal : 

Send: M81 192.168.0.44
Recv: Deletion(bin) failed.
Connection closed, closing down monitor
Changing monitoring state from "Operational" to "Offline"

jneilliii, 2020-06-21:
Yep, increase the gcode off delay in newer versions because the phase being used to detect the gcode was changed. It's getting blocked because when the off command is detected in gcode it gets blocked by the printer still being in a "printing" state. Once that is increased it should work fine.

rominetb44, 2020-06-21:
Hello,

Thank you.
I tried with 5, 20, 30 and 60 seconds and it's the same result.

I will keep the 0.9.20 release.

Thanks.

undeliverable, 2020-06-21:
I am also not able to get the M81 shutdown command to work after 9.20
I dont get the 'Deletion (bin) Failed' error.  It just never shuts off.  If it helps I can roll forward to 9.22 and get some logs.

jneilliii, 2020-06-21:
Logs are always helpful.

jneilliii, 2020-06-21:
Curious if the `@TPLINKOFF <ip>` works or not?

undeliverable, 2020-06-21:
I will test and get some logs as soon as my current print is complete.  Should be about 2 hours.

SeekaMartin, 2020-06-22:
I am having the same issue. how can i get logs? @tplinkoff works however

jneilliii, 2020-06-22:
Enable debug logging in settings, make sure your gcode off delay is long enough for your printer's buffer to clear and the print to finish, try 60 to start with for example and restart octoprint. Send a print, go into octoprint's logging section in settings and grab plugin_tplinksmartplug_debug.log. Drag it into a comment here to upload. 

Glad to know there's a functional work around using the @ command.

bvalat, 2020-06-22:
Same problem here.
Switch printer on and off works.
But auto off with shutdown printer plugin is not working anymore.
Octoprint log  (Should'nt be the IP instead of the None?) :
2020-06-22 18:16:39,971 - octoprint.plugins.shutdownprinter - INFO - Starting abort shutdown printer timer.
2020-06-22 18:17:10,048 - octoprint.plugins.shutdownprinter - INFO - Shutting down printer with command: M81 192.168.1.40
2020-06-22 18:17:10,091 - octoprint.plugins.tplinksmartplug - INFO - PowerOff: None

No logs at all one the tplinksmartplug plugin even if i put it in debug.
[octoprint (1).log](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/files/4814708/octoprint.1.log)
[plugin_tplinksmartplug_debug.log](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/files/4814709/plugin_tplinksmartplug_debug.log)

bvalat, 2020-06-22:
https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/pull/178 definitivly broke somthing for gcode parsing.
In __init__.py line 900: if gcode == "M80":
You never do a if/elif for M81

jneilliii, 2020-06-22:
You are completely correct. Just haven't had the time to investigate yet, thanks for the check. That was a refactoring error on my part. Working to resolve that now.

bvalat, 2020-06-22:
Glad i could help :)

jneilliii, 2020-06-22:
Test version ready for you guys, you can install with this URL in plugin manager.

https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/archive/0.9.23.zip

bvalat, 2020-06-22:
cmd?  shouldn't it be gcode instead?
if gcode == "M80":
if cmd == "M81":

I tested anyway : not working.

bvalat, 2020-06-22:
I can confirm with `if gcode == "M81":` it's working 
I tested with the test button in shutdown printer plugin.

jneilliii, 2020-06-22:
You are correct, I just pushed an update that should fix that. Try again using the URL linked above.

bvalat, 2020-06-22:
First tests are OK (with test buton from other plugin)
I let @rominetb44  confirm if everything is fine for him as well.

closed 2020-06-23 by jneilliii

rominetb44, 2020-06-23:
It works fine. Than you.

PR #187 (2020-06-23): 0.9.23
* Fix M81 issue, resolves #186
* add new supporters