jneilliii/OctoPrint-TPLinkSmartplug #190: [BUG]: On File Upload Doesn't Trigger

ryancole38, 2020-07-05:
**Describe the bug**
I'm unable to get my smart plug to turn on when a file is uploaded.  I can turn the outlet on with the icon just fine, but it will not automatically trigger when I upload a file.

**To Reproduce**
Steps to reproduce the behavior:
1. Install the plugin via plugin manager
2. Go to plugin settings
3. Set settings as shown in screenshots
4. Upload file
5. Outlet does not trigger
**Expected behavior**
I expect to be able to upload a file, and then have the smart outlet trigger.

**Screenshots**
[image]
[image]
**Desktop (please complete the following information):**
 - OS: macOS and Windows
 - Browser: Safari (and Chrome on Windows)
 - OctoPrint Version: 1.4.0
 - Plugin Version: 0.9.23

**Additional context**
In release 0.9.22, the release notes state "enhance upload event monitoring to only process when auto print is enabled, requires OctoPrint version 1.4.1" but I could not find OctoPrint release 1.4.1, I'm assuming this refers to release 1.4.0.

The aforementioned release also mentions auto print, but I could not find any official OctoPrint reference to auto print.  I installed the plugin "Autoselect" to see if that would make it work, but that did not help.  I do not have any other non-stock plugins installed.

jneilliii, 2020-07-05:
OctoPrint version 1.4.1 is correct and is getting close to becoming a release candidate. You'd have to manually upgrade to the maintenance branch in order for that functionality to work. It should be usable once that version is released.

ryancole38, 2020-07-06:
Wow, you weren't kidding about it being close to release!  I still couldn't get it to work on the maintenance branch or the newest 1.4.1rc, but I'm guessing I'm missing the auto print that was alluded to.  I've dug through all of the settings but haven't found anything, could you point me in the right direction?

I had also wondered if there needed to be an active connection to the printer for this to work.  I'm using "uhubctl" to turn off my USB ports when the smart outlet is off, but I'm still having the same problem with an active connection to the printer.

jneilliii, 2020-07-06:
So the auto print options I was alluding to is from the API file upload process. It's a checkbox in PrusaSlicer when you send directly to OctoPrint. I see your point now, this may not be ideal for all use cases.

ryancole38, 2020-07-06:
I actually had just found the checkbox in Cura- works perfectly now. Thanks for the help, and thanks for all the plugins you create!

jneilliii, 2020-07-06:
Oh if you're using Cura you should use Cura's OctoPrint Connection plugin from fieldOfView (available in the marketplace). It detects all my power related plugins and will allow you to manage that as well directly from Cura, rather than depend on this plugin to power on your printer. It will also detect the presence of my UltimakerFormatPackage plugin in OctoPrint and upload in that format to give you thumbnail support.

ryancole38, 2020-07-06:
Oh, awesome! Yeah, I'm using fieldOfView's plugin. I'll also have to install UltimakerFormatPackage, which looks great.

ShagoY, 2020-11-19:
hi,
I have version 1.4.2 and it doesn't work, is this normal?
I have "upload Event monitoring" and "On with upload" checked.
I upload with the PrusaSlicer API.

jneilliii, 2020-11-19:
not expected ShagoY, please open a new issue with debug logs to investigate further.

ShagoY, 2020-11-20:
Ok i need to check "Start print after download" for trigger "On with upload" on PrusaSlicer.
I don't know if it's wanted or if it's possible to trigger it without this option.
Btw i won't bother you with that :)
Thank !

jneilliii, 2020-11-20:
> Ok i need to check "Start print after download" for trigger "On with upload" on PrusaSlicer.

That is correct and was the intended design.

timur-tabi, 2021-03-18:
> > Ok i need to check "Start print after download" for trigger "On with upload" on PrusaSlicer.
> 
> That is correct and was the intended design.

Can you change this?  I don't want to start the print immediately and automatically after upload.  I just want the printer to be turned on.

jneilliii, 2021-03-19:
What would the workflow be here @timur-tabi in this case? If you're just uploading and powering on, then wouldn't you have to go to the web interface to start the print? In that case you could just use the navbar to power on the printer.

timur-tabi, 2021-03-19:
For starters, even if I select "start print on upload" from the PrusaSlicer dialog box, it doesn't actually start the print if the printer is currently off.  TPLink does turn the printer on, but that's it.  So I guess technically it's already doing what I want.

I do see your point.  However, I'm a little paranoid so I usually want to check the printer in octoprint before I actually start printing.  I thought this feature could be used to save me a few seconds by turning on the printer right away.

jneilliii, 2021-03-19:
> For starters, even if I select "start print on upload" from the PrusaSlicer dialog box, it doesn't actually start the print if the printer is currently off. TPLink does turn the printer on, but that's it. So I guess technically it's already doing what I want.

That might be the case if autoconnect isn't enabled. But I have been using the autostart on upload feature for quite some time and it does work. But I see no reason why I couldn't change the logic here and have power on with upload, and then detect the auto start option if enabled.

timur-tabi, 2021-03-19:
Autoconnect is enabled for me, but it takes almost 10 seconds before Octoprint is fully online after turning the printer on.

jneilliii, 2022-04-14:
I've added the option to Always on Upload and will be available in the next release candidate published.

PR #304 (2022-04-17): 1.0.2
* resolves issue related to shutdown monitoring not powering off devices due to SIGTERM, resolves #274
* fix timelapse race condition, resolves for #281
* add JSON response to on/off API commands
* add `Always on Upload` option to not require start print from slicer upload, resolves #190, resolves #267
* replace deprecated getBoolen with get_boolean
* remove duplicated bundling of fontawesome 5 if running on OctoPrint 1.5.0+
closed 2022-04-17 by jneilliii