jneilliii/OctoPrint-TPLinkSmartplug #192: [BUG]:

ibsmith04, 2020-07-10:
**Describe the bug**
A clear and concise description of what the bug is.

I have just installed a Octoprint with TPLinksmartplug on my new pi4b to control my ender 3 fitted with an SKR1.4 32 bit board.

I am trying to get the plugin to turn off my pi, wait for 40 seconds, then turn off the power supply to the pi.

I am a little confused by the following screen

[image]

If I hover over the input for "Run System Command Before OFF", I get a little help bubble saying "System Command to run AFTER powering off this plug".

I tested powering off the pi, with putty open and connected to the pi.

When I turn off the Pi using smartplug, what appeared to happen was
After about 5-10 seconds, octoprint disconnected from my printer, although putty was still active.
I kept hitting enter in putty up to the 40 seconds elapsed, and at that point the plug powered off, even though the pi was still running.

I then tried issuing the command sudo shutdown -h now through putty
This shutdown the pi, and the putty screen lost connection immediately, although it took a little longer for the pi to completely close down.

So to me it looks like the paraemeter is run after powering off the plug, wherease the label says before. I want it to be before.

Or have I got this horribly wrong with an incorrect setup ?

I do have a 2nd smart plug on the printer itself, and that works great.

**To Reproduce**
Steps to reproduce the behavior:
Setup smartplug as shown above

**Expected behavior**
I expected it to shutdown the pi, and after 40 seconds to turn off the plug.
What it actually did was diconnect octoprint from the printer, waited 40 seconds then switched off the plug.

**Screenshots**
see above

**Desktop (please complete the following information):**
 - OS: [e.g. iOS] - Windows 10
 - Browser [e.g. chrome, safari] Chrome
 - OctoPrint Version [e.g. 22] 1.4
 - Plugin Version 0.9.23

**Additional context**
Add any other context about the problem here.

jneilliii, 2020-07-11:
> I then tried issuing the command sudo shutdown -h now through putty

Is it possibly because of the capitalization of the command shutdown in your settings?

ibsmith04, 2020-07-12:
Thank you jneilliii, Now working perfectly, although the Help bubble that pops up when you hover over the input field is incorrect.

Thanks again
Ian

closed 2020-07-12 by ibsmith04