jneilliii/OctoPrint-TPLinkSmartplug #193: [BUG]: Plugin stops OctoPrint from firing Shutdown event, possibly due to use of threadinf

cp2004, 2020-07-16:
**Describe the bug**
On shutdown of OctoPrint, this plugin's thread's are not terminated so Python doesn't exit cleanly. OctoPrint waits for 15 secs, before forcing everything to terminate. 

**To Reproduce**
Steps to reproduce the behavior:
1. Enable OctoPrint-TPLinkSmartplug
2. Try to shutdown OctoPrint using the web interface, or `ctrl-c` on the command line.
3. Watch logs, look for the shutdown event.

**Expected behavior**
OctoPrint to exit cleanly, and record a shutdown event

**Screenshots**
Logs: 
https://gist.github.com/cp2004/213d96ba951a24d3e74e4e48ffbd5a0d

Note in the second file, lots of plugins are disabled. This just part I found where TPLink is disabled, later on in the testing I narrowed it down to TPLink, verified it is not a conflict with other plugins.

**Desktop (please complete the following information):**
 - OS: OctoPi 0.17.0
 - Browser Chrome
 - OctoPrint Version 0.4.0/0.4.1rc1/2/3
 - Plugin Version 0.9.23

**Additional context**
As discussed on discord, I am willing to fix this, however not for 2 weeks or so. I believe it is something to do with the threading, daemon threads and not cancelling timers, but need to look into it further. Will PR if I can fix it.

cp2004, 2020-08-06:
Haven't forgotten about this! Did some preliminary investigation, is the Resettable timer run as a daemon? It's the non-daemon threads (almost definitely) that cause this.
Here's what I did:
1. Made all uses of `threading.Timer` daemon - didn't seem to solve it
2. Commented out all `resettable timer` lines. Probably broken the functionality of the plugin, but it seemed to shutdown correctly.

Now I'm not *certain* about these, but have a pretty good idea.

At some point I can see if I can make the resettable timer a daemon, and then PR if it works.

cp2004, 2020-08-09:
Have made the resettable timer daemon, and it appears to work well. Although I made the mistake of making the changes on your repo, not my fork so I can't push them 🤦 Will get it sorted out and send a PR in to fix this. Once I've tested it against my plugin too, to check the orginal issue is truly fixed.

cp2004, 2020-08-09:
PR sent :) was easier than I thought

#200

jneilliii, 2020-08-10:
Thanks @cp2004. I've merged the change into my dev branch and was going to see if it will resolve the issue with the CPU load too prior to official merge.

cp2004, 2020-08-10:
I'm going to say I doubt it, as my problem is only with the shutdown of the Python interpreter - only having an effect at shutdown. If you want any more help with the threading stuff I can give it a go.

PR #200 (2020-08-10): Fix use of threading to be all daemons
Would close #193 
Threads used in OctoPrint should be daemon threads, as this allows it to shutdown cleanly. Issue means that plugin using multiprocessing meant OctoPrint failed to restart correctly. 

This PR sets all threads to `daemon = True`, so they do not block interpreter shutdown.
Still yet to test if this fixes the issue with my plugin, but OctoPrint now exits with the full `on_shutdown` logging etc. so I am confident it will.
closed 2020-08-11 by cp2004