jneilliii/OctoPrint-TPLinkSmartplug #194: [Feature Request]: Toggle smart plug state based on state change of RPi's GPIO pin

nickbuhling, 2020-07-18:
Let's start by saying that I love this plugin; great work!

Currently I have a relay on one of my printers that I can toggle by flipping a switch. The RPi detects the state change and toggles the state of the relay. With this setup I use the PSU-control plugin. The problem is that I cannot shut the printer down from outside my network. With the TP-Link I can always use their app. Because the outlet is on the backside of the printer's enclosure, I would like to wire a switch on the front of the enclosure to my RPi and use a GPIO input to toggle the state of the TP-Link, however currently this is not supported if I'm not mistaken. I'd love to hear whether this is a feature that could be included?

Thanks for your consideration and keep up the good work.

jneilliii, 2020-07-18:
I'm not too familiar with GPIO programming for the Pi, but if you know of a way to monitor that button press outside of OctoPrint you could run a curl command that calls the plugin's [API](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/wiki/API#curl-examples) to turn on/off. This is not something that I would integrate within the plugin itself though because of the additional complexity of GPIO python programming. The plugin has already kind of gotten a little out of control with the current feature set.

closed 2020-07-18 by jneilliii

jneilliii, 2020-07-18:
You might also be able to get away with using the plugin below and enable error monitoring in the TPLink plugin and your plug's settings.

https://plugins.octoprint.org/plugins/emergencystopsimplified/

nickbuhling, 2020-07-21:
Thank you for the reply. I was thinking you could add it since the plugin is based on the PSU control plugin which already has this feature. It is the 'sense gpio' feature. Is there not a way to take that feature and integrate it in this plugin? In my opinion this one setting would make the plugin much more valuable.

I know very little about programming for RPi's so unfortunately can not write it myself.  If this is not an option I might try the curl script route however I have no idea how that works yet.

Anyway, thanks for the already great plugin again ;).