jneilliii/OctoPrint-TPLinkSmartplug #195: [BUG]: P100 (tapo plug) doesn't connect

Doormat1, 2020-07-19:
**Describe the bug**
I have a brand new Tapo TP link P100 and I can control it from my phone but not via octoprint

**To Reproduce**
Steps to reproduce the behavior:
I have followed the set up and have the correct ip for the plug but all I get is a grey icon at the top of the page

**Expected behavior**
Octoprint to control the plug

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 Latest version of the plug in
Octoprint 1.4
Plug firmware 1.1.8
**Additional context**
Add any other context about the problem here.
I assume it may not work because it's a tapo not a kasa?

jneilliii, 2020-07-19:
Yeah, Kapo devices are not supported, and never will be unfortunately. Kasa devices are the only ones that can be controlled locally outside of a tplink app.

Doormat1, 2020-07-19:
Ah OK. Might have to send this plug back then and get a new one.

jneilliii, 2020-07-19:
I just updated the readme here to hopefully make that more clear. Will probably update the plugin repository too.

closed 2020-07-19 by jneilliii