jneilliii/OctoPrint-TPLinkSmartplug #196: [Feature Request]: Use browser locale for graph time

jw59, 2020-07-25:
It'd be handy if the graph times were presented in my local timezone.  Looks like they're currently in GMT.   Its 9pm local when this screenshot was taken on the 24th.  Confusing to see the 25th on the graph.  

[image]

jneilliii, 2020-07-25:
The record date time should be pulled from the pi's local settings, so you might want to double check those settings via `sudo raspi-config`.

tideline3d, 2020-07-26:
That did fix my issue, but all the other Date/Time's on OctoPrint UI were just following browser locale, so I didn't think to look there.

closed 2020-07-26 by jneilliii

jneilliii, 2020-07-26:
sorry @tideline3d, the data logging has no knowledge of the browser and can be polling status even while your browser is closed. It's easy enough to configure your device for appropriate timezone/etc. I don't feel it's worth the programming effort to work around it.

tideline3d, 2020-07-26:
Makes sense, didn't think about that the logging is done on the backend.