jneilliii/OctoPrint-TPLinkSmartplug #197: [BUG]: Lightbulb icon takes a long time to load

timur-tabi, 2020-07-27:
I have two TP link devices attached, one for the printer and another for some lights.  I assigned the "far fa-lightbulb" icon to the light device and "fas fa-print" for the printer.

When I load the Octoprint web page for the first time, the "fas fa-print" icon loads immediately, but the "far fa-lightbulb" icon initially appears as some generic unicode box icon before changing into a light bulb after about 5-10 seconds.

I don't know anything about HTML fonts, but I suspect that there are some things that can be done to speed up the loading and displaying of the lightbulb font.

[octoprint.log](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/files/4985022/octoprint.log)

jneilliii, 2020-07-27:
I'm unsure if there's anything I can do to resolve this. Both fonts come from the same webfont library so I don't know why one would load quicker than the other.

timur-tabi, 2020-07-28:
Well, the light bulb icon is "far fa-lightbulb" and the printer icon is "fas fa-print".  What's the difference between "far" and "fas"?

jneilliii, 2020-07-28:
Good point, `far` means a regular fontawesome font, where `fas` indicates a solid fontawesome font. Those do come from different files, but the strange part is that the solid file is much larger than the regular file. 

[image]

What browser are you using, I do see a slightly slower load of the bulb but barely noticeable, not the 5-10 second load time you are mentioning. Do you have any other plugins installed that may also include a fontawesome library like TouchUI?

jneilliii, 2020-07-28:
Try manually entering the icon `icon-lightbulb` in the settings and I think that will improve your situation. That will load the icon from a different font library and seems to load as quickly as print.

timur-tabi, 2020-08-03:
Using icon-lightbulb doesn't change anything.  This only happens on my Android tablet.  My PC either already has the font downloaded or is just a lot faster, I don't know.  But I guess there's not much that can be done.

closed 2020-08-03 by timur-tabi

timur-tabi, 2020-08-05:
I do have one more observation.  I think the problem occurs when there are two devices.  The first one loads fine.  The second one appears to be stuck at the generic unicode icon until the rest of the web page loads.  The problem is more noticeable when there are several update notifications (new versions of plugins, etc).  So I see the generic icon, and then I the other notifications pop up one at a time, and then only after the entire page is loaded, is the icon corrected.

So my guess is that this is a side-effect of the threading model of the browser.

timur-tabi, 2022-06-22:
I have another observation.  The problem is apparently that my browser sometimes stalls when it tries to load http://octopi.local/static/vendor/font-awesome-3.2.1/fonts/fontawesome-webfont.woff.