jneilliii/OctoPrint-TPLinkSmartplug #204: [Question]: Is it possible to power on the smart plug when Octoprint starts?

Alwinator, 2020-08-17:
It would be very convenient. By the way thank you for your great plugin! Thank you in advance.

jneilliii, 2020-08-17:
There is a way, although a little convoluted. Create a new file called startup.sh in `/home/pi/.octoprint/scripts/` and put this into it, replacing the API KEY (you can get that from your user settings) and plug's ip address.
```
#!/bin/bash
curl -s -H "Content-Type: application/json" -H "X-Api-Key: PUTYOUTAPIKEYHERE" -X POST -d '{"command":"turnOn", "ip":"PUTYOURPLUGSIPHERE"}' https://127.0.0.1/api/plugin/tplinksmartplug
```

after creating the file you need to run this command to make it executable.

```
chmod +x /home/pi/.octoprint/scripts/startup.sh
```

and then edit your config.yaml to include the startup event by adding this.

```
events:
    subscriptions:
    -   command: /home/pi/.octoprint/scripts/startup.sh
        event: Startup
        type: system
```

Alwinator, 2020-08-17:
Thanks a lot for your answer! But it doesn't work yet.

> #!/bin/bash
> curl -s -H "Content-Type: application/json" -H "X-Api-Key: PUTYOUTAPIKEYHERE" -X POST -d '{"command":"turnOn", "ip":"PUTYOURPLUGSIPHERE"}' https://127.0.0.1/api/plugin/tplinksmartplug

Works after removing the S from HTTPS. Now it turns on my printer like it should. 

> events:
>     subscriptions:
>     -   command: /home/pi/.octoprint/scripts/startup.sh
>         event: Startup
>         type: system

When I add this lines to my config.yaml and restart Octoprint it crashes. The status is active (exited). Nothing written to  ~/.octoprint/logs/octoprint.log

Do you have an idea?

jneilliii, 2020-08-17:
The space indentation of those lines are extremely important in config.yaml. Just make sure that you match up the space indentation in the areas around it. Some config.yam files use 2 spaces, others use 4. Just make sure it matches the way it looks elsewhere.

Alwinator, 2020-08-18:
Thank you, now everything works!

I replaced the tabs with 4 spaces and this fixed the problem.

Then it showed the following error in the logs:
```
2020-08-18 13:42:43,555 - octoprint.events - WARNING - Command failed with return code 7: Command '/home/pi/.octoprint/scripts/startup.sh' returned non-zero exit status 7
```

I fixed it by adding `sleep 5` in the startup.sh before the turn on command. It is definitely not the best way to solve this, but it works.

@jneilliii Is it possible to add this directly to the plugin? That you have a checkbox for every object in the web interface: Power on when Octoprint starts

jneilliii, 2020-08-18:
Yes, it's possible. I was just giving you a temporary solution until I'm able to get around to making the change.

Alwinator, 2020-08-18:
@jneilliii Thank you a lot! I am looking forward to the feature.

jneilliii, 2020-09-03:
Startup event monitoring implemented in the above commit and will be included in next release. If you want to try it out you can install by copying/pasting the URL below in plugin manager > get more > ...from URL and clicking install.

```
https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/archive/0.9.25.zip
```

Alwinator, 2020-09-03:
Thank you a lot for implementing, but I have just tested it and it does not work. When I just restart Octoprint it works, but when I restart my whole Raspberry Pi it does not work. I had exactly the same problem with my script. I have solved it with adding sleep 5 before the turn on command, but maybe there is a better way to solve this.

jneilliii, 2020-09-03:
Interesting, I'll have to do some additional testing with a full power up. I was just testing by restarting octoprint from the system menu.

Alwinator, 2020-09-04:
Okay, thank you!

Alwinator, 2020-11-13:
@jneilliii Thank you for your work, but I still have this problem:

> Thank you a lot for implementing, but I have just tested it and it does not work. When I just restart Octoprint it works, but when I restart my whole Raspberry Pi it does not work. I had exactly the [...]

jneilliii, 2020-11-13:
Sorry @Alwinator I forgot to test a full power on. Probably just need to move the stuff into a different startup event section.

jneilliii, 2020-11-14:
I think I got this startup issue really resolved now. I have tested with my pi here going from a cold boot and it works as expected here. If you want to test it out you can copy/paste the URL below into plugin manager > get more > ...from URL and clicking install.

```
https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/archive/0.9.26rc2.zip
```

Alwinator, 2020-11-16:
@jneilliii Thank you a lot! It works perfectly now!

jneilliii, 2020-11-16:
Great news, thanks for reporting back.

commit d65263c1 (2020-09-03): add startup event monitoring to automatically power on during octoprint startup, resolves #204
PR #220 (2020-11-12): 0.9.25
- fix typo in title for "Power Off on Idle" setting,
- fix power on with startup
- fix start print on upload logic, resolves #207
- add custom access permissions hook for controlling who can send API commands for controlling plugs
- add release channels available in OctoPrint 1.5.0
- add startup event monitoring to automatically power on during octoprint startup, resolves #204
- update knockout sortable module to version 1.2.0 to resolve knockout upgrade issue in OctoPrint 1.5.0
- update plotly to version 1.54.7 to resolve iPad 12.9 detection issues
closed 2020-11-21 by jneilliii