jneilliii/OctoPrint-TPLinkSmartplug #207: [BUG]: Print doesn't automatically start after printer turns on

chriscla, 2020-09-02:
I'm expecting this plugin to turn on the Printer and then start a print when uploading from PrusaSlicer. The printer will turn on using this plugin but the print won't automatically start. 

**To Reproduce**
Steps to reproduce the behavior:
0. Printer is off. 
1. In PrusaSlicer, press the Send to Printer button and select "Start printing after upload"
2. Printer turns on but doesn't start printing

**Expected behavior**
The printer turns on, waits the Auto-Connect duration and then starts printing.

**Screenshots**
Sending my config:

[image]

[image]

**Desktop (please complete the following information):**
 - OS: windows
 - Browser chrome
 - OctoPrint Version 1.4.2
 - Plugin Version 0.9.24

**Additional context**
This may be how it's expected to work or I messed up the config. Thanks for creating this plugin!

jneilliii, 2020-09-02:
Your settings do look correct, and your assumption is correct that is how it is intended to work. It does work that way for me, so to debug your issue I'll need you to enable debug logging, restart octoprint and put the plug into a powered off state, therefore turning your printer off. Then send the file from PrusaSlicer with that option enabled to auto start the print and the printer should power on, octoprint should then auto connect, and once the connection is established the file should be automatically selected and start printing. If it doesn't then I'll need to get your plugin_tplinksmartplug_debug.log file from OctoPrint's logging section.

jneilliii, 2020-09-02:
One clarification, when you say OS: windows, do you mean that's what octoprint is installed on or the machine that you're accessing octoprint from?

chriscla, 2020-09-02:
Octoprint is running on a Raspberry Pi 4 using the Octopi setup. I set this up a few days ago.   Attached are the logs. I'm new at deciphering this but I think that it's timing out trying to negotiate the serial connection.  

[plugin_tplinksmartplug_debug (2).log](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/files/5160050/plugin_tplinksmartplug_debug.2.log)
[octoprint (1).log](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/files/5160051/octoprint.1.log)

jneilliii, 2020-09-02:
Try manually setting your port and baud rate with the option save settings and connect to octoprint while it's powered on instead of using Auto. That way when the power on occurs and it tries to connect it will know what port to send the connect command. You may consider increasing the autoconnect delay too just in case the printer is not fully on after 20 seconds.

chriscla, 2020-09-02:
Yep, that helped.  I disabled "automatically connect to printer on startup" and ensured the baud rate was correct. So now, the printer turns on and it becomes operational.  It still doesn't start printing.
Updated logs:
[octoprint (3).log](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/files/5160129/octoprint.3.log)
[plugin_tplinksmartplug_debug (4).log](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/files/5160130/plugin_tplinksmartplug_debug.4.log)

The file does appear in the upload queue

chriscla, 2020-09-02:
I think I figured it out (I have it working now by modifying the plugin source):

After turning on the plug, the plugin starts a timer for connection

https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/blob/da780d4bff30e77e45780c41fd3944401c6dc67f/octoprint_tplinksmartplug/__init__.py#L34

```
			if plug["autoConnect"] and self._printer.is_closed_or_error():
				c = threading.Timer(int(plug["autoConnectDelay"]), self._printer.connect)
				c.daemon = True
				c.start()
```

[This method returns with whether the plugged turned on](https://github.com/jneilliii/OctoPrint-
TPLinkSmartplug/blob/da780d4bff30e77e45780c41fd3944401c6dc67f/octoprint_tplinksmartplug/__init__.py#L667).  There is then a sleep that sleeps for the autoConnectDelay + 1. It then checks if the printer is ready before kicking off the print:

```
if self._printer.is_ready() != False:
  self._tplinksmartplug_logger.debug("printer connected starting print of %s" % (payload.get("path", "")))
  self._printer.select_file(payload.get("path"), False, printAfterSelect=True)
```

I think it's failing because the connect method hasn't completed and the printer is not yet ready. I hacked this by adding 10 seconds to the sleep. I'm new to Octoprint so not sure the best way to fix this. I could see:

1. Adding a configurable connecting sleep. 
2. Checking the state_id and seeing if the state is "connecting" 
3. Add a signal so that the timer can signal to the main thread that the connect happened.

jneilliii, 2020-09-02:
Yeah, it would make sense to increase that sleep value, or potentially switched the logic to something like the below. I'll do some additional testing on this approach.

```
While True:
    if not self._printer.is_ready():
        return True
    self._printer.select_file(payload.get("path"), False, printAfterSelect=True)
```

jneilliii, 2020-09-03:
So I've changed the logic to how the auto start happens and moved it to the connected event in the above commit, This way we're not clogging up anything in a busy while loop. Will be included as part of the next release. If you want to test it out for me you can copy/paste the URL below into plugin manager > get more> ...from URL and click Install. The version will still show 0.9.24 as I tend to keep that at current release version until I'm done with additional fixes.

```
https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/archive/0.9.25.zip
```

chriscla, 2020-09-08:
Just installed the new version and uploading from Prusaslicer started it and the print started automatically after the machine turned on.  I'll do some more testing and report back if there are issues but "no news is good news". Thanks for working on this and making the plugin :)

chriscla, 2020-09-10:
Hey, I need to do more  testing but I think that if I turn the printer on using the "lightning bolt" in Octopi, then it will start printing the last print that was uploaded.  I'll do some more testing.

jneilliii, 2020-09-10:
So with that latest dev version linked above I wonder if you just increase your automatic connect time to 30 seconds if it makes a difference or not here? Sorry haven't had a chance to really dig into this yet.

commit f48f75aa (2020-09-03): change auto start on upload logic, resolves #207
PR #220 (2020-11-12): 0.9.25
- fix typo in title for "Power Off on Idle" setting,
- fix power on with startup
- fix start print on upload logic, resolves #207
- add custom access permissions hook for controlling who can send API commands for controlling plugs
- add release channels available in OctoPrint 1.5.0
- add startup event monitoring to automatically power on during octoprint startup, resolves #204
- update knockout sortable module to version 1.2.0 to resolve knockout upgrade issue in OctoPrint 1.5.0
- update plotly to version 1.54.7 to resolve iPad 12.9 detection issues
closed 2020-11-12 by jneilliii