jneilliii/OctoPrint-TPLinkSmartplug #208: [Feature Request]: Octoprint connect to printer after power off

derekantrican, 2020-09-03:
**Setup:** I have an Ender 3 plugged into a TP-Link outlet and connected to a raspberry pi 4 running octoprint
**Workflow:** I plan out the build plate in Cura, then use the Octoprint plugin to send to my printer
**Problem:** When a print is finished printing the TPLink plugin turns off the printer (which is great! exactly what I want) but that also disconnects Octoprint from the printer. Octoprint can connect to the printer when it's powered off, so all I have to do is click "Connect" in Octoprint and I'm back up and running. But since this happens every print it gets annoying.

**Request:** Similar to the `@TPLINKOFF` command, could you also add something like a `@TPLINKCONNECT` command that would call the [Octoprint API method](https://docs.octoprint.org/en/master/api/connection.html#issue-a-connection-command) to connect to the printer? That save me a lot of pain!

jneilliii, 2020-09-03:
Sounds like you have back-powered printer control board over USB. This is why you are able to connect to printer even with printer off, and IMO doesn't make sense. You may want to look into this article about the situation.

https://community.octoprint.org/t/put-tape-on-the-5v-pin-why-and-how/13574

derekantrican, 2020-09-03:
It's not causing me any problems so I don't see a reason to do that. Even if I did do it, it still wouldn't solve my problem where I can't print directly from Cura because the printer is offline. How else would you suggest solving this?

jneilliii, 2020-09-03:
I personally use the OctoPrint Connection plugin in Cura and my printer is not back-powered by USB and it powers on the plug for me from the settings. It's possible that the on upload event monitoring will also power on without this option in OctoPrint 1.4.1+., I just don't know if the Cura plugin flags it the same way as Slic3r/PrusaSlicer/SuperSlicer do it when they upload the file with autostart option.

[image]

derekantrican, 2020-09-03:
Ah - perfect! That setting was exactly what I was looking for

For people who look at this in the future, the setting via the Cura plugin is: Settings > [Printer] > Manager Printers > Connect OctoPrint > Turn on printer with [TP Link device]

Thanks for the help!

closed 2020-09-03 by derekantrican

jneilliii, 2020-09-03:
The accolades for that one are all on @fieldOfView. He integrated all the power plugin options that are out there.