jneilliii/OctoPrint-TPLinkSmartplug #209: [BUG]: Printer turns off every time I start a new print

Tarzanelly, 2020-09-08:
Hello everyone. First of all sorry for my english. It's not perfect and I will try to do my best.
Like the title says, every time I launch a new print my printer (Creality Ender 3) power off after few seconds. 
The extruder goes up on the Z axis and while the temperature goes up it power off
This is my setup for the plugin: 
[image]
[image]

and these are my start and end gcodes:
**Start G-code:**
_M83;
G28;
G1 Z30.0;
M104 S[first_layer_temperature] ; set extruder temp
M140 S[first_layer_bed_temperature] ; set bed temp
M190 S[first_layer_bed_temperature] ; wait for bed temp
M109 S[first_layer_temperature] ; wait for extruder temp
G28;
G92 E0.0;
G1 Y100.0 E9.0 F1000.0;
G1 Y 200.0 E12.5 F1000.0;
G92 E0.0;
M80 192.168.1.89;_

**End G-code:** 
_G4;
M104 S0;
M140 S0;
M107;
G1 X0.0 Y200.0;
M84;
M81 192.168.1.189;_

I'm using "TP-Link Smartplug (0.9.24)" and a TP-Link HS100

I really can't understand why it happens. Probably there are some gcode issues... I don't know. I tried to launch a new print with an old gcode when I wasn't using this plugin and everything worked.

Tarzanelly, 2020-09-08:
I'm using Octoprint on a Raspberry Pi 3 A+
I manage it through Chrome on a windows 7 PC

LarryVand, 2020-09-08:
Change your "abort power off timeout" from 30 sec to 300 sec. Then when you start your printer, when the bug occurs, you will get a 300sec countdown warning prompt that you need to cancel. After canceling that warning prompt, everything should work fine, till the next time you shut off your printer longer than the idle timeout (which in your case is only 8 minutes). I suggest you set the idle timeout to 30min, not 8min, as 8min is quite short.

Tarzanelly, 2020-09-08:
I saw this in the other discussion, but I have no warning prompt to cancel and it just turns off. This is really strange

LarryVand, 2020-09-08:
See if you can copy the settings below but set the "abort power off timeout" to 300 secs". And when you turn on your TP-LINK HS-100, make sure within 90 sec that you get to a browser and load octoprint. That countdown warning prompt should appear on the upper right corner.
https://user-images.githubusercontent.com/69426512/89898374-73585400-db95-11ea-820e-8304b2741d03.JPG

jneilliii, 2020-09-08:
Looking at your settings I think you might have your max temps reversed. You have 80 for extruder (hot end) and 250 for the bed.

LarryVand, 2020-09-08:
I don't know what settings you are looking at (80? 250?) but the picture I send you has Max bed temp 125 and max extruder temp 270, which would be correct.
https://user-images.githubusercontent.com/69426512/89898374-73585400-db95-11ea-820e-8304b2741d03.JPG

Tarzanelly, 2020-09-08:
> I don't know what settings you are looking at (80? 250?) but the picture I send you has Max bed temp 125 and max extruder temp 270, which would be correct.
> https://user-images.githubusercontent.com/69426512/89898374-73585400-db95-11ea-820e-8304b2741d03.JPG

I think his comment was for me. It has sense... I'm an idiot! I will try tomorrow but I think this is my problem. I'll let you know

jneilliii, 2020-09-08:
Yeah, I was referring to @Tarzanelly screenshots from his original post. They are reversed.

LarryVand, 2020-09-09:
> Yeah, I was referring to @Tarzanelly screenshots from his original post. They are reversed.

Sorry jneillii. I totally missed that the post was from you and not Tarzanelly, :(

Tarzanelly, 2020-09-09:
Ok now it doesn't turn off anymore. But I noticed it doesn't turn off by itself when the print is finished. Did I miss something with the end gcode? Is the start g-code (M80) needed to make the end one working?

jneilliii, 2020-09-09:
Delays of 0 with warn while printing will cause the gcode commands to not power off your plug. Increase the off delay to allow the print to finish completely before attempting the power off.

LarryVand, 2020-09-27:
Just putting a post down to un-stale the issue... lol

jneilliii, 2020-09-27:
Are you still having issues @LarryVand per mmt last reply? I still haven't heard from @Tarzanelly and that's what I thought we'r were waiting on. The purpose of the stalebot is exactly that, if there's no response from my replies I don't want things lingering.

LarryVand, 2020-09-27:
@jneilliii Oh yes. The issue is still there. The "work around" is not a true "work around" that fixes the issue, but instead delays the bug long enough to manually catch it and abort it. The increase to the "Abort Power Off Timeout" #1 requires the attention of the user to catch it and abort it, and #2 that increase is added on to the power off on idle when you really want for the printer to shut off . So Yes. The bug is still there. It happens when the plug in (octoprint) has been off longer than the "Idle Timeout" minutes time, so when Octoprint restarts the plug in thinks immediately it has idle the timeout minutes and starts right away to count down the "Abort Power Off Timeout seconds" (which is clearly NOT the case as the user just turned on Octoprint). If you restart Octoprint in less time than the "idle timeout minutes" the bug does not happen. It's only when Octoprint has been off for longer time than the "idle timeout minutes" that the "Abort Power Off Timeout seconds" starts counting down immediately on a Octoprint restart.

LarryVand, 2020-09-27:
Darn it. Sorry. My fault. Disregard everything. I thought this was the original thread of the idle timeout bug. I apologize.

closed 2020-10-22 by stale[bot]