jneilliii/OctoPrint-TPLinkSmartplug #210: Issue: Intermittent Connection to an HS110

war1000, 2020-09-20:
Hi, so far I have been able to connect once. I got all the usage reports as well but it is no longer connecting. I have rebooted all devices multiple time.

[2020-09-20 23:04:30,118] DEBUG: ImmutableMultiDict([('checkStatus', u'192.168.1.139')])
[2020-09-20 23:04:30,120] DEBUG: Checking status of 192.168.1.139.
[2020-09-20 23:04:30,121] DEBUG: {'system': {'get_sysinfo': {}}}
[2020-09-20 23:04:30,121] DEBUG: IP 192.168.1.139 is valid.
[2020-09-20 23:04:30,122] DEBUG: Sending command {'system': {'get_sysinfo': {}}} to 192.168.1.139
[2020-09-20 23:04:33,207] DEBUG: Could not connect to 192.168.1.139.
[2020-09-20 23:04:33,209] DEBUG: 
[2020-09-20 23:04:33,210] DEBUG: {'emeter': {'err_code': True}, 'system': {'get_sysinfo': {'relay_state': 3}}}
[2020-09-20 23:04:51,247] DEBUG: ImmutableMultiDict([('checkStatus', u'192.168.1.139')])
[2020-09-20 23:04:51,248] DEBUG: Checking status of 192.168.1.139.
[2020-09-20 23:04:51,251] DEBUG: {'system': {'get_sysinfo': {}}}
[2020-09-20 23:04:51,252] DEBUG: IP 192.168.1.139 is valid.
[2020-09-20 23:04:51,252] DEBUG: Sending command {'system': {'get_sysinfo': {}}} to 192.168.1.139
[2020-09-20 23:04:54,403] DEBUG: Could not connect to 192.168.1.139.
[2020-09-20 23:04:54,404] DEBUG: 
[2020-09-20 23:04:54,405] DEBUG: {'emeter': {'err_code': True}, 'system': {'get_sysinfo': {'relay_state': 3}}}
[2020-09-20 23:04:55,986] DEBUG: {'energy_data': [(u'2020-09-20 21:28:23.098284', 1.79739, 113.200209, 2.138, 119.720856), (u'2020-09-20 20:21:55.115658', 1.92758, 120.287356, 2.01, 119.517085), (u'2020-09-20 18:06:57.661238', 0.135136, 7.359501, 1.779, 120.657265), (u'2020-09-20 14:26:25.116881', 1.685915, 108.12367, 1.441, 120.674061), (u'2020-09-19 21:09:06.977409', 1.722701, 110.221523, 1.145, 121.210792), (u'2020-09-19 20:48:27.421353', 1.742552, 110.467631, 1.106, 121.306424), (u'2020-09-19 20:47:55.294765', 1.765893, 112.344106, 1.105, 121.451613), (u'2020-09-19 20:26:41.812801', 1.820653, 116.159836, 1.065, 121.261003), (u'2020-09-19 20:24:32.303185', 1.759896, 112.31786, 1.061, 121.085927), (u'2020-09-19 20:06:09.490707', 1.745149, 110.455761, 1.026, 121.251867)]}
[2020-09-20 23:05:06,869] DEBUG: ImmutableMultiDict([('checkStatus', u'192.168.1.139')])
[2020-09-20 23:05:06,871] DEBUG: Checking status of 192.168.1.139.
[2020-09-20 23:05:06,872] DEBUG: {'system': {'get_sysinfo': {}}}
[2020-09-20 23:05:06,873] DEBUG: IP 192.168.1.139 is valid.
[2020-09-20 23:05:06,873] DEBUG: Sending command {'system': {'get_sysinfo': {}}} to 192.168.1.139
[2020-09-20 23:05:10,004] DEBUG: Could not connect to 192.168.1.139.
[2020-09-20 23:05:10,004] DEBUG: 
[2020-09-20 23:05:10,005] DEBUG: {'emeter': {'err_code': True}, 'system': {'get_sysinfo': {'relay_state': 3}}}
[2020-09-20 23:05:43,164] DEBUG: ImmutableMultiDict([('checkStatus', u'192.168.1.139')])
[2020-09-20 23:05:43,165] DEBUG: Checking status of 192.168.1.139.
[2020-09-20 23:05:43,166] DEBUG: {'system': {'get_sysinfo': {}}}
[2020-09-20 23:05:43,166] DEBUG: IP 192.168.1.139 is valid.
[2020-09-20 23:05:43,167] DEBUG: Sending command {'system': {'get_sysinfo': {}}} to 192.168.1.139
[2020-09-20 23:05:46,324] DEBUG: Could not connect to 192.168.1.139.
[2020-09-20 23:05:46,324] DEBUG: 
[2020-09-20 23:05:46,325] DEBUG: {'emeter': {'err_code': True}, 'system': {'get_sysinfo': {'relay_state': 3}}}
[2020-09-20 23:05:58,981] DEBUG: Automatic power off setting changed: False
[2020-09-20 23:06:02,053] DEBUG: Automatic power off setting changed: True
[2020-09-20 23:06:06,446] DEBUG: CE3PRO_case.gcode
[2020-09-20 23:06:06,447] DEBUG: Checking status of 192.168.1.139.
[2020-09-20 23:06:06,447] DEBUG: {'system': {'get_sysinfo': {}}}
[2020-09-20 23:06:06,447] DEBUG: IP 192.168.1.139 is valid.
[2020-09-20 23:06:06,448] DEBUG: Sending command {'system': {'get_sysinfo': {}}} to 192.168.1.139
[2020-09-20 23:06:09,603] DEBUG: Could not connect to 192.168.1.139.
[2020-09-20 23:06:09,604] DEBUG: 
[2020-09-20 23:06:09,605] DEBUG: {'emeter': {'err_code': True}, 'system': {'get_sysinfo': {'relay_state': 3}}}
[2020-09-20 23:06:09,605] DEBUG: 0.0
[2020-09-20 23:06:26,628] DEBUG: Print cancelled, reseting job_power to 0
[2020-09-20 23:07:08,651] DEBUG: ImmutableMultiDict([('checkStatus', u'192.168.1.139')])
[2020-09-20 23:07:08,652] DEBUG: Checking status of 192.168.1.139.
[2020-09-20 23:07:08,653] DEBUG: {'system': {'get_sysinfo': {}}}
[2020-09-20 23:07:08,653] DEBUG: IP 192.168.1.139 is valid.
[2020-09-20 23:07:08,654] DEBUG: Sending command {'system': {'get_sysinfo': {}}} to 192.168.1.139
[2020-09-20 23:07:11,763] DEBUG: Could not connect to 192.168.1.139.
[2020-09-20 23:07:11,764] DEBUG: 
[2020-09-20 23:07:11,765] DEBUG: {'emeter': {'err_code': True}, 'system': {'get_sysinfo': {'relay_state': 3}}}
[2020-09-20 23:07:24,817] DEBUG: ImmutableMultiDict([('checkStatus', u'192.168.1.139')])
[2020-09-20 23:07:24,818] DEBUG: Checking status of 192.168.1.139.
[2020-09-20 23:07:24,818] DEBUG: {'system': {'get_sysinfo': {}}}
[2020-09-20 23:07:24,819] DEBUG: IP 192.168.1.139 is valid.
[2020-09-20 23:07:24,819] DEBUG: Sending command {'system': {'get_sysinfo': {}}} to 192.168.1.139
[2020-09-20 23:07:27,923] DEBUG: Could not connect to 192.168.1.139.
[2020-09-20 23:07:27,924] DEBUG: 
[2020-09-20 23:07:27,925] DEBUG: {'emeter': {'err_code': True}, 'system': {'get_sysinfo': {'relay_state': 3}}}
[2020-09-20 23:07:34,983] DEBUG: ImmutableMultiDict([('checkStatus', u'192.168.1.139')])
[2020-09-20 23:07:34,984] DEBUG: Checking status of 192.168.1.139.
[2020-09-20 23:07:34,984] DEBUG: {'system': {'get_sysinfo': {}}}
[2020-09-20 23:07:34,985] DEBUG: IP 192.168.1.139 is valid.
[2020-09-20 23:07:34,985] DEBUG: Sending command {'system': {'get_sysinfo': {}}} to 192.168.1.139
[2020-09-20 23:07:38,084] DEBUG: Could not connect to 192.168.1.139.
[2020-09-20 23:07:38,084] DEBUG: 
[2020-09-20 23:07:38,085] DEBUG: {'emeter': {'err_code': True}, 'system': {'get_sysinfo': {'relay_state': 3}}}
[2020-09-20 23:08:41,315] DEBUG: ImmutableMultiDict([('checkStatus', u'192.168.1.139')])
[2020-09-20 23:08:41,316] DEBUG: Checking status of 192.168.1.139.
[2020-09-20 23:08:41,317] DEBUG: {'system': {'get_sysinfo': {}}}
[2020-09-20 23:08:41,317] DEBUG: IP 192.168.1.139 is valid.
[2020-09-20 23:08:41,318] DEBUG: Sending command {'system': {'get_sysinfo': {}}} to 192.168.1.139
[2020-09-20 23:08:44,404] DEBUG: Could not connect to 192.168.1.139.
[2020-09-20 23:08:44,404] DEBUG: 
[2020-09-20 23:08:44,405] DEBUG: {'emeter': {'err_code': True}, 'system': {'get_sysinfo': {'relay_state': 3}}}
[2020-09-20 23:10:03,672] DEBUG: abortTimeout: 30
[2020-09-20 23:10:03,685] DEBUG: powerOffWhenIdle: True
[2020-09-20 23:10:03,686] DEBUG: idleTimeout: 30
[2020-09-20 23:10:03,687] DEBUG: idleIgnoreCommands: M105
[2020-09-20 23:10:03,734] DEBUG: idleTimeoutWaitTemp: 50
[2020-09-20 23:10:34,828] DEBUG: ImmutableMultiDict([('checkStatus', u'192.168.1.139')])
[2020-09-20 23:10:34,829] DEBUG: Checking status of 192.168.1.139.
[2020-09-20 23:10:34,829] DEBUG: {'system': {'get_sysinfo': {}}}
[2020-09-20 23:10:34,830] DEBUG: IP 192.168.1.139 is valid.
[2020-09-20 23:10:34,830] DEBUG: Sending command {'system': {'get_sysinfo': {}}} to 192.168.1.139
[2020-09-20 23:10:37,954] DEBUG: Could not connect to 192.168.1.139.
[2020-09-20 23:10:37,958] DEBUG: 
[2020-09-20 23:10:37,958] DEBUG: {'emeter': {'err_code': True}, 'system': {'get_sysinfo': {'relay_state': 3}}}
[2020-09-20 23:10:38,805] DEBUG: {'energy_data': [(u'2020-09-20 21:28:23.098284', 1.79739, 113.200209, 2.138, 119.720856), (u'2020-09-20 20:21:55.115658', 1.92758, 120.287356, 2.01, 119.517085), (u'2020-09-20 18:06:57.661238', 0.135136, 7.359501, 1.779, 120.657265), (u'2020-09-20 14:26:25.116881', 1.685915, 108.12367, 1.441, 120.674061), (u'2020-09-19 21:09:06.977409', 1.722701, 110.221523, 1.145, 121.210792), (u'2020-09-19 20:48:27.421353', 1.742552, 110.467631, 1.106, 121.306424), (u'2020-09-19 20:47:55.294765', 1.765893, 112.344106, 1.105, 121.451613), (u'2020-09-19 20:26:41.812801', 1.820653, 116.159836, 1.065, 121.261003), (u'2020-09-19 20:24:32.303185', 1.759896, 112.31786, 1.061, 121.085927), (u'2020-09-19 20:06:09.490707', 1.745149, 110.455761, 1.026, 121.251867)]}

jneilliii, 2020-09-20:
Did you connect the device to the cloud? There have been reports that this will break communication from this plugin in newer versions of the firmware for some devices.

war1000, 2020-09-20:
yes, this one is also connected to the cloud. Should I remove it from the Kasa app?

jneilliii, 2020-09-20:
Check this link. You can run in local only mode and it might work. 

https://www.tp-link.com/us/support/faq/2707/

war1000, 2020-09-21:
Thanks for the link. I just did this. Still no success. I am just baffled that it worked and now it isn't.

[2020-09-20 20:12:02,112] DEBUG: abortTimeout: 30
[2020-09-20 20:12:02,113] DEBUG: powerOffWhenIdle: True
[2020-09-20 20:12:02,127] DEBUG: idleTimeout: 30
[2020-09-20 20:12:02,128] DEBUG: idleIgnoreCommands: M105
[2020-09-20 20:12:02,130] DEBUG: idleTimeoutWaitTemp: 50
[2020-09-20 20:12:13,604] DEBUG: ImmutableMultiDict([('checkStatus', u'192.168.1.139')])
[2020-09-20 20:12:13,612] DEBUG: Checking status of 192.168.1.139.
[2020-09-20 20:12:13,614] DEBUG: {'system': {'get_sysinfo': {}}}
[2020-09-20 20:12:13,618] DEBUG: IP 192.168.1.139 is valid.
[2020-09-20 20:12:13,620] DEBUG: Sending command {'system': {'get_sysinfo': {}}} to 192.168.1.139
[2020-09-20 20:12:16,702] DEBUG: Could not connect to 192.168.1.139.
[2020-09-20 20:12:16,703] DEBUG: 
[2020-09-20 20:12:16,703] DEBUG: {'emeter': {'err_code': True}, 'system': {'get_sysinfo': {'relay_state': 3}}}
[2020-09-20 20:12:36,521] DEBUG: powering off due to Error event.
[2020-09-20 20:12:36,524] DEBUG: powering off 192.168.1.139 due to Error event.
[2020-09-20 20:12:36,525] DEBUG: Turning off 192.168.1.139.
[2020-09-20 20:12:36,526] DEBUG: {'autoConnect': True, 'ip': '192.168.1.139', 'btnColor': '#808080', 'autoDisconnectDelay': 0, 'sysCmdOff': False, 'gcodeEnabled': True, 'gcodeOnDelay': 0, 'emeter': {'get_realtime': {'current': 1.742552, 'total': 1.106, 'voltage': 121.306424, 'power': 110.467631, 'err_code': 0}}, 'countdownOnDelay': 1, 'label': 'HS110-KASA-Plug1', 'warnPrinting': True, 'thermal_runaway': True, 'gcodeOffDelay': 0, 'sysCmdOnDelay': 0, 'event_on_error': True, 'countdownOffDelay': 1, 'sysRunCmdOff': '', 'sysRunCmdOn': '', 'autoDisconnect': True, 'automaticShutdownEnabled': True, 'event_on_upload': True, 'useCountdownRules': True, 'icon': 'icon-bolt', 'displayWarning': True, 'autoConnectDelay': 10, 'event_on_disconnect': False, 'sysCmdOn': False, 'currentState': 'unknown', 'sysCmdOffDelay': 0}
[2020-09-20 20:12:36,528] DEBUG: IP 192.168.1.139 is valid.
[2020-09-20 20:12:36,529] DEBUG: Sending command {u'count_down': {u'delete_all_rules': None}} to 192.168.1.139
[2020-09-20 20:12:39,638] DEBUG: Could not connect to 192.168.1.139.
[2020-09-20 20:12:39,639] DEBUG: IP 192.168.1.139 is valid.
[2020-09-20 20:12:39,639] DEBUG: Sending command {u'count_down': {u'add_rule': {u'delay': 1, u'enable': 1, u'name': u'turn off', u'act': 0}}} to 192.168.1.139
[2020-09-20 20:12:42,758] DEBUG: Could not connect to 192.168.1.139.
[2020-09-20 20:12:42,759] DEBUG: None

jneilliii, 2020-09-21:
Did the plug's ip change maybe?

war1000, 2020-09-21:
No...I tried with another plug, removed it from the cloud and did local...same issue...

[2020-09-20 21:48:57,676] DEBUG: {'emeter': {'err_code': True}, 'system': {'get_sysinfo': {'relay_state': 3}}}
[2020-09-20 21:49:42,055] DEBUG: ImmutableMultiDict([('checkStatus', u'192.168.1.189')])
[2020-09-20 21:49:42,056] DEBUG: Checking status of 192.168.1.189.
[2020-09-20 21:49:42,056] DEBUG: {'system': {'get_sysinfo': {}}}
[2020-09-20 21:49:42,056] DEBUG: IP 192.168.1.189 is valid.
[2020-09-20 21:49:42,056] DEBUG: Sending command {'system': {'get_sysinfo': {}}} to 192.168.1.189
[2020-09-20 21:49:45,195] DEBUG: Could not connect to 192.168.1.189.
[2020-09-20 21:49:45,195] DEBUG: 
[2020-09-20 21:49:45,197] DEBUG: {'emeter': {'err_code': True}, 'system': {'get_sysinfo': {'relay_state': 3}}}
[2020-09-20 21:49:45,360] DEBUG: Settings saved, Automatic Power Off Enabled, starting idle timer...
[2020-09-20 21:49:51,177] DEBUG: ImmutableMultiDict([('checkStatus', u'192.168.1.189')])
[2020-09-20 21:49:51,178] DEBUG: Checking status of 192.168.1.189.
[2020-09-20 21:49:51,178] DEBUG: {'system': {'get_sysinfo': {}}}
[2020-09-20 21:49:51,179] DEBUG: IP 192.168.1.189 is valid.
[2020-09-20 21:49:51,179] DEBUG: Sending command {'system': {'get_sysinfo': {}}} to 192.168.1.189
[2020-09-20 21:49:54,315] DEBUG: Could not connect to 192.168.1.189.
[2020-09-20 21:49:54,316] DEBUG: 
[2020-09-20 21:49:54,317] DEBUG: {'emeter': {'err_code': True}, 'system': {'get_sysinfo': {'relay_state': 3}}}

jneilliii, 2020-09-21:
Yeah, I'm afraid you may have been hit with the new firmware bug. It's not just this plugin, but other projects that also use the same reverse engineered hack to control these devices. Not sure I'll ever be able to fix this unless that firmware communication is reverse engineered again, if that's even if possible. This is the reason I'm saying that.

```
[2020-09-20 21:49:51,179] DEBUG: Sending command {'system': {'get_sysinfo': {}}} to 192.168.1.189
[2020-09-20 21:49:54,315] DEBUG: Could not connect to 192.168.1.189.
```

The device itself is not responding to the connection request.

closed 2020-10-17 by stale[bot]