jneilliii/OctoPrint-TPLinkSmartplug #211: [BUG]: Automatic Triggers not activating

bjsmyth, 2020-09-29:
The plugin isn't doing the automatic triggers like turning on the printer when uploading a new file, or even things like the automatic timeout. This is a HS103 that I just bought new and connected locally, but may have the new firmware. However, the top bar button to toggle the outlet on and off seems to work just fine, it's the automatic switches that aren't working for me. So I can use the plugin to manually turn the outlet on and off with the octoprint webpage, but not with the automatic ways.

[Here's the config I have for the switch](https://i.imgur.com/GJPernx.png)

[And here's the global config for the plugin](https://i.imgur.com/uYiEOyL.png)

And here's a copy of my debug log for the plugin:

[2020-09-28 20:03:54,792] DEBUG: Turning on 10.0.0.4.
[2020-09-28 20:03:54,796] DEBUG: {'autoConnect': True, 'ip': '10.0.0.4', 'btnColor': '#808080'                                                                                                                                               , 'autoDisconnectDelay': 0, 'sysCmdOff': False, 'gcodeEnabled': False, 'gcodeOnDelay': 0, 'eme                                                                                                                                               ter': {'get_realtime': {}}, 'countdownOnDelay': 1, 'label': 'Ender3', 'warnPrinting': True, 't                                                                                                                                               hermal_runaway': True, 'gcodeOffDelay': 0, 'sysCmdOnDelay': 0, 'event_on_error': True, 'countd                                                                                                                                               ownOffDelay': 1, 'sysRunCmdOff': '', 'sysRunCmdOn': '', 'autoDisconnect': True, 'automaticShut                                                                                                                                               downEnabled': True, 'event_on_upload': True, 'useCountdownRules': False, 'icon': 'icon-bolt',                                                                                                                                                'displayWarning': True, 'autoConnectDelay': 10, 'event_on_disconnect': False, 'sysCmdOn': Fals                                                                                                                                               e, 'currentState': 'unknown', 'sysCmdOffDelay': 0}
[2020-09-28 20:03:54,798] DEBUG: IP 10.0.0.4 is valid.
[2020-09-28 20:03:54,799] DEBUG: Sending command {'system': {'set_relay_state': {'state': 1}}}                                                                                                                                                to 10.0.0.4
[2020-09-28 20:03:54,907] DEBUG: «-ý"system":{"set_relay_state":{"err_code":0}}}
[2020-09-28 20:03:54,922] DEBUG: 0
[2020-09-28 20:03:54,925] DEBUG: Resetting idle timer since plug 10.0.0.4 was just turned on.
[2020-09-28 20:03:54,926] DEBUG: Checking status of 10.0.0.4.
[2020-09-28 20:03:54,928] DEBUG: {'system': {'get_sysinfo': {}}}
[2020-09-28 20:03:54,929] DEBUG: IP 10.0.0.4 is valid.
[2020-09-28 20:03:54,930] DEBUG: Sending command {'system': {'get_sysinfo': {}}} to 10.0.0.4
[2020-09-28 20:03:55,023] DEBUG: «5ç"system":{"get_sysinfo":{"sw_ver":"1.1.2 Build 191113 Rel.                                                                                                                                               095623","hw_ver":"2.1","model":"HS103(US)","deviceId":"800661DA1325720147134968618592CC1D2EC8E                                                                                                                                               5","oemId":"24B27E003E98D053CF8B51CE8086FDA8","hwId":"18967AEED1C89BED1BBC7F62FD06468A","rssi"                                                                                                                                               :-30,"longitude_i":-832619,"latitude_i":423194,"alias":"Ender 3","status":"new","mic_type":"IO                                                                                                                                               T.SMARTPLUGSWITCH","feature":"TIM","mac":"D8:07:B6:53:9C:CF","updating":0,"led_off":0,"relay_s                                                                                                                                               tate":1,"on_time":0,"active_mode":"none","icon_hash":"","dev_name":"Smart Wi-Fi Plug Lite","ne                                                                                                                                               xt_action":{"type":-1},"err_code":0}}}
[2020-09-28 20:03:55,093] DEBUG: TIM
[2020-09-28 20:06:49,718] DEBUG: ImmutableMultiDict([('checkStatus', u'10.0.0.4')])
[2020-09-28 20:06:49,720] DEBUG: Checking status of 10.0.0.4.
[2020-09-28 20:06:49,721] DEBUG: {'system': {'get_sysinfo': {}}}
[2020-09-28 20:06:49,722] DEBUG: IP 10.0.0.4 is valid.
[2020-09-28 20:06:49,723] DEBUG: Sending command {'system': {'get_sysinfo': {}}} to 10.0.0.4
[2020-09-28 20:06:49,853] DEBUG: «;é"system":{"get_sysinfo":{"sw_ver":"1.1.2 Build 191113 Rel.                                                                                                                                               095623","hw_ver":"2.1","model":"HS103(US)","deviceId":"800661DA1325720147134968618592CC1D2EC8E                                                                                                                                               5","oemId":"24B27E003E98D053CF8B51CE8086FDA8","hwId":"18967AEED1C89BED1BBC7F62FD06468A","rssi"                                                                                                                                               :-30,"longitude_i":-832619,"latitude_i":423194,"alias":"Ender 3","status":"new","mic_type":"IO                                                                                                                                               T.SMARTPLUGSWITCH","feature":"TIM","mac":"D8:07:B6:53:9C:CF","updating":0,"led_off":0,"relay_s                                                                                                                                               tate":1,"on_time":175,"active_mode":"none","icon_hash":"","dev_name":"Smart Wi-Fi Plug Lite","                                                                                                                                               next_action":{"type":-1},"err_code":0}}}
[2020-09-28 20:06:49,919] DEBUG: TIM
[2020-09-28 20:06:49,959] DEBUG: Settings saved, Automatic Power Off Enabled, starting idle ti                                                                                                                                               mer...

jneilliii, 2020-09-29:
Can you try increasing your idle timeout value to something a little longer like 5 minutes to see if there's something weird going on with just 1 minute? I haven't ever tried a timeout that low before and am just curious. As for the on with upload option, I too have recently come across the issue reported in #207 since using SuperSlicer, but it sounds like your problem is the switch isn't even turning on. Your log indicates the plugin is still able to communicate with the plug, so I don't think it's a firmware related issue.

bjsmyth, 2020-09-29:
I was able to get it to trigger after 1 minute if I leave the octoprint page open in a browser.

I just tried printing something, closing the webpage after the print started with a 5 minute timeout. I waited 15 minutes after the print completed with nothing happened. I open the octoprint webpage and even then it didn't trigger after sitting for 5 additional minutes.

[2020-09-28 21:19:51,587] DEBUG: Checking statuses during print progress (94).
[2020-09-28 21:19:51,589] DEBUG: Resetting idle timer during print progress (94)...
[2020-09-28 21:19:52,592] DEBUG: Checking status of 10.0.0.4.
[2020-09-28 21:19:52,593] DEBUG: {'system': {'get_sysinfo': {}}}
[2020-09-28 21:19:52,594] DEBUG: IP 10.0.0.4 is valid.
[2020-09-28 21:19:52,595] DEBUG: Sending command {'system': {'get_sysinfo': {}}} to 10.0.0.4
[2020-09-28 21:19:52,730] DEBUG: «8ê"system":{"get_sysinfo":{"sw_ver":"1.1.2 Build 191113 Rel.095623","hw_ver":"2.1","model":"HS103(US)","deviceId":"800661DA1325720147134968618592CC1D2EC8E5","oemId":"24B27E003E98D053CF8B51CE8086FDA8","hwId":"18967AEED1C89BED1BBC7F62FD06468A","rssi":-31,"longitude_i":-832619,"latitude_i":423194,"alias":"Ender 3","status":"new","mic_type":"IOT.SMARTPLUGSWITCH","feature":"TIM","mac":"D8:07:B6:53:9C:CF","updating":0,"led_off":0,"relay_state":1,"on_time":1943,"active_mode":"none","icon_hash":"","dev_name":"Smart Wi-Fi Plug Lite","next_action":{"type":-1},"err_code":0}}}
[2020-09-28 21:19:52,816] DEBUG: TIM
[2020-09-28 21:19:52,849] DEBUG: Checking statuses during print progress (100).
[2020-09-28 21:19:52,855] DEBUG: Resetting idle timer during print progress (100)...
[2020-09-28 21:19:53,862] DEBUG: Checking status of 10.0.0.4.
[2020-09-28 21:19:53,864] DEBUG: {'system': {'get_sysinfo': {}}}
[2020-09-28 21:19:53,865] DEBUG: IP 10.0.0.4 is valid.
[2020-09-28 21:19:53,866] DEBUG: Sending command {'system': {'get_sysinfo': {}}} to 10.0.0.4
[2020-09-28 21:19:53,993] DEBUG: «8ê"system":{"get_sysinfo":{"sw_ver":"1.1.2 Build 191113 Rel.095623","hw_ver":"2.1","model":"HS103(US)","deviceId":"800661DA1325720147134968618592CC1D2EC8E5","oemId":"24B27E003E98D053CF8B51CE8086FDA8","hwId":"18967AEED1C89BED1BBC7F62FD06468A","rssi":-30,"longitude_i":-832619,"latitude_i":423194,"alias":"Ender 3","status":"new","mic_type":"IOT.SMARTPLUGSWITCH","feature":"TIM","mac":"D8:07:B6:53:9C:CF","updating":0,"led_off":0,"relay_state":1,"on_time":1944,"active_mode":"none","icon_hash":"","dev_name":"Smart Wi-Fi Plug Lite","next_action":{"type":-1},"err_code":0}}}
[2020-09-28 21:19:54,059] DEBUG: TIM
[2020-09-28 21:24:40,977] DEBUG: Checking status of 10.0.0.4.
[2020-09-28 21:24:40,978] DEBUG: {'system': {'get_sysinfo': {}}}
[2020-09-28 21:24:40,978] DEBUG: IP 10.0.0.4 is valid.
[2020-09-28 21:24:40,979] DEBUG: Sending command {'system': {'get_sysinfo': {}}} to 10.0.0.4
[2020-09-28 21:24:41,083] DEBUG: «8ê"system":{"get_sysinfo":{"sw_ver":"1.1.2 Build 191113 Rel.095623","hw_ver":"2.1","model":"HS103(US)","deviceId":"800661DA1325720147134968618592CC1D2EC8E5","oemId":"24B27E003E98D053CF8B51CE8086FDA8","hwId":"18967AEED1C89BED1BBC7F62FD06468A","rssi":-31,"longitude_i":-832619,"latitude_i":423194,"alias":"Ender 3","status":"new","mic_type":"IOT.SMARTPLUGSWITCH","feature":"TIM","mac":"D8:07:B6:53:9C:CF","updating":0,"led_off":0,"relay_state":1,"on_time":2231,"active_mode":"none","icon_hash":"","dev_name":"Smart Wi-Fi Plug Lite","next_action":{"type":-1},"err_code":0}}}
[2020-09-28 21:24:41,152] DEBUG: TIM
[2020-09-28 21:32:57,576] DEBUG: ImmutableMultiDict([('checkStatus', u'10.0.0.4')])
[2020-09-28 21:32:57,577] DEBUG: Checking status of 10.0.0.4.
[2020-09-28 21:32:57,578] DEBUG: {'system': {'get_sysinfo': {}}}
[2020-09-28 21:32:57,578] DEBUG: IP 10.0.0.4 is valid.
[2020-09-28 21:32:57,579] DEBUG: Sending command {'system': {'get_sysinfo': {}}} to 10.0.0.4
[2020-09-28 21:32:57,767] DEBUG: «8ê"system":{"get_sysinfo":{"sw_ver":"1.1.2 Build 191113 Rel.095623","hw_ver":"2.1","model":"HS103(US)","deviceId":"800661DA1325720147134968618592CC1D2EC8E5","oemId":"24B27E003E98D053CF8B51CE8086FDA8","hwId":"18967AEED1C89BED1BBC7F62FD06468A","rssi":-31,"longitude_i":-832619,"latitude_i":423194,"alias":"Ender 3","status":"new","mic_type":"IOT.SMARTPLUGSWITCH","feature":"TIM","mac":"D8:07:B6:53:9C:CF","updating":0,"led_off":0,"relay_state":1,"on_time":2728,"active_mode":"none","icon_hash":"","dev_name":"Smart Wi-Fi Plug Lite","next_action":{"type":-1},"err_code":0}}}
[2020-09-28 21:32:57,834] DEBUG: TIM

I also looked in octoprint.log and it looks like there's some sort of call to the plugin about an upload but it's not doing anything with it.

2020-09-28 20:44:01,695 - octoprint.plugins.tplinksmartplug - INFO - Upload: {'print': False, 'path': u'CE3PRO_3DBenchy_PETG.gcode', 'target': u'local', 'select': False, 'name': u'CE3PRO_3DBenchy_PETG.gcode'}
2020-09-28 20:44:01,606 - octoprint.plugins.tplinksmartplug - INFO - FileAdded: {'path': u'CE3PRO_3DBenchy_PETG.gcode', 'storage': u'local', 'name': u'CE3PRO_3DBenchy_PETG.gcode', 'type': ['machinecode', 'gcode']}

jneilliii, 2020-09-29:
Ok, the on upload feature actually only works when uploading a file from PrusaSlicer/SuperSlicer/Slic3r directly with the option to automatically start the print when uploaded. It doesn't trigger on a pure upload event from the web interface. 

> I was able to get it to trigger after 1 minute if I leave the octoprint page open in a browser.

All of the timeout logic is programmed into the server side code, so this shouldn't have an impact one way or another outside of the abort dialog that pops up in the web interface to allow you to cancel the power off. What I find weird is the timeout never seems to be reached, it only gets reset. Have you tried restarting OctoPrint since setting the timeout value? I can only think that there is some logic I'm missing or that is getting bypassed because of current status not being in sync with the plugin for some reason.

bjsmyth, 2020-09-29:
> It doesn't trigger on a pure upload event from the web interface.

Ah, ok. Is that a feature that could be implemented in the future, or not possible with OctoPrint's API? I've been using Cura, maybe I should give Slic3r a try lol

I gave it a try after restarting OctoPrint and that seemed to get it to work. Gave it a couple more tries and looks like it does turn the switch off at the appropriate timing after OctoPrint is restarted.

jneilliii, 2020-09-29:
Oh shoot, if you are using Cura, then you should install the OctoPrint Connection plugin in the marketplace and send files directly from it. It will see the presence of this plugin as well as the Cura Thumbnails plugin and automatically send the file, with a thumbnail directly to octoprint and auto power on your printer for you. That is my standard workflow.

jneilliii, 2020-09-29:
This one...

https://marketplace.ultimaker.com/app/cura/plugins/fieldofview/OctoPrintPlugin

jneilliii, 2020-09-29:
And then you set the connection up in the printer settings.

[image]

bjsmyth, 2020-09-29:
Wow, I didn't even know that was a thing! That makes things so much easier to send prints down to my printer! Thanks for the help, I really appreciate it!

closed 2020-11-14 by jneilliii