jneilliii/OctoPrint-TPLinkSmartplug #212: Raspberry pi 4 8gb

2-k, 2020-10-05:
I can not get tp link plug in to work with hs100.

I have connected them on same router, same 2.4ghz, I can see them both on the ROG AX11000.
 So IP is correct, connection is correct.

For sure I tried the different firmwares. 1.5.2, 1.5.3 and 1.6. Set up plug off the cloud so anything on local only can control it. 
Tried setting up plug then leaving it on search mode. 

Octopi is 18.1, updated raspberry pi firmware, software and plugins.

Only problem I've had is getting the 12.3mp camera to work, but I got the 5mp to work easily so I moved to the smart plug.

I have been working on this for a while so not sure what to try now.

jneilliii, 2020-10-05:
Enable debug logging, restart octoprint and try again to check the status of your plug or control it. Once that fails grab the file plugin_tplinksmartplug_debug.log from the logging section of octoprint's settings and drag it into a comment here to upload for further diagnostics.

2-k, 2020-10-05:
Ok, will do once I get home from work. Thanks for the fast response.

2-k, 2020-10-05:
[2020-10-04 17:29:10,630] DEBUG: Automatic power off setting changed: False
[2020-10-04 17:29:13,750] DEBUG: Automatic power off setting changed: True
[2020-10-04 17:29:40,842] DEBUG: Automatic power off setting changed: False
[2020-10-04 17:29:42,196] DEBUG: ImmutableMultiDict([('checkStatus', '192.168.50.227')])
[2020-10-04 17:29:42,196] DEBUG: Checking status of 192.168.50.227.
[2020-10-04 17:29:42,197] DEBUG: {'system': {'get_sysinfo': {}}}
[2020-10-04 17:29:42,197] DEBUG: IP 192.168.50.227 is valid.
[2020-10-04 17:29:42,197] DEBUG: Sending command {'system': {'get_sysinfo': {}}} to 192.168.50.227
[2020-10-04 17:29:42,405] DEBUG: Â«�LÂž"system":{"get_sysinfo":{"sw_ver":"1.5.6 Build 191118 Rel.140307","hw_ver":"2.0","type":"IOT.SMARTPLUGSWITCH","model":"HS100(US)","mac":"98:DA:C4:47:96:67","dev_name":"Smart Wi-Fi Plug","alias":"Predator 3d","relay_state":0,"on_time":0,"active_mode":"none","feature":"TIM","updating":0,"icon_hash":"","rssi":-49,"led_off":0,"longitude_i":-1209709,"latitude_i":375871,"hwId":"0DC28CDD0B7E6C55F52AD35B8B68277E","fwId":"00000000000000000000000000000000","deviceId":"800680EDA7CAB196AFE774C35761D8A21B9D242A","oemId":"80CB8C7ED4CAD7DA4570979BFF731BA1","next_action":{"type":-1},"err_code":0}}}
[2020-10-04 17:29:42,407] DEBUG: TIM
[2020-10-04 17:29:44,738] DEBUG: ImmutableMultiDict([('checkStatus', '192.168.50.227')])
[2020-10-04 17:29:44,739] DEBUG: Checking status of 192.168.50.227.
[2020-10-04 17:29:44,739] DEBUG: {'system': {'get_sysinfo': {}}}
[2020-10-04 17:29:44,739] DEBUG: IP 192.168.50.227 is valid.
[2020-10-04 17:29:44,740] DEBUG: Sending command {'system': {'get_sysinfo': {}}} to 192.168.50.227
[2020-10-04 17:29:44,895] DEBUG: Â«�LÂž"system":{"get_sysinfo":{"sw_ver":"1.5.6 Build 191118 Rel.140307","hw_ver":"2.0","type":"IOT.SMARTPLUGSWITCH","model":"HS100(US)","mac":"98:DA:C4:47:96:67","dev_name":"Smart Wi-Fi Plug","alias":"Predator 3d","relay_state":0,"on_time":0,"active_mode":"none","feature":"TIM","updating":0,"icon_hash":"","rssi":-49,"led_off":0,"longitude_i":-1209709,"latitude_i":375871,"hwId":"0DC28CDD0B7E6C55F52AD35B8B68277E","fwId":"00000000000000000000000000000000","deviceId":"800680EDA7CAB196AFE774C35761D8A21B9D242A","oemId":"80CB8C7ED4CAD7DA4570979BFF731BA1","next_action":{"type":-1},"err_code":0}}}
[2020-10-04 17:29:44,897] DEBUG: TIM
[2020-10-04 17:29:45,154] DEBUG: ImmutableMultiDict([('checkStatus', '192.168.50.227')])
[2020-10-04 17:29:45,154] DEBUG: Checking status of 192.168.50.227.
[2020-10-04 17:29:45,155] DEBUG: {'system': {'get_sysinfo': {}}}
[2020-10-04 17:29:45,155] DEBUG: IP 192.168.50.227 is valid.
[2020-10-04 17:29:45,155] DEBUG: Sending command {'system': {'get_sysinfo': {}}} to 192.168.50.227
[2020-10-04 17:29:45,308] DEBUG: Â«�LÂž"system":{"get_sysinfo":{"sw_ver":"1.5.6 Build 191118 Rel.140307","hw_ver":"2.0","type":"IOT.SMARTPLUGSWITCH","model":"HS100(US)","mac":"98:DA:C4:47:96:67","dev_name":"Smart Wi-Fi Plug","alias":"Predator 3d","relay_state":0,"on_time":0,"active_mode":"none","feature":"TIM","updating":0,"icon_hash":"","rssi":-49,"led_off":0,"longitude_i":-1209709,"latitude_i":375871,"hwId":"0DC28CDD0B7E6C55F52AD35B8B68277E","fwId":"00000000000000000000000000000000","deviceId":"800680EDA7CAB196AFE774C35761D8A21B9D242A","oemId":"80CB8C7ED4CAD7DA4570979BFF731BA1","next_action":{"type":-1},"err_code":0}}}
[2020-10-04 17:29:45,310] DEBUG: TIM
[2020-10-04 17:29:45,423] DEBUG: ImmutableMultiDict([('checkStatus', '192.168.50.227')])
[2020-10-04 17:29:45,423] DEBUG: Checking status of 192.168.50.227.
[2020-10-04 17:29:45,424] DEBUG: {'system': {'get_sysinfo': {}}}
[2020-10-04 17:29:45,424] DEBUG: IP 192.168.50.227 is valid.
[2020-10-04 17:29:45,424] DEBUG: Sending command {'system': {'get_sysinfo': {}}} to 192.168.50.227
[2020-10-04 17:29:45,455] DEBUG: Â«�LÂž"system":{"get_sysinfo":{"sw_ver":"1.5.6 Build 191118 Rel.140307","hw_ver":"2.0","type":"IOT.SMARTPLUGSWITCH","model":"HS100(US)","mac":"98:DA:C4:47:96:67","dev_name":"Smart Wi-Fi Plug","alias":"Predator 3d","relay_state":0,"on_time":0,"active_mode":"none","feature":"TIM","updating":0,"icon_hash":"","rssi":-49,"led_off":0,"longitude_i":-1209709,"latitude_i":375871,"hwId":"0DC28CDD0B7E6C55F52AD35B8B68277E","fwId":"00000000000000000000000000000000","deviceId":"800680EDA7CAB196AFE774C35761D8A21B9D242A","oemId":"80CB8C7ED4CAD7DA4570979BFF731BA1","next_action":{"type":-1},"err_code":0}}}
[2020-10-04 17:29:45,456] DEBUG: TIM
[2020-10-04 17:29:45,866] DEBUG: ImmutableMultiDict([('checkStatus', '192.168.50.227')])
[2020-10-04 17:29:45,867] DEBUG: Checking status of 192.168.50.227.
[2020-10-04 17:29:45,867] DEBUG: {'system': {'get_sysinfo': {}}}
[2020-10-04 17:29:45,867] DEBUG: IP 192.168.50.227 is valid.
[2020-10-04 17:29:45,868] DEBUG: Sending command {'system': {'get_sysinfo': {}}} to 192.168.50.227
[2020-10-04 17:29:46,041] DEBUG: Â«�LÂž"system":{"get_sysinfo":{"sw_ver":"1.5.6 Build 191118 Rel.140307","hw_ver":"2.0","type":"IOT.SMARTPLUGSWITCH","model":"HS100(US)","mac":"98:DA:C4:47:96:67","dev_name":"Smart Wi-Fi Plug","alias":"Predator 3d","relay_state":0,"on_time":0,"active_mode":"none","feature":"TIM","updating":0,"icon_hash":"","rssi":-49,"led_off":0,"longitude_i":-1209709,"latitude_i":375871,"hwId":"0DC28CDD0B7E6C55F52AD35B8B68277E","fwId":"00000000000000000000000000000000","deviceId":"800680EDA7CAB196AFE774C35761D8A21B9D242A","oemId":"80CB8C7ED4CAD7DA4570979BFF731BA1","next_action":{"type":-1},"err_code":0}}}
[2020-10-04 17:29:46,044] DEBUG: TIM
[... cut ...]
[2020-10-04 23:12:13,372] DEBUG: Checking status of 102.168.50.11.
[2020-10-04 23:12:13,372] DEBUG: {'system': {'get_sysinfo': {}}}
[2020-10-04 23:12:13,372] DEBUG: IP 102.168.50.11 is valid.
[2020-10-04 23:12:13,372] DEBUG: Sending command {'system': {'get_sysinfo': {}}} to 102.168.50.11
[2020-10-04 23:12:23,383] DEBUG: Could not connect to 102.168.50.11.
[2020-10-04 23:12:23,384] DEBUG: 
[2020-10-04 23:12:23,385] DEBUG: {'system': {'get_sysinfo': {'relay_state': 3}}, 'emeter': {'err_code': True}}

2-k, 2020-10-05:
I noticed that I put the IP address wrong this time, 102 instead of 192. Will fix and try again. 50.11 is firmware never update. 50.227 has had firmware update then reset.

2-k, 2020-10-05:
Lol it just started to work for me. Awesome guess I just need to put in the wrong IP address then Correct it. I don't see what else I did different this time other than make the IP address mistake.

2-k, 2020-10-05:
Thank you have a good one.

closed 2020-10-05 by 2-k