jneilliii/OctoPrint-TPLinkSmartplug #214: KP303 support

BigTed76, 2020-10-18:
The TP Link smart power strips (3 plug KP303) don't seem to work with this plugin.  This plugin only appears to work with the single socket smart plugs.  It would be cool if power strip support with individual socket switching could be added.

jneilliii, 2020-10-18:
They are. I use the 6 outlet model strip. You just have to pay attention to the hover text in how to enter the ip, ie 192.168.0.2/0.

BigTed76, 2020-10-19:
Thanks for that - yep got it.  Mine has plugs labelled 1 to 3 but I have to use ip xxx.xxx.xxx.27/0 to ip xxx.xxx.xxx.27/2 I.e plug number -1.  Not a problem just thought I’d share.

This is a very good plugin and extremely useful - top marks for developing it.

Many thanks for you speedy response.

closed 2020-10-19 by BigTed76