jneilliii/OctoPrint-TPLinkSmartplug #215: [BUG]: No action after sending off command

thelewiss1, 2020-10-27:
**Describe the bug**
After clicking the power button in the top bar, and pressing "yes" to turn off, the green power icon just spins continually and the plug does not turn off. I waited quite a while, so long that even the Auto Power Off triggered, which also did not work. Printer not connected.

 - OS: Windows 10
 - Browser Edge 86.0.622.43
 - OctoPrint 1.4.2
 - Plugin Version 0.9.24
 - Kasa HS100

jneilliii, 2020-10-28:
Could you please enable debug logging, restart octoprint, and then share your logs after trying the process again?

thelewiss1, 2020-10-28:
Hi jneilliii, here's my log. I've censored my individual IDs and long/lat. let me know if you needed any of these.
[censored_log.txt](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/files/5454093/censored_log.txt)

jneilliii, 2020-10-28:
I think it might be your countdown timer settings. Can you see if increasing those to something larger will actually resolve this or possibly without that option enabled as a test?

thelewiss1, 2020-10-28:
I did have them disabled, but I enabled them as a troubleshooting step in
case that would fix it.

Increasing the delay to 5 seconds did seem to help, but when turning off
the plug from the toolbar, it turns the switch off, but the icon continues
to spin and stay green.

thelewiss1, 2020-10-28:
Edit: seems like I was slightly impatient, it has now stopped spinning and
gone red.

jneilliii, 2020-10-28:
sometimes the timings of these things are a little tricky and then the known state from the plugin side gets out of sync with the plug itself. This can be exaggerated if you control the device outside of the plugin (ie with the app). it's one of the reasons I added the polling settings, to try and keep them in line with each other by checking on regular intervals. not saying that is the case here, but it seems like it might be working now. I'll leave this ticket open in case it starts acting up again. The big thing to look out for is if the plug's ip address changes. I personally have a reserved ip address in my routers DHCP server so it always stays the same.

thelewiss1, 2020-10-28:
>  I personally have a reserved ip address in my routers DHCP server so it always stays the same.

I have done the same. Not sure why it has only just started happening, but i'm glad it's working again for now,

Many thanks, and i'll let you know if it breaks again.

DanHudson15, 2020-10-29:
I have the same issue also, and its happening on both OctoPrint systems.  Icon stays grey.

jneilliii, 2020-10-29:
please follow same instructions...and provide logs. Sounds like your plug is not accepting/responding to the check status request.

DanHudson15, 2020-10-29:
It was in my router, I turned on isolate AP on and off and it reestablished WAN communications fOR my Wi-Fi

jneilliii, 2020-10-29:
Ok cool. I'll keep that in mind for future troubleshooting assistance. Thanks for the info.

closed 2020-11-14 by jneilliii