jneilliii/OctoPrint-TPLinkSmartplug #218: [BUG]:  Plugin stopped working

corautem, 2020-11-12:
**Describe the bug**
The plugin stopped working yesterday morning out of nowhere; I have the HS110 and used to work perfectly. I just saw the warning about the firmware update of the smartplug but I don't know if it's that or not. I can see the plug connected on my router and I can access it with the KASA app, but not with plugin. If it's because of smartplug firmware update, there is anything to do to gain control? Thank you.

 - OctoPrint Version 1.42
 - Plugin Version 0.9.24

jneilliii, 2020-11-12:
I need you to enable debug logging, restart octoprint, try to toggle the plug. If it's still an issue, then grab your log files and share here. I've also seen where the ip changed if not reserved via dhcp server options. Have you tried pulling the plug from power and plugging back in?

jneilliii, 2020-11-12:
It seems you and @jerryfudd are having the same issue. It would be helpful to get logs from you both if possible, as described in the post above.

jneilliii, 2020-11-12:
Issue definitely appears to be firmware breaking change, as reported [here](https://github.com/softScheck/tplink-smartplug/issues/81).

corautem, 2020-11-12:
Hi, the smartplug ip is reserved and there's no change of ip address, I plugged out/plugged in, reset the smartplug, restart system/ Octoprint, no change.
[octoprint (2).log](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/files/5532157/octoprint.2.log)

jneilliii, 2020-11-12:
I need the plugin_tplinksmartplug_debug.log file from the logging section @corautem, but based on that linked issue above I don't believe this is going to be fixable until someone is able to reverse engineer the protocol again.

corautem, 2020-11-12:
Oups, sorry. There it is 

[plugin_tplinksmartplug_debug.log](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/files/5532318/plugin_tplinksmartplug_debug.log)

jneilliii, 2020-11-12:
Yep, that's what I kind of expected.

```
Could not connect to 192.168.1.202.
```

If it can't connect it can't send commands.

jneilliii, 2020-11-12:
What firmware version does your plug have @corautem ?

corautem, 2020-11-12:
Hardware is 4.1 and firmware is 1.1.0

jerryfudd, 2020-11-12:
> I need the plugin_tplinksmartplug_debug.log file from the logging section @corautem, but based on that linked issue above I don't believe this is going to be fixable until someone is able to reverse engineer the protocol again.

Would you be able to do something similar for the Hue Plugs as they seem to support API calls..... been using OctoHue for controlling the bulbs I'd imagine the plug wouldnt be a far stretch....

jneilliii, 2020-11-12:
I'm not sure if there is an API available from TPLINK, but if there is a way to communicate with the cloud service then yes, something similar could be achieved with some effort. Curious, are both of your plugs UK models? Doing a little discussion in the discord, it seems that this may only be effecting those devices from our quick testing of people that have my plugin isntalled.

corautem, 2020-11-12:
I'll give it a try to OctoHue tomorrow, meantime I installed your new version and the last 150 octoprint version, but now this new version won't save my plug ip address, every time I insert the ip, save and exit a new icon appears on the main screen, after 3 attempts I had 3 icons on the main screen. I uninstalled it with clean data, reinstalled it but still won't save the ip.

I only have 1 plug and yes it UK version

jneilliii, 2020-11-12:
I'll check into that other issue, but definitely the UK plugs with that firmware are definitely broken based on what I'm seeing in several other repository issues.

jneilliii, 2020-11-12:
@corautem do you have any of my other plugins installed in 1.5.0? I haven't released the same fix in those to stable versions for what I think you are experiencing with that list. There are RC releases on all my plugins that are effected by this issue, and you may have to install those manually.

jerryfudd, 2020-11-12:
> I'm not sure if there is an API available from TPLINK, but if there is a way to communicate with the cloud service then yes, something similar could be achieved with some effort. Curious, are both [...]

I meant make the same plugin for the Hue smart plug instead as it seems what we are trying to achieve is officially supported by them with their API calls.

In answer to your other question, yes mine are UK Kasa plugs.

jneilliii, 2020-11-12:
Oh, gotcha. Yes definitely a separate plugin scenario. If it's all API calls though it might be easier to fork my Tasmota plugin, as all of it's commands are using API calls instead of some weird encrypted thing like the TPLink stuff uses. Another option is my wemo plugin, which I know still works with the wemo plug I have.

danyboy88, 2020-11-12:
I just tried this today with the plugin and no joy. I have the uk version and it won't connect to the IP address but the app works which is weird

jneilliii, 2020-11-12:
Yeah, I think they may have changed the connection protocols. There's someone in the discord that's going to test a tapo p100 module to see if it's able to talk to the broken UK firmware. From what I'm seeing TPLink may have converted the firmware over to use a similar approach to how it works.

jerryfudd, 2020-11-12:
> Yeah, I think they may have changed the connection protocols. There's someone in the discord that's going to test a tapo p100 module to see if it's able to talk to the broken UK firmware. From what I'm seeing TPLink may have converted the firmware over to use a similar approach to how it works.

Appreciate you looking into it

ben-j-harris, 2020-11-13:
Another instance to add here too, showing exactly the same symptoms - UK also. Was working fine a couple of days ago through Octoprint, but now only accessible via the Kasa app.

Model: HS110
Hardware: 4.1
Firmware: 1.1.0

2hlc, 2020-11-14:
Same issue here as well, was working fine last week. Is not the update of the plugin itself because I have two PI's and one has the previous version and that one isn't working either anymore. 

Model: HS110
Hardware: 4.1
Firmware: 1.1.0

Checking status of 192.168.1.20.
[2020-11-14 09:41:55,393] DEBUG: {'system': {'get_sysinfo': {}}}
[2020-11-14 09:41:55,394] DEBUG: IP 192.168.1.20 is valid.
[2020-11-14 09:41:55,394] DEBUG: Sending command {'system': {'get_sysinfo': {}}} to 192.168.1.20
[2020-11-14 09:41:55,398] DEBUG: Could not connect to 192.168.1.20.

snecklifter, 2020-11-14:
Yeah, same here, my plug just updated and same issue.

jneilliii, 2020-11-14:
There is hope that this will work again, it's just a matter of time before someone is able to reverse engineer the protocol again.

robbrad, 2020-11-14:
If got the same issue. Would this help https://github.com/softScheck/tplink-smartplug

Looks like they closed port: 9000

I have an openwrt router and could potentially do a wire shark on the plug.

snecklifter, 2020-11-14:
> If got the same issue. Would this help https://github.com/softScheck/tplink-smartplug
> 
> Looks like they closed port: 9999
> 
> I have an openwrt router and could potentially do a wire shark on the plug.

@robbrad yeah they are working on it there:

https://github.com/softScheck/tplink-smartplug/issues/81

robbrad, 2020-11-14:
This could help https://k4czp3r.xyz/posts/reverse-engineering-tp-link-tapo/

jneilliii, 2020-11-14:
@robbrad yeah, that's the implementation [here](https://github.com/fishbigger/TapoP100) that one of the Discord users has already tried and it did not work for them unfortunately. However, there are much smarter guys than me working on this stuff and I'm sure we'll have a solution soon hopefully.

robbrad, 2020-11-14:
> @robbrad yeah, that's the implementation [here](https://github.com/fishbigger/TapoP100) that one of the Discord users has already tried and it did not work for them unfortunately. However, there are much smarter guys than me working on this stuff and I'm sure we'll have a solution soon hopefully.

Thank you and also kudos to your awesome Octoprint plugin!  

I nearly tweeted TPLink tonight asking for an opt in opt out of remote management

ibsmith04, 2020-11-16:
I also have the same problem UK also. Was working fine a couple of days ago through Octoprint, but now only accessible via the Kasa app.

Model: HS110
Hardware: 4.1
Firmware: 1.1.0

cheers

jneilliii, 2020-11-16:
There has been some headway in the softScheck issue thread and @cp2004 has been able to verify that it works for pulling status and minimal control. Once that upstream code gets flushed out hopefully we will have a usable python module soon that can be integrated with this plugin (or potentially a separate fork of this plugin) since it's only effecting the HK models right now. As other firmware updates get rolled out to other regions we may start seeing the same thing.

cp2004, 2020-11-22:
@jneilliii If I get a chance after finishing a re-write of the Marlin EEPROM editor, I don't mind having a go at the new logic for these devices. I keep remembering this needs to be done every time I print, forget to turn the printer off...

IMO, communication like this should be done using a class, that an instance of can be created for each plug. That *should* simplify the code in this plugin, which to be honest has grown and grown over time as great features have been added, and is mildly chaotic 🙂

Similar approach to what I designed for the WLED plugin - although I haven't got around to actually writing the plugin yet, it has a class for communication with each device.

robbrad, 2020-11-24:
FYI 

Firmware Rollback Option : https://www.home-assistant.io/blog/2020/11/23/tplink-local-access/

jerryfudd, 2020-11-24:
> FYI
> 
> Firmware Rollback Option : https://www.home-assistant.io/blog/2020/11/23/tplink-local-access/

Thanks, just sent a message.... its worth a shot. This plugin is the only reason I bought a Kasa plug (HS110) and quite liked it so ended up changing my entire fleet of smart plugs. Quite like that they did the KP303 power strip too.

jneilliii, 2020-11-24:
Yeah, I personally use the HS300 with the plugin, and it's great having the ability to control all the different things individually right around my printers.

Dark360, 2020-11-26:
I have the same issue with a UK Plug, but looking at the link above there was a comment added yesterday (25/11) by **Steven_​​Rollason** stating that there is a newer protocol.

jerryfudd, 2020-12-16:
> Thanks, just sent a message.... its worth a shot. This plugin is the only reason I bought a Kasa plug (HS110) and quite liked it so ended up changing my entire fleet of smart plugs. Quite like that [...]

@jneilliii just to report that I followed the link @robbrad submitted and provided my mac address etc as per my previous post 22 days ago and just got a firmware update - my plugin is back working with my uk plug :D

I've got quite a few TP-Link HS100 and HS110 units and the only one received a firmware update today (I'd been regularly checking) corresponded with the one I submitted so I have no doubt it was due to the message and subsequent pm conversation I had with the stated user.

jneilliii, 2020-12-16:
Thanks for reporting back @jerryfudd and I'm glad that support was able to roll your device back. Hopefully others that come here will have the same success.

corautem, 2020-12-18:
> > > FYI
> > > Firmware Rollback Option : https://www.home-assistant.io/blog/2020/11/23/tplink-local-access/
> > 
> > 
> > Thanks, just sent a message.... its worth a shot. This plugin is the only reason I bought a Kasa plug (HS110) and quite liked it so ended up changing my entire fleet of smart plugs. Quite like that they did the KP303 power strip too.
> 
> @jneilliii just to report that I followed the link @robbrad submitted and provided my mac address etc as per my previous post 22 days ago and just got a firmware update - the plugin is back working with my uk plug :D
> 
> I've got quite a few TP-Link HS100 and HS110 units and the only one to receive a firmware update today (I had been regularly checking) corresponded with the details I submitted so I have no doubt it was due to the message and subsequent pm conversation I had with the stated user. I have no need for my other plugs to work in this way so left them as is.

I just completely lost hope on this, but I have seen your comment and I tried to see if there's any update available and here it is, firmware version 1.1.1 and back on control. Thank you.

closed 2020-12-21 by corautem

ppriest, 2021-01-23:
FYI, since I don't see it mentioned, but the 'KLAP' protocol was reverse-engineered and integrated into Homebridge. 
There's an implementation here: https://github.com/GadgetReactor/pyHS100/compare/master...SimonWilkinson:new-protocol
Note that from my reading it requires the username and password (well, md5 of them) of the cloud account the device is registered with, unless you skip that step.

filamental, 2021-02-08:
Plugin works with HS100, firmware version 1.1.5. Consider removing the warning from the readme

cp2004, 2021-02-08:
The warning still applies, since it says **1.1.0**, you have **1.1.5**.

nirurin, 2021-04-20:
https://community.tp-link.com/en/smart-home/forum/topic/239364

I have tried to contact TPLink through the suggested forum, however I have been told: "Since the beta firmware has been stopped pushing already, there is no more beta available now. And We're also planning to push a more secure cloud API in the future before upgrading a more secure local communication authentication method on all Kasa devices. At that time, other third-party platforms/applications can register a developer account on our official website and integrate with us through our more secure APIs. If possible, Please wait for a while for the later official firmware. For more details, please refer to this link: https://community.tp-link.com/en/smart-home/forum/topic/239364 Thanks for your understanding and support."

So there is no longer a way to make the existing plug work, unless the app is updated to their new API. And there's no way that I know of to get the 'right' plug (I have a second HS110 which works fine, because it's an older 'hardware version', but they are identical in looks and I got them from the same place.

jneilliii, 2021-04-21:
If TPLink provides a developer API to work with their cloud platform I would be happy to convert the plugin to use that method instead. I wasn't aware of such a thing to this point or if it even exists.

nirurin, 2021-04-21:
That link in my post seems to imply that it exists, though I don't know how you get involved with it. It seems to say you just have to contact tplink.

Firefishy, 2023-07-06:
TP-Link are now actively refusing to release the fixed v1.1.1 firmware for HS110 (UK) HW: 4.1
I now have 5x dead devices. They tell me to use their mobile app. 🤮