jneilliii/OctoPrint-TPLinkSmartplug #221: [BUG]: Updating Plugin caused it to malfunction

Doormat1, 2020-11-13:
**Describe the bug**
Updated today and the pluging has stopped working. It is not appearing in OctoPrint. If I try installing it, the plugin manager says its already installed and nothing changes. 
Attached is the plugin console log. 

**Desktop (please complete the following information):**
Octoprint 1.3.12.
[plugin_pluginmanager_console.log](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/files/5538938/plugin_pluginmanager_console.log)

**Additional context**
Add any other context about the problem here.

jneilliii, 2020-11-13:
Huh, that logs indicate an error with the pip command, which is outside of my control. 
```
no such option: --process-dependency-links
```

Not sure what would cause that error and have not seen any other upgrade issues mentioned. If you're able to SSH to your pi can you run the command below for me and report back?

```
~/oprint/bin/pip freeze
```

Doormat1, 2020-11-13:
Hi, thanks for the quick response. Here is what I got

DEPRECATION: Python 2.7 will reach the end of its life on January 1st,
2020. Please upgrade your Python as Python 2.7 won't be maintained after
that date. A future version of pip will drop support for Python 2.7. More
details about Python 2 support in pip, can be found at
https://pip.pypa.io/en/latest/development/release-process/#python-2-support
argh==0.26.2
awesome-slugify==1.6.5
Babel==2.7.0
backoff==1.10.0
backports-abc==0.5
Bed-Visualizer==0.1.15
blinker==1.4
bson==0.5.10
cachelib==0.1
certifi==2019.9.11
chainmap==1.0.3
chardet==3.0.4
Click==7.0
colorzero==1.1
contextlib2==0.6.0.post1
emoji==0.5.4
feedparser==5.2.1
file-read-backwards==2.0.0
filetype==1.0.5
Flask==0.10.1
Flask-Assets==0.12
Flask-Babel==0.12.2
Flask-Login==0.2.11
Flask-Principal==0.4.0
frozendict==1.2
future==0.17.1
futures==3.2.0
idna==2.8
itsdangerous==1.1.0
Jinja2==2.8.1
Markdown==3.0.1
MarkupSafe==1.1.1
monotonic==1.5
netaddr==0.7.19
netifaces==0.10.9
numpy==1.16.6
Octolapse==0.4.0
OctoPrint==1.3.12
OctoPrint-Autocalibration==0.1
OctoPrint-GcodeEditor==0.2.9
OctoPrint-TPLinkSmartplug==0.9.25
pathtools==0.1.2
picamerab==1.13b1
Pillow==6.2.2
pkg-resources==0.0.0
pkginfo==1.5.0.1
psutil==5.6.3
pyasn1==0.4.7
pybonjour==1.1.1
pylru==1.2.0
pyserial==3.4
python-dateutil==2.8.1
pytz==2019.3
PyYAML==5.1.2
raven==6.10.0
regex==2019.8.19
requests==2.22.0
rsa==4.0
sarge==0.1.5.post0
scandir==1.10.0
semantic-version==2.6.0
sentry-sdk==0.7.7
singledispatch==3.4.0.3
six==1.12.0
TheSpaghettiDetective==1.4.4
tornado==4.5.3
TouchUI==0.3.16
typing==3.7.4.1
Unidecode==0.4.21
urllib3==1.25.6
watchdog==0.9.0
webassets==0.12.1
websocket-client==0.56.0
Werkzeug==0.15.6
wrapt==1.11.2

jneilliii, 2020-11-14:
Okay, that's saying you have OctoPrint-TPLinkSmartplug version 0.9.25. You sure it's not disabled?

Doormat1, 2020-11-14:
It's not shwoing on the octoprint Web page and doesn't come up in the
software update section of settings

jneilliii, 2020-11-14:
ok, we can attempt to manually uninstall. `~/oprint/bin/pip uninstall OctoPrint-TPLinkSmartplug` and see if that clears it out completely. You can verify by running `~/oprint/bin/pip freeze` again to see if it's still listed.

jneilliii, 2020-11-14:
If that doesn't work, please share your octoprint.log.

Doormat1, 2020-11-14:
OK thanks will try this later on today

Doormat1, 2020-11-14:
That worked a treat. I have also updated to octoprint 1.4.2.

jneilliii, 2020-11-14:
great news, thanks for reporting back.

closed 2020-11-14 by jneilliii