jneilliii/OctoPrint-TPLinkSmartplug #225: [BUG]: Idle timeout power-off triggered when plug is already off

benlye, 2020-12-04:
**Describe the bug**
The idle timeout power-off ignores the current state of the smart plug and kicks in even when the plug is already off.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect to OctoPrint
1. Don't turn the printer on
1. Wait for the idle time out to elapse
1. Observe the idle timeout abort dialog appearing

**Expected behavior**
The idle timeout power off should not fire and the abort dialog should not appear if the plug is already off.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows
 - Browser: Microsoft Edge
 - OctoPrint Version: 1.5.0
 - Plugin Version: 0.9.26

**Additional context**
Plugin debug log excerpt:
```
[2020-12-04 08:52:19,246] DEBUG: abortTimeout: 30
[2020-12-04 08:52:19,248] DEBUG: powerOffWhenIdle: True
[2020-12-04 08:52:19,249] DEBUG: idleTimeout: 10
[2020-12-04 08:52:19,255] DEBUG: idleIgnoreCommands: M105
[2020-12-04 08:52:19,286] DEBUG: idleTimeoutWaitTemp: 40
[2020-12-04 08:52:33,558] DEBUG: ImmutableMultiDict([('checkStatus', '192.168.1.13')])
[2020-12-04 08:52:33,559] DEBUG: Checking status of 192.168.1.13.
[2020-12-04 08:52:33,559] DEBUG: {'system': {'get_sysinfo': {}}}
[2020-12-04 08:52:33,561] DEBUG: IP 192.168.1.13 is valid.
[2020-12-04 08:52:33,561] DEBUG: Sending command {'system': {'get_sysinfo': {}}} to 192.168.1.13
[2020-12-04 08:52:33,846] DEBUG: «5ç"system":{"get_sysinfo":{"err_code":0,"sw_ver":"1.2.5 Build 171213 Rel.101014","hw_ver":"1.0","type":"IOT.SMARTPLUGSWITCH","model":"HS100(UK)","mac":"50:C7:BF:10:33:6D","deviceId":"8006D571241B0342B9634F289A1646C617BD90DE","hwId":"522CBD4857687F1AE7D7EE9B440BF26B","fwId":"00000000000000000000000000000000","oemId":"9F26B8A5A0266993C796D0BF593B8415","alias":"3D Printer","dev_name":"Wi-Fi Smart Plug","icon_hash":"","relay_state":0,"on_time":0,"active_mode":"schedule","feature":"TIM","updating":0,"rssi":-73,"led_off":0,"latitude":0.0,"longitude":0.0}}}
[2020-12-04 08:52:33,850] DEBUG: TIM
[2020-12-04 09:02:31,882] DEBUG: Idle timeout reached after 10 minute(s). Turning heaters off prior to powering off plugs.
[2020-12-04 09:02:31,884] DEBUG: Heaters below temperature.
[2020-12-04 09:02:31,885] DEBUG: Checking timelapse status before shutting off power...
[2020-12-04 09:02:31,886] DEBUG: Starting abort power off timer.
[2020-12-04 09:03:01,963] DEBUG: Automatically powering off enabled plugs.
```

jneilliii, 2020-12-04:
Yeah, it's really more of a cosmetic issue as sending the off to an already off plug doesn't harm anything, but I have been planning to look into this when I begin work on splitting the state data from settings,

jerryfudd, 2021-05-26:
Is it possible to just an option to disable the Abort Power Off Timeout notification? I know we can set the time, but I'd be happy just letting it do its thing retaining the Idle Timeout and not having to see the box.

jneilliii, 2021-05-26:
That could definitely be an option, but still wouldn't resolve the issue completely.

jerryfudd, 2021-05-26:
> That could definitely be an option, but still wouldn't resolve the issue completely.

I was just working on your last reply that sending an off to an off plug doesn't harm anything - which is true, I just don't need to know about it :D ....and I'm quite happy for it to kill the power if it is idle and never stop it.

I gained an extra monitor so have a dedicated Octoprint screen now and keep seeing it flash up all through the day (working from home)