jneilliii/OctoPrint-TPLinkSmartplug #226: Problem with gcode / OctoPrint / TPLinkSmartplug-Plugin and CheckSum

ferdi0807, 2020-12-06:
Hello everybody,

I am having a hard time getting the TPLink OctoPrint plugin to work. I set up the plugin and now everything is working except the automatic shutdown after printing using my Cura-generated endcode. 

I figured out the cause of this, but I can't seem to find a solution, so I am asking you for your help :-)

I understand, the plugin works this way:
- send "M81 192.168.xxx.xxx" (IP of the smart plug)
- smart plug turns off

Using the OctoPrint terminal, I can manually do this and it works perfectly. Using it through the Cura endcode, unfortunately the checksum seems to mess up the command.

Instead of "M81 192.168.xxx.xxx" the gcode is sent with a checksum "M81 192.168.xxx.xxx * 29" and with the checksum the plugin does not get triggered and nothing happens.

Anyone has an idea what I could do to get this to work?

Thanks in advance, your help is very appreciated!

- ferdi0807

jneilliii, 2020-12-06:
add additional delay to the gcode options or don't use the gcode method and use the idle timer option. the issue with gcode option is that it checks to see if the printer is still printing and if it is doesn't power off the plug. Your printer has a buffer so the command is seen by octoprint well in advance of when it actually happens on the printer, so increasing the delay will give the printer time to finish and the off to work.

ferdi0807, 2020-12-07:
Hi jneilliii, thx for the quick reply, I will try to use the idle timer instead. :-)

closed 2021-01-31 by github-actions[bot]