jneilliii/OctoPrint-TPLinkSmartplug #227: [BUG]:

shadrak-source, 2020-12-10:
**Describe the bug**
I changed the heating cartridge of my hotend to increase the temperature faster, when I set a temperature above 230 °, the connected plug turns off via the plugin. If I turn off TPlink smartplug everything is fine.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Octopi
2. Click on "Set hotend temp. to 250°"
After little moment, my  smartplug HS110 turn off.

**Desktop (please complete the following information):**
 - OS: windows 10 pro
 - Browser  chrome
 - OctoPrint Version 1.5.1
 - Plugin Version : TP-Link Smartplug (0.9.26)

jneilliii, 2020-12-10:
Sounds like you have temperature runaway monitoring enabled. Have you increased the temp there to compensate for your higher temps? Please share screenshots of your settings.

shadrak-source, 2020-12-12:
Ok find it, it's not a bug, there is a monitoring for max extruder temp, it was set to 250°, thank you for your help.

closed 2020-12-12 by shadrak-source