jneilliii/OctoPrint-TPLinkSmartplug #228: [BUG]: Light not turning on automatically when print starts

timur-tabi, 2020-12-10:
I'm hoping this is just simply a misunderstanding of the feature, but I was hoping I would be able to have TPLinkSmartplug automatically turn on the LEDs over my printer whenever a print starts.  Here are my configuration pages:

[image]
[image]

OctoPrint 1.5.2
Python 2.7.13
OctoPi 0.15.1

jneilliii, 2020-12-11:
Ah, I think I see the confusion. The lights will be turned on when OctoPrint starts, not a print. If you want to turn the light on when a print starts, enable the `gcode trigger` option and add `@TPLINKON 192.168.1.9` to your start gcode in your slicer or in octoprint's gcode scripts section for print start.

timur-tabi, 2020-12-15:
That works, thanks!

closed 2020-12-15 by timur-tabi