jneilliii/OctoPrint-TPLinkSmartplug #229: [BUG]:

Jon-So, 2020-12-14:
**Describe the bug**
A clear and concise description of what the bug is.
The plug Hs103 is added but doesn't show up on the list once it's created

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

I have it shown on the other printers with no problem...

**Expected behavior**
A clear and concise description of what you expected to happen. 

When you add the plug it's not visible but it's there...

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS] windows 10
 - Browser [e.g. chrome, safari] Mozilla Firefox 
 - OctoPrint Version [e.g. 22] 1.5.2, python 3.7.3 Octopi 0.17.0
 - Plugin Version 0.9.26

**Additional context**
Add any other context about the problem here.

Jon-So, 2020-12-14:
![Uploading 102045583...]

Jon-So, 2020-12-14:
[image]

Jon-So, 2020-12-14:
I did an uninstall and a reinstall and I cant seem to see the plug nor its settings... :(

jneilliii, 2020-12-14:
Is the plugin getting disabled?

Jon-So, 2020-12-14:
No it's not disabled, it still works but I can't see the plug that was there nor can I add or edit it's settings

jneilliii, 2020-12-14:
That issue was something that should have been fixed in 0.9.25 and would only show like this if you have another plugin that might be causing a conflict with jQuery sortable. Do you have Custom Controls or System Command Editor installed?

Jon-So, 2020-12-14:
Ok so when I removed unnecessary plugins it was still not showing up, then I updated the Ocotprint-WideScreen to the latest and then it showed up. And I noticed why there was an extra bolt icon in my Navbar and I was able to see everything again... so it might have something to do with the widescreen plugin... but I could be mistaken... either way problem solved!!!

closed 2020-12-15 by jneilliii