jneilliii/OctoPrint-TPLinkSmartplug #233: [Feature Request]: Automatically turn off plug when shutdown

Alwinator, 2020-12-27:
I have connected a power button to my Raspberry Pi like in this [video](https://www.youtube.com/watch?v=qAzheiZlB9k).
I just connects GPIO3 with GND when I press it. Then I added this line:
```
dtoverlay=gpio-shutdown,gpio_pin=3,active_low=1,gpio_pull=up
```
to /boot/config.txt

Done. Now I can turn my Pi on and of with a button.

BUT, I forgot about the smart plug.

Can you add the possibility to also turn off the smart plug if I turn off the Pi. It should be possible by adding a script into rc6.d like described in this [answer](https://askubuntu.com/a/416301).

Thank you a lot for your great work!

jneilliii, 2020-12-27:
Without looking at those links I assume this is running a script when you press the button? If so then you could just update your script to also send the API command to power off the printer. Check out the API documentation in the wiki [here](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/wiki/API). Otherwise this is a duplicate of #194 which I don't want to get into.

closed 2020-12-27 by jneilliii

Alwinator, 2020-12-28:
@jneilliii I don't have any script! It is the "official" to use a power button using the boot config.