jneilliii/OctoPrint-TPLinkSmartplug #234: [BUG]: "On with upload" greyed - not useable

epalzeolithe, 2021-01-01:
**Describe the bug**
Octoprint 1.52
TP-Link smartplug 0.9.26
Cura 4.8.0
Cura Fielofview plugin

> Can't select "on with upload" in the TPlink plugin configuration- it's greyed

**To Reproduce**
So i need to manually click on octoprint TPlink icon to power up the printer, then send from Cura the print job.
Otherwise, Cura cannot connect to octoprint.

I think its due to the "On with upload" option in TPlink, but it's greyed so i cannot activate it

Thanks in advance

Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
When Cura send job to octoprint, it should power on the printer before with the TP link plug

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - OctoPrint Version [e.g. 22]
 - Plugin Version

**Additional context**
Add any other context about the problem here.

epalzeolithe, 2021-01-01:
In fact i shoudl see the tplink plugin in Cura like this window : 
[image]

But i dont have this option !!!!

epalzeolithe, 2021-01-01:
SOLVED 👍
i read the python code, and understand it needs a label in the tp link configuration, otherwise it doesn't find the plug.

Perhaps we could add some code to define a generic label, if no label is defined :)

        common_api_plugins = [
            ("tplinksmartplug", "TP-Link Smartplug", []), # ip
            ("orvibos20", "Orvibo S20", []), # ip
            ("wemoswitch", "Wemo Switch", []), # ip
            ("tuyasmartplug", "Tuya Smartplug", []), # label
            ("domoticz", "Domoticz", ["idx", "username", "password"]), # ip, idx, username, password
            ("tasmota", "Tasmota", ["idx"]), # ip, idx, username, password, backlog_delay
        ]  # type: List[Tuple[str, str, List[str]]]
        for (plugin_id, plugin_name, additional_data) in common_api_plugins:
            if plugin_id in plugin_data and "arrSmartplugs" in plugin_data[plugin_id]:
                for plug_data in plugin_data[plugin_id]["arrSmartplugs"]:
                    if plug_data["ip"] and plug_data["label"]:
                        plug = OrderedDict([
                            ("plugin", plugin_id),
                            ("name", ("%s (%s)" % (plug_data["label"], plugin_name))),
                            ("label", plug_data["label"]),
                            ("ip", plug_data["ip"])
                        ])
                        for key in additional_data:
                            plug[key] = plug_data.get(key, "")
                        self._available_plugs[self._createPlugId(plug)] = plug

jneilliii, 2021-01-01:
I see you've already opened a ticket [here](https://github.com/fieldOfView/Cura-OctoPrintPlugin/issues/217) on the OctoPrint Connection plugin for cura and @fieldOfView appears to be on top of it.

closed 2021-01-01 by jneilliii

fieldOfView, 2021-01-01:
As for your original report:
> Can't select "on with upload" in the TPlink plugin configuration- it's greyed

You have to enable the `Upload Event Monitoring` option in the TP-Link Smartplug Settings in OctoPrint for that option to be enabled.

jneilliii, 2021-01-02:
Oh right, oops. forgot to mention that. Thanks Aldo.