jneilliii/OctoPrint-TPLinkSmartplug #235: Upload event monitor not working

npyoung, 2021-01-02:
**Describe the bug**
When you upload gcode for printing, the TP-Link plugin doesn't detect this and turn on the printer. The printer must be turned on and connected to manually.

**To Reproduce**
Steps to reproduce the behavior:
1. Enable upload event monitoring and autoconnect for the printer's smart plug.
2. Upload gcode (for example using the Cura octoprint plugin)
3. Wait several minutes.

**Expected behavior**
The printer should turn on when gcode is uploaded, then after 10 seconds (the auto connect delay) octoprint should connect and printing should begin.

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Browser Chrome
 - OctoPrint Version 1.5.2
 - Plugin Version 0.9.26

**Additional context**
- Slicer: Cura with Octoprint plugin

jneilliii, 2021-01-02:
I've recently found out that you may need to have the label filled in for the cura plugin to be able to function properly. See related issue https://github.com/fieldOfView/Cura-OctoPrintPlugin/issues/217 which is going to be fixed in a future update of that plugin. Is that the case for you? Otherwise, make sure that your OctoPrint Connection configuration in Cura has the plug selected to auto-power on.

[image]

npyoung, 2021-01-03:
I did not check the Cura side of things closely enough! I will test this out and report back if it's a fix, although if not then it sounds like a Cura plugin issue rather than an Octoprint-TPlink issue, so sorry for cluttering the issue tracker here.

npyoung, 2021-01-13:
Works great with the suggestion above - user error on my part.

closed 2021-01-13 by npyoung