jneilliii/OctoPrint-TPLinkSmartplug #238: [BUG]: The plug does not turn off at the end of a print job.

Fleoxi, 2021-01-12:
**Describe the bug**
The plug does not turn off at the end of a print job, can't be more explicit.

**To Reproduce**
Steps to reproduce the behavior:
1. Start Print
2. Wait till' the end

**Expected behavior**
I expect the plug to turn off.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS] : IOS & Windows
 - Browser [e.g. chrome, safari] : Opera
 - OctoPrint Version [e.g. 22] : Latest
 - Plugin Version : Latest

**Additional context**
-

Fleoxi, 2021-01-12:
Additional precision: Manually via the button in the navbar, everything works perfectly.

jneilliii, 2021-01-12:
Please share screenshots of your settings. To automatically power off you have to enable either gcode commands and put that in your end gcode or idle timeout options.

Fleoxi, 2021-01-12:
[image]
[image]
[image]

jneilliii, 2021-01-12:
Ok, thanks. I believe what issue you might be running into is related to the printer's command buffer. If you increase your gcode off delay to something a little higher to give the print time to actually complete before the off command is triggered I think it will work as expected. Start with something like 10 or 15 and increase a little at a time. It's really going to depend on how big the firmware's buffer is as to how long that delay needs to be.

Fleoxi, 2021-01-12:
How to set a delay as you advise in this case? I am not very used to the GCode

jneilliii, 2021-01-12:
I was referring to the plugin's settings here.

[image]

Fleoxi, 2021-01-12:
Okay perfect, I'll try this

closed 2021-02-01 by Fleoxi