jneilliii/OctoPrint-TPLinkSmartplug #239: No Connect to Device

guennid, 2021-01-21:
Describe the bug
The plug dont react

To Reproduce
Configured the plug, but no reaction

Desktop (please complete the following information):

OS: [e.g. iOS] : IOS & Windows
Browser [e.g. chrome, safari] : Chrome
OctoPrint Version [e.g. 22] : Latest
Plugin Version : Latest
Additional contextDescribe the bug
The plug does not turn off at the end of a print job, can't be more explicit.

To Reproduce
Steps to reproduce the behavior:

Start Print
Wait till' the end
Expected behavior
I expect the plug to turn off.

Desktop (please complete the following information):

OS: [e.g. iOS] : IOS & Windows
Browser [e.g. chrome, safari] : Opera
OctoPrint Version [e.g. 22] : Latest
Plugin Version : Latest

Log file:
2021-01-21 18:31:14,837] DEBUG: {'emeter': {'err_code': True}, 'system': {'get_sysinfo': {'relay_state': 3}}}
[2021-01-21 18:31:15,020] DEBUG: ImmutableMultiDict([('checkStatus', u'192.168.1.49')])
[2021-01-21 18:31:15,020] DEBUG: Checking status of 192.168.1.49.
[2021-01-21 18:31:15,021] DEBUG: {'system': {'get_sysinfo': {}}}
[2021-01-21 18:31:15,021] DEBUG: IP 192.168.1.49 is valid.
[2021-01-21 18:31:15,022] DEBUG: Sending command {'system': {'get_sysinfo': {}}} to 192.168.1.49
[2021-01-21 18:31:15,027] DEBUG: Could not connect to 192.168.1.49.

I checked the Lan. I can ping from the raspberry to the ip of the plug.

jneilliii, 2021-01-21:
Are you in the UK? There's a known bug if so, which you might be able to get reverted on your firmware. More info in the issue linked below.

https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/issues/218

jneilliii, 2021-01-21:
Also, make sure you enabled the `off on idle` option in the plug's individual settings...

closed 2021-01-24 by guennid