jneilliii/OctoPrint-TPLinkSmartplug #240: [BUG]: Plug turns on, Octoprint wants to connect, plug shuts down.

OxygenGS, 2021-01-23:
Hello,

I did gte a Tp link HS100 today. Did install the plugin and it responds.. jeej.
However.. Once i click the icon on the top to turn the plug on, all is fine. 
BUT as soon as octorpint wants to connect to the printer, the plug turns off...
Someone who can help me out :(?

jneilliii, 2021-01-23:
Please share your settings. There are some known conflicts with some printers that are USB back powered and disconnect monitoring enabled.

OxygenGS, 2021-01-23:
Nvm.. i feel so stupid! i didnt had the USB cable connected...
Because i changed the PSU before i installed the plugin haha.. sorry!

OxygenGS, 2021-01-23:
But i still have 1 question: how can i shut down my pi also once the printer goes off ?

jneilliii, 2021-01-23:
Something like this should work. Just make sure the timer is longer than it takes for your pi to shutdown completely, 30 seconds is probably fine I would think, but you might want to time it manually first just to be sure.

[image]

OxygenGS, 2021-01-23:
Thnx!!!

closed 2021-01-23 by jneilliii