jneilliii/OctoPrint-TPLinkSmartplug #241: [BUG]:  Plug turns on, Octoprint wants to connect, plug shuts down.

OxygenGS, 2021-01-24:
Hello,

I did gte a Tp link HS100 today. Did install the plugin and it responds.. jeej.
However.. Once i click the icon on the top to turn the plug on, all is fine.
BUT as soon as octorpint wants to connect to the printer, the plug turns off...
Someone who can help me out :(?
My settings:
https://prnt.sc/xjhk7d
https://prnt.sc/xjhkub

OxygenGS, 2021-01-24:
UPDATE: i did turn those off, and now it wokr.. but i would like to have it enabled
https://prnt.sc/xjhtlj

jneilliii, 2021-01-24:
Try it with just everything but error monitoring. The other two are not known to cause this. If you enable debug logging in the main plugin setting's we can possibly determine if this is a bug. Enable that and restart OctoPrint and then try again. My guess is that your printer is sending an error back to OctoPrint, which the monitoring triggers.

closed 2021-02-16 by github-actions[bot]