jneilliii/OctoPrint-TPLinkSmartplug #243: [BUG]: plug numbers do not match

ingo916, 2021-01-25:
**Describe the bug**
In the TP-Link Kasa app, the outlets on my HS300 are listed as 1-6 as shown in the following screenshot. 
However in the plugin it's off, 1 would be outlet #2 on the power strip then 2 would be outlet #3 and so on. 
**Screenshots**
[image]

 - OctoPrint Version 1.5.3
 - Plugin Version 0.9.26

jneilliii, 2021-01-25:
The instructions for multiple plug/strip devices is to use a 0 based number, so for now you should be able to subtract 1 from the number listed in the plugin's settings.

jneilliii, 2021-02-28:
adjusted code in above commit to match the index to the labels shown on the device and in the app, you can switch the release channel to `Release Candidate` in OctoPrint's software update settings and upgrade when prompted to try it out. @tideline3d may be interested in this one as well 😄

PR #264 (2021-04-14): 1.0.0
* clear _autostart_file on print done or canceled events to prevent from auto starting a file on next connect.
* fix missing poll_status in init
* resolve idle timeout issues related to plugs powering off with gcode commands, resolves #254
* change gcode trigger logic for off commands to account for printer buffer, resolves #258
* change index values for strip devices to match labels and the Kasa app, resolves #243
* add sending gcode commands before off and after on, resolves #160, resolves #170
* turn on only if we are closed or errored state with upload event, thanks to @ayufan
* adde [... cut]
closed 2021-02-28 by jneilliii