jneilliii/OctoPrint-TPLinkSmartplug #244: [BUG]: Turning the plug on after idle shutdown causes the printer to re-start last job

samwiseg0, 2021-01-26:
**Describe the bug**
Turning the plug on from the UI/manually causes the printer to start printing the last job it had before it was powered off. 

**To Reproduce**
Steps to reproduce the behavior:
1. Have the plugin shutdown the printer after idle for x time
2. Turn the plug on manually/from the UI
3. The printer will start printing the last job instead of sitting idle.

**Expected behavior**
When the printer is turned back on it should remain in an idle state. 

**Screenshots**
N/A

**Desktop (please complete the following information):**
 - OS: Any
 - Browser Any
 - OctoPrint Version 1.5.3
 - Plugin Version 0.9.26

**Additional context**
N/A

jneilliii, 2021-01-26:
Thanks @samwiseg0, I know what I need to do to fix this, it's definitely a bug since I'm not clearing out the flag for autostart on print completion/cancel.

jneilliii, 2021-01-27:
This should be resolved by the above commit.

closed 2021-02-28 by jneilliii