jneilliii/OctoPrint-TPLinkSmartplug #247: [BUG]: one device doesn't start at print start, but seems configured the same as others that do.

rrauenza, 2021-02-01:
**Describe the bug**
 I have four plugs, all on a tplink power strip.  Three are all configured the same, one is the printer itself.  One of the three doesn't turn on when a print starts, but the others do:
 
[image]

**To Reproduce**
Steps to reproduce the behavior:

- Start a print..
- Thordsen and Light turn on
- Fans do not turn on

I CAN manually turn the fan on and off with the icon on main page.

**Expected behavior**

Fan should turn on just like the other two devices.

**Desktop (please complete the following information):**
 -
 - OctoPrint Version OctoPrint: 1.5.3 
 - Plugin Version TP-Link Smartplug (0.9.26)

**Additional context**

I looked at the config diff ... and I noticed that emeter and status are missing in the yaml -- although the icons do turn green/red for the fans when I click on them.  

Config:

```
    tplinksmartplug:
        _config_version: 13
        abortTimeout: '60'
        arrSmartplugs:
        -   autoConnect: true
            autoConnectDelay: '5'
            autoDisconnect: true
            autoDisconnectDelay: 0
            automaticShutdownEnabled: false
            btnColor: '#808080'
            countdownOffDelay: '1'
            countdownOnDelay: '0'
            currentState: 'on'
            displayWarning: false
            emeter:
                get_realtime:
                    current_ma: 118
                    err_code: 0
                    power_mw: 8830
                    total_wh: 193
                    voltage_mv: 123243
            event_on_disconnect: false
            event_on_error: false
            event_on_startup: false
            event_on_upload: false
            gcodeEnabled: false
            gcodeOffDelay: '600'
            gcodeOnDelay: 0
            icon: icon-print
            ip: 3dprinterswitch/0
            label: Prusa Mk3
            sysCmdOff: false
            sysCmdOffDelay: 0
            sysCmdOn: false
            sysCmdOnDelay: 0
            sysRunCmdOff: ''
            sysRunCmdOn: ''
            thermal_runaway: false
            useCountdownRules: false
            warnPrinting: true
[... 370 lines cut ...]
[2021-01-31 17:44:38,215] DEBUG: 0
[2021-01-31 17:44:38,215] DEBUG: Checking status of 3dprinterswitch/3.
[2021-01-31 17:44:38,216] DEBUG: {'system': {'get_sysinfo': {}}}
[2021-01-31 17:44:38,216] DEBUG: Invalid ip 3dprinterswitch trying hostname.
[2021-01-31 17:44:38,229] DEBUG: Hostname 3dprinterswitch is valid.
[2021-01-31 17:44:38,230] DEBUG: {'system': {'get_sysinfo': {}}}
[2021-01-31 17:44:38,230] DEBUG: Invalid ip 3dprinterswitch trying hostname.
[2021-01-31 17:44:38,236] DEBUG: Hostname 3dprinterswitch is valid.
[2021-01-31 17:44:38,237] DEBUG: Sending command {'system': {'get_sysinfo': {}}} to 3dprinterswitch
[2021-01-31 17:44:38,374] DEBUG: «²f"system":{"get_sysinfo":{"sw_ver":"1.0.19 Build 200224 Rel.090814","hw_ver":"1.0","model":"HS300(US)","deviceId":"800671B8B6CC08249B39955E766F0C281B1C1245","oemId":"5C9E6254BEBAED63B2B6102966D24C17","hwId":"34C41AA028022D0CCEA5E678E8547C54","rssi":-73,"longitude_i":-1220547,"latitude_i":373498,"alias":"TP-LINK_Power Strip_CE34","status":"new","mic_type":"IOT.SMARTPLUGSWITCH","feature":"TIM:ENE","mac":"D8:0D:17:BA:CE:34","updating":0,"led_off":0,"children":[{"id":"800671B8B6CC08249B39955E766F0C281B1C124500","state":1,"alias":"Printer","on_time":1119377,"next_action":{"type":-1}},{"id":"800671B8B6CC08249B39955E766F0C281B1C124501","state":1,"alias":"Dryer","on_time":482,"next_action":{"type":-1}},{"id":"800671B8B6CC08249B39955E766F0C281B1C124502","state":1,"alias":"Light","on_time":482,"next_action":{"type":-1}},{"id":"800671B8B6CC08249B39955E766F0C281B1C124503","state":1,"alias":"Fans","on_time":0,"next_action":{"type":-1}},{"id":"800671B8B6CC08249B39955E766F0C281B1C124504","state":0,"alias":"Plug 5","on_time":0,"next_action":{"type":-1}},{"id":"800671B8B6CC08249B39955E766F0C281B1C124505","state":1,"alias":"Cat Light","on_time":935,"next_action":{"type":-1}}],"child_num":6,"err_code":0}}}
[2021-01-31 17:44:38,380] DEBUG: get_device_id response: 800671B8B6CC08249B39955E766F0C281B1C124503
[2021-01-31 17:44:38,380] DEBUG: Sending command {'system': {'get_sysinfo': {}}, 'context': {'child_ids': ['800671B8B6CC08249B39955E766F0C281B1C124503']}} to 3dprinterswitch
[2021-01-31 17:44:38,549] DEBUG: «²f"system":{"get_sysinfo":{"sw_ver":"1.0.19 Build 200224 Rel.090814","hw_ver":"1.0","model":"HS300(US)","deviceId":"800671B8B6CC08249B39955E766F0C281B1C1245","oemId":"5C9E6254BEBAED63B2B6102966D24C17","hwId":"34C41AA028022D0CCEA5E678E8547C54","rssi":-73,"longitude_i":-1220547,"latitude_i":373498,"alias":"TP-LINK_Power Strip_CE34","status":"new","mic_type":"IOT.SMARTPLUGSWITCH","feature":"TIM:ENE","mac":"D8:0D:17:BA:CE:34","updating":0,"led_off":0,"children":[{"id":"800671B8B6CC08249B39955E766F0C281B1C124500","state":1,"alias":"Printer","on_time":1119377,"next_action":{"type":-1}},{"id":"800671B8B6CC08249B39955E766F0C281B1C124501","state":1,"alias":"Dryer","on_time":482,"next_action":{"type":-1}},{"id":"800671B8B6CC08249B39955E766F0C281B1C124502","state":1,"alias":"Light","on_time":482,"next_action":{"type":-1}},{"id":"800671B8B6CC08249B39955E766F0C281B1C124503","state":1,"alias":"Fans","on_time":0,"next_action":{"type":-1}},{"id":"800671B8B6CC08249B39955E766F0C281B1C124504","state":0,"alias":"Plug 5","on_time":0,"next_action":{"type":-1}},{"id":"800671B8B6CC08249B39955E766F0C281B1C124505","state":1,"alias":"Cat Light","on_time":935,"next_action":{"type":-1}}],"child_num":6,"err_code":0}}}
[2021-01-31 17:44:38,582] DEBUG: TIM:ENE
[2021-01-31 17:44:38,591] DEBUG: Invalid ip 3dprinterswitch trying hostname.
[2021-01-31 17:44:38,599] DEBUG: Hostname 3dprinterswitch is valid.
[2021-01-31 17:44:38,601] DEBUG: {'system': {'get_sysinfo': {}}}
[2021-01-31 17:44:38,604] DEBUG: Invalid ip 3dprinterswitch trying hostname.
[2021-01-31 17:44:38,631] DEBUG: Hostname 3dprinterswitch is valid.
[2021-01-31 17:44:38,632] DEBUG: Sending command {'system': {'get_sysinfo': {}}} to 3dprinterswitch
[2021-01-31 17:44:38,818] DEBUG: «²f"system":{"get_sysinfo":{"sw_ver":"1.0.19 Build 200224 Rel.090814","hw_ver":"1.0","model":"HS300(US)","deviceId":"800671B8B6CC08249B39955E766F0C281B1C1245","oemId":"5C9E6254BEBAED63B2B6102966D24C17","hwId":"34C41AA028022D0CCEA5E678E8547C54","rssi":-73,"longitude_i":-1220547,"latitude_i":373498,"alias":"TP-LINK_Power Strip_CE34","status":"new","mic_type":"IOT.SMARTPLUGSWITCH","feature":"TIM:ENE","mac":"D8:0D:17:BA:CE:34","updating":0,"led_off":0,"children":[{"id":"800671B8B6CC08249B39955E766F0C281B1C124500","state":1,"alias":"Printer","on_time":1119377,"next_action":{"type":-1}},{"id":"800671B8B6CC08249B39955E766F0C281B1C124501","state":1,"alias":"Dryer","on_time":482,"next_action":{"type":-1}},{"id":"800671B8B6CC08249B39955E766F0C281B1C124502","state":1,"alias":"Light","on_time":482,"next_action":{"type":-1}},{"id":"800671B8B6CC08249B39955E766F0C281B1C124503","state":1,"alias":"Fans","on_time":0,"next_action":{"type":-1}},{"id":"800671B8B6CC08249B39955E766F0C281B1C124504","state":0,"alias":"Plug 5","on_time":0,"next_action":{"type":-1}},{"id":"800671B8B6CC08249B39955E766F0C281B1C124505","state":1,"alias":"Cat Light","on_time":935,"next_action":{"type":-1}}],"child_num":6,"err_code":0}}}
[2021-01-31 17:44:38,824] DEBUG: get_device_id response: 800671B8B6CC08249B39955E766F0C281B1C124503
[2021-01-31 17:44:38,824] DEBUG: Sending command {'emeter': {'get_realtime': {}}, 'context': {'child_ids': ['800671B8B6CC08249B39955E766F0C281B1C124503']}} to 3dprinterswitch
[2021-01-31 17:44:38,996] DEBUG: «i¹"emeter":{"get_realtime":{"voltage_mv":122431,"current_ma":10,"power_mw":1,"total_wh":16,"err_code":0}}}
```

rrauenza, 2021-02-01:
So... after having this issue for as long as I can remember, it worked correctly this morning when I started a print.  I just rebooted the pi ... which maybe is related.

I wouldn't look too deep into this until I reproduce it again.

jneilliii, 2021-02-02:
Thanks for following up @rrauenza. I'll leave this one open for now, and the stale bot will grab it and auto-close after a bit. If I don't hear back from you I'll just let it do its job.

rrauenza, 2021-02-02:
I think this is user error.  I can't explain why it worked that one time, though!

I just realized I'm using GCODE Trigger and have these two scripts:

pi@octopi:~/.octoprint/logs $ less ../scripts/gcode/beforePrintStarted 
@TPLINKON 3dprinterswitch/2
@TPLINKON 3dprinterswitch/1
@TPLINKON 3dprinterswitch/0

pi@octopi:~/.octoprint/logs $ less ../scripts/gcode/afterPrintDone 
@TPLINKOFF 3dprinterswitch/2
@TPLINKOFF 3dprinterswitch/1
@TPLINKOFF 3dprinterswitch/0
pi@octopi:~/.octoprint/logs $

realbadfox, 2021-02-04:
I have the same Issue, with one plug no problem.
But with two plugs the automatic shutdown doesn't work anymore!

jneilliii, 2021-02-04:
Thanks for adding in @realbadfox there very well could be a bug here, I just haven't had the time to try and reproduce yet. Can you verify if it doesn't work at all or only works on one?

realbadfox, 2021-02-05:
i seems that it doesn't work at all

jneilliii, 2021-02-05:
Well, that definitely seems like a different issue. You have the idle options enabled both on the general settings area and on the device itself? Please enable debug logging and restart OctoPrint and then provide your plugin_tplinksmartplug_debug.log file for further analysis. You can just drag the whole file here into a comment on github. Make sure you've waiting long enough for the idle timeout to supposedly trigger.

realbadfox, 2021-02-05:
Yes both enabled.

And where do i find the logfile?

jneilliii, 2021-02-05:
In OctoPrint's settings under logging.

jneilliii, 2021-02-06:
Hey, just had a possible test for you guys to run for me in the above linked issue. That user was experiencing the same thing I think and once they disabled gcode triggers the idle timer worked. Could you guys verify you too have gcode triggers enabled and test with those turned off as well to see if the idle timers work or not then?

realbadfox, 2021-02-06:
here is my logging of yesterday, where it should turn of with idle. but haven't
[octoprint.log](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/files/5937180/octoprint.log)

jneilliii, 2021-02-06:
I need the file plugin_tplinksmartplug_debug.log file as well @realbadfox. I notice a bunch of errors in your log relative to the octopi support plugin and temps. are you running on a pi or not?

jneilliii, 2021-02-06:
Oh it looks like that's a different plugin...Opitemp Plugin.

realbadfox, 2021-02-06:
yes of course, I am using a PI!
Why it is a different plugin? i use TPLink Smartplug!
here is the second log file
[plugin_tplinksmartplug_debug.log](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/files/5937272/plugin_tplinksmartplug_debug.log)

jneilliii, 2021-02-06:
I was just mentioning that the only thing I saw in the octoprint.log was errors with that other plugin OpiTemp. This is what I find interesting in this log though...

```
[2021-02-06 07:45:48,731] DEBUG: Waiting for heaters() before shutting power off...
[2021-02-06 07:45:52,957] DEBUG: Turning on 192.168.0.221.
[2021-02-06 07:45:52,959] DEBUG: {'autoConnect': True, 'ip': '192.168.0.221', 'btnColor': '#808080', 'autoDisconnectDelay': '5', 'sysCmdOff': False, 'gcodeEnabled': True, 'gcodeOnDelay': '0', 'emeter': {'get_realtime': {}}, 'countdownOnDelay': 1, 'label': '3DDruckerSteckdose', 'warnPrinting': True, 'thermal_runaway': False, 'gcodeOffDelay': '0', 'sysCmdOnDelay': 0, 'event_on_error': False, 'countdownOffDelay': 1, 'sysRunCmdOff': 'M81 192.168.0.221', 'sysRunCmdOn': '', 'autoDisconnect': True, 'automaticShutdownEnabled': True, 'event_on_upload': False, 'useCountdownRules': False, 'event_on_startup': False, 'icon': 'icon-bolt', 'displayWarning': False, 'autoConnectDelay': '5', 'event_on_disconnect': False, 'sysCmdOn': False, 'currentState': 'unknown', 'sysCmdOffDelay': '10'}
[2021-02-06 07:45:52,960] DEBUG: IP 192.168.0.221 is valid.
[2021-02-06 07:45:52,960] DEBUG: Sending command {'system': {'set_relay_state': {'state': 1}}} to 192.168.0.221
[2021-02-06 07:45:53,049] DEBUG: «  -ý"system":{"set_relay_state":{"err_code":0}}}
[2021-02-06 07:45:53,052] DEBUG: 0
[2021-02-06 07:45:53,054] DEBUG: Resetting idle timer since plug 192.168.0.221 was just turned on.
[2021-02-06 07:45:53,055] DEBUG: Checking status of 192.168.0.221.
[2021-02-06 07:45:53,056] DEBUG: {'system': {'get_sysinfo': {}}}
[2021-02-06 07:45:53,056] DEBUG: IP 192.168.0.221 is valid.
[2021-02-06 07:45:53,057] DEBUG: Sending command {'system': {'get_sysinfo': {}}} to 192.168.0.221
[2021-02-06 07:45:53,108] DEBUG: « 1ã"system":{"get_sysinfo":{"sw_ver":"1.1.5 Build 200828 Rel.081157","hw_ver":"4.0","model":"HS100(EU)","deviceId":"80063CB163C0D1EA2F477018FFE5076D1C54E216","oemId":"76ABF4259BE3FBA5A287BFE55D1848D7","hwId":"4CF2439894182E79E641A055C0519425","rssi":-48,"longitude_i":163728,"latitude_i":482185,"alias":"3DDrucker","status":"new","mic_type":"IOT.SMARTPLUGSWITCH","feature":"TIM","mac":"74:DA:88:FD:26:65","updating":0,"led_off":0,"relay_state":1,"on_time":0,"active_mode":"none","icon_hash":"","dev_name":"Smart Wi-Fi Plug","next_action":{"type":-1},"err_code":0}}}
[2021-02-06 07:45:53,141] DEBUG: TIM
[2021-02-06 07:45:53,736] DEBUG: Aborted power off due to activity.
```
It seems at the end of the wait for heaters part when it would be powering off it's powering on, and then there's an abort due to activity. I was hoping to look into this today to see if I can reproduce with my HS300.

realbadfox, 2021-02-06:
I turn the light (192.168.0.221) on 15sec after print.
Make a picture and turn them off.

And the printer (192.168.0.220) should turn off 2min after idle.

realbadfox, 2021-02-09:
Always same. Lights go off but not the printer

jneilliii, 2021-02-23:
This should be fixed in the latest 1.0.0rc3 release candidate. If you want to test it out and confirm that would be great, just go into software update settings and change the release channel to `Release Candidate` for the plugin and you should get an update notification.

realbadfox, 2021-02-23:
ok i will try

realbadfox, 2021-02-23:
still running and running, i turn 2 min idle timing on, but still didn't turn off.
[octoprint.log](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/files/6031750/octoprint.log)

jneilliii, 2021-02-23:
Something looks messed up with your settings possibly. 

```
Could not convert '' to a valid integer when getting option ['plugins', 'tplinksmartplug', 'idleTimeoutWaitTemp']
```

can you check your temperature setting for the idle section please?

[image]

realbadfox, 2021-02-23:
Oh yes. I have nothing in there. I thought with no value it won't be restricted for the idle mode.
I will try again.

realbadfox, 2021-02-25:
Now it is working! Perfect, thank you.

closed 2021-02-28 by jneilliii