jneilliii/OctoPrint-TPLinkSmartplug #248: [BUG]: PrusaSlicer upload via API does not trigger upload event

Gh0stWalk3r, 2021-02-01:
**Describe the bug**
When I upload my sliced gcode file with PrusaSlicer (via API) to Octoprint, the upload event does not activate the printer.

**To Reproduce**
Steps to reproduce the behavior:
1. Enable 'Upload Event Monitoring' in the plugin settings
2. Enable 'On with Upload' in the smartplug editor
3. Upload sliced GCode directly in PrucaSlicer (setup instructions from [Prusa Knowledge Base](https://help.prusa3d.com/en/article/sending-files-to-octoprint-duet_1663))
4. Issue: Smartplug does not turn on

**Expected behavior**
When I upload a file via API or UI, I expect the smartplug to turn on, when it is configured to do so.

**Desktop (please complete the following information):**
 - OS: Windows 10
 - PrusaSlicer Version 2.3.0
 - OctoPrint Version 1.5.3
 - Plugin Version 0.9.26

**Additional context**
I followed the instructions from the [Prusa Knowledge Base](https://help.prusa3d.com/en/article/sending-files-to-octoprint-duet_1663) to connect PrusaSlicer with Octoprint.

jneilliii, 2021-02-01:
Just to confirm, are you checking the box in the upload from PrusaSlicer dialog to auto start printing?

Gh0stWalk3r, 2021-02-01:
Hi @jneilliii, thanks for your fast response and great plugins 😄 

No, I did not try that. I thought that the upload should be enough to trigger it.
I can confirm that enabling the 'Start printing after upload' checkbox in the PrusaSlicer dialog triggers the event to turn on the smartplug.

I will adapt my workflow to prepare the printer prior upload to Octoprint, with enabled 'Start printing after upload'.
Thanks a lot for your help!

jneilliii, 2021-02-01:
Yeah, that is the flag on the API that is required to trigger the on event. I felt it more appropriate as you may want to upload files outside of these direct send methods in the web interface and that could cause some issues potentially.

closed 2021-02-01 by jneilliii