jneilliii/OctoPrint-TPLinkSmartplug #250: Idle Power Test is working. After print not working

Jack77777777, 2021-02-05:
I have installed your great plugin yesterday and it´s great. I am using two Plugs 192.168.178.120 for printer PSU and 192.168.178.121 for light.
192.168.178.120 is the one not working.
I don´t know if i did a mistake in the config or it is a bug. 
I did a test first by power on in thorugh navbar button, raised temp to 40°, set to 0° again and printer powered off as expected.
[image]

Then changed settings to:
[image]
and started two test prints. Bith times the printer doesn´t power off.

Additional settings and debug:
[image]
[image]
[image]
[plugin_tplinksmartplug_debug.log](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/files/5932421/plugin_tplinksmartplug_debug.log)

jneilliii, 2021-02-05:
Thank you for your wonderful complete issue submission. Your issue I believe that you are trying to use GCODE triggers for powering off on print completion? The issue I think you are running into is that the delay for the off trigger needs to be high enough for the printer's buffer to clear and OctoPrint to go back into `Operational` mode if the warn while printing option is enabled as well. If you change your gcode off delay to 20 or 30 I bet it will work as expected.

Jack77777777, 2021-02-05:
Hi @jneilliii,

Thank you for the quick reply. I use Gcode Trigger for the light not for the printer. Do you mean this can cause the issue anyway?

jneilliii, 2021-02-05:
It could definitely prevent the gcode trigger from working, but shouldn't effect the idle timers from working. It's starting to sound a lot like #247

Jack77777777, 2021-02-05:
Disabled Gcode Triggers and and Popups. Printer powered off on idle, Yippie!
Here are some screenshots from the settings:
[image]

[image]

[image]

jneilliii, 2021-02-06:
Thanks. Let me check in the other issue to see if it's the same for those guys as well. Once we have a consensus it will be easier to narrow down where the issue is.

Jack77777777, 2021-02-08:
@jneilliii could you isolate the issue? I can do more tests and provide logs if needed.

jneilliii, 2021-02-09:
So I did do some testing over the weekend and couldn't reproduce this 100%. One thing I noticed just now on your last set of screenshots is that your `Abort Power Off Timeout` is set at 0, but wasn't before. Can you put that back to something like 30 seconds? The one difference during my testing was that I used the `M80/M81` gcode commands rather than the custom `@TPLINKON/@TPLINKOFF` commands. When I get another opportunity I'll see if testing with those instead makes a difference or not.

Jack77777777, 2021-02-09:
Hi @jneilliii i already tested with Abort Power Off Timeout 30 and disabled gcodd trigger, what worked. Actually i have the power off warning enabled and this works too. Shall i test with M80, M81 and Abort Power Off Timeout 30?

jneilliii, 2021-02-09:
By all means, the more data I have the better. Here are the settings that I was using for my testing as reference. 

[image]

[image]

[image]

[image]

Jack77777777, 2021-02-09:
Tested with same settings. Waited 1h after print but plug doesn't power off on idle. Plug with M81 powers off. Opened Octoprint in Chrome. After a minute or so Popup for power off appeared and powered off idle plug. BTW: I couldn't cancel the power off by clicking cancel on Popup. Tested a second time and behaviour was the same. Doing another test now.

Jack77777777, 2021-02-09:
@jneilliii i checked logs and think the below snippets show exactly what i mean.
### octoprint.log
2021-02-09 11:19:25,842 - octoprint.printer.standard.job - INFO - Print job selected - origin: local, path: Printer_Upgrades/Waveshare_LCD_Case/Test_TPLink_v1.gcode
...
2021-02-09 11:19:25,962 - octoprint.plugins.tplinksmartplug - INFO - WS_LIGHTSON (**What´s going on here??? Weired**)
2021-02-09 11:19:25,962 - octoprint.plugins.tplinksmartplug - INFO - 
2021-02-09 11:19:25,963 - octoprint.plugins.ws281x_led_status - INFO - Turning lights on
2021-02-09 11:19:25,966 - octoprint.plugins.tplinksmartplug - INFO - TPLINKON
2021-02-09 11:19:25,968 - octoprint.plugins.tplinksmartplug - INFO - 192.168.178.121
...
2021-02-09 11:19:30,062 - octoprint.server.util.sockjs - INFO - Client connection closed: ::ffff:192.168.178.94
...
2021-02-09 11:26:36,780 - **octoprint.plugins.tplinksmartplug** - INFO - **WS_LIGHTSOFF** (**again weired**)
2021-02-09 11:26:36,781 - octoprint.plugins.tplinksmartplug - INFO - 
2021-02-09 11:26:36,782 - octoprint.plugins.ws281x_led_status - INFO - Turning light off
2021-02-09 11:26:36,784 - octoprint.plugins.tplinksmartplug - INFO - TPLINKOFF
2021-02-09 11:26:36,784 - octoprint.plugins.tplinksmartplug - INFO - 192.168.178.121
2021-02-09 11:26:36,787 - octoprint.util.comm - INFO - Changing monitoring state from "Finishing" to "Operational"
...
2021-02-09 11:49:41,173 - octoprint.server.util.sockjs - INFO - New connection from client: ::ffff:192.168.178.96
2021-02-09 **11:49:42,579** - octoprint.server.util.flask - INFO - Passively logging in user _api from ::ffff:192.168.178.96
2021-02-09 11:49:42,579 - octoprint.access.users - INFO - Logged in user: _api
2021-02-09 11:49:42,604 - octoprint.server.util.sockjs - INFO - User _api logged in on the socket from client ::ffff:192.168.178.96
2021-02-09 11:49:53,805 - octoprint.server.util.sockjs - INFO - Client connection closed: ::ffff:192.168.178.96
2021-02-09 12:02:28,714 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2021-02-09 12:09:44,342 - octoprint.util.comm - INFO - Changing monitoring state from "Operational" to "Offline"

### plugin_tplinksmartplug_debug.log
[2021-02-09 **12:09:44,197**] INFO: Turning off 192.168.178.120 at 2021-02-09 12:09:44.197606

The marked timestamps show after logging in again the delay time of 20 minutes starts and powers off idle plug.

However i will remove WS_LIGHTSON and WS_LIGHTSOFF from Gcode Scripts and test again.

Jack77777777, 2021-02-09:
> However i will remove WS_LIGHTSON and WS_LIGHTSOFF from Gcode Scripts and test again.

No success. I have tousands of:
**tornado.access - WARNING - 409 GET /api/printer ...**
Could this cause the issue ?

Fact is that the idle counter first starts as soon as i login to octoprint, if gcode trigger is enabled.
Without Gcode Trigger it is working as designed.

Jack77777777, 2021-02-09:
Tested several times with WS281x Plugin disabled. For me the idle power off is only working when octoprint gets an login request like:
Chrome Browser _**Passively logging in user myuser from ::ffff:192.168.178.94**_ or
App (Octopod for IOS) _**Passively logging in user _api from ::ffff:192.168.178.96**_

This is only the case if Gcode Trigger is enabled for any device within **tplinksmartplug plugin**
Without any Gcode Trigger enabled idle power off is worling wothout login

Logs:
[octoprint (3).log](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/files/5952108/octoprint.3.log)
[plugin_tplinksmartplug_debug (5).log](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/files/5952122/plugin_tplinksmartplug_debug.5.log)

jneilliii, 2021-02-09:
The API errors are typically related to an outside app connecting to octoprint, but the printer not being available. I see them from OctoDash integrations. Not much you can do there for that, but I think in the next release of OctoPrint the logging for those gets a little quieter. 

I'll evaluate your logs. The "weird" parts you put in there are from the plugin processing the @ commands, it's just logging that an action command came through the pipeline, shouldn't effect the logic. 

I want to also thank you for the detailed testing. This should help me with getting down to root cause hopefully.

Jack77777777, 2021-02-09:
> The API errors are typically related to an outside app connecting to octoprint, but the printer not being available. I see them from OctoDash integrations. Not much you can do there for that, but I think in the next release of OctoPrint the logging for those gets a little quieter.
> 

I am using Octodash since Sunday. Thanks for clarifying

Jack77777777, 2021-02-12:
@jneilliii is there anything else i can help?

jneilliii, 2021-02-12:
I don't think so at this time. Once I've had time to evaluate and walk through my processing logic again I may reach back out for you to test a rc branch, but that's probably not going to happen until later this weekend.

jneilliii, 2021-02-14:
I was hoping to work on this issue over the weekend, but unfortunately cut my hand removing supports last night and it might be a few days before I'm able to debug this situation.

Jack77777777, 2021-02-14:
😖
Get Well soon

Jack77777777, 2021-02-19:
Hi @jneilliii 
Any chance to get a rc this weekend?

jneilliii, 2021-02-19:
My finger is doing better now (although still sore) so I probably will be looking at this in depth over the weekend to try and figure out the workflow with your settings.

jneilliii, 2021-02-20:
Can you try something for me @Jack77777777? Go into plugin manager and click this icon to clear the plugin's settings and add your plugs back in from scratch.

[image]

Looking through that octoprint.log I noticed that there was a setting that was causing an error and think your settings got corrupted somehow.

jneilliii, 2021-02-20:
I take that back. I just found where that error was coming from and have corrected. 1.0.0rc2 inbound to fix that error, and hopefully that will resolve this problem.

Jack77777777, 2021-02-20:
@jneilliii 
Unfortunately same issue with this rc. I deinstalled previous version and deleted data before installing 1.0.0rc2. Any recommended settings i can test?

jneilliii, 2021-02-20:
Can you share your exact settings again? It would make it way easier for me if you could grab the config.yaml file from your pi using winscp or filezilla and strip out everything but the `plugins > tplinksmartplug` section.

Jack77777777, 2021-02-21:
@jneilliii 
Here you go. I did two tests with different settings.

[config1.yaml.txt](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/files/6017192/config1.yaml.txt)
[config2.yaml.txt](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/files/6017193/config2.yaml.txt)

jneilliii, 2021-02-23:
Just released 1.0.0rc3 version that I think will resolve this issue.

Jack77777777, 2021-02-23:
I can confirm 1.0.0rc3 fixed this issue. 192.168.178.121 is Plug LED Strip, 192.168.178.120 is Plug Printer
Snipped _plugin_tplinksmartplug_debug.log_:
```
[2021-02-23 14:01:25,240] INFO: Turning off 192.168.178.121 at 2021-02-23 14:01:25.240776
[2021-02-23 14:11:26,246] INFO: Turning off 192.168.178.120 at 2021-02-23 14:11:26.245925
```
Snipped _octoprint.log_:
```
2021-02-23 14:01:24,464 - octoprint.util.comm - INFO - Changing monitoring state from "Printing" to "Finishing"
...
2021-02-23 14:11:26,376 - octoprint.util.comm - INFO - Changing monitoring state from "Operational" to "Offline"
```

arathald, 2021-02-27:
> Thank you for your wonderful complete issue submission. Your issue I believe that you are trying to use GCODE triggers for powering off on print completion? The issue I think you are running into is [...]

Hi, firstly, thanks for your work on this plugin, it's reliable and responsive enough that it's bumped adding in a proper relay for my PSU essentially to the bottom of my 3d printing to do list :)

I was having this exact issue and it really wasn't obvious to me what was going on. Setting my delay to 5s (as the m81 command is the last in my print job) solved the issue for me. Since this isn't really obvious, would you consider adding this in the tooltips for the gcode watch? Or, at least in my opinion, gcode actions shouldn't need a warning even if they're enabled for manual actions, so the gcode should override the warning option.

closed 2021-02-28 by jneilliii