jneilliii/OctoPrint-TPLinkSmartplug #251: [Feature Request]:

DocZagreus, 2021-02-11:
Hi!
Do you think you'll update this plugin in order to work with Tapo P100 smart plugs?

jneilliii, 2021-02-11:
Already answered a couple of times, I'm not personally going to make this plugin work with those devices but someone did mention that they forked it for that purpose, just don't know if it's fully functional or not.

https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/issues/131

https://github.com/Tyfy/OctoPrint-TapoSmartplug

closed 2021-02-11 by jneilliii

jneilliii, 2025-08-17:
Version 2.0.0rc1 is out and is working with my Kasa based HS300, Kasa based HS103 and Tapo based P115. You can change the release channel in OctoPrint's Software Update settings to Release Candidate and update when prompted to get this version. 

Things I've noted so far, the P115 doesn't provide as much energy data as the kasa devices do. Tapo devices will require entering your username and password registered in the Tapo app, Kasa devices don't require those settings to be entered.