jneilliii/OctoPrint-TPLinkSmartplug #254: [ISSUE]: Not powering off TP-Link Plug

oxicottin, 2021-02-21:
**Describe the bug**
I have been using this life saver Plugin for a few months without any trouble. Yesterday I printed and at my prints end my plug never powered down and shut everything off. I thought maybe it was a conflicting plugin so I removed all of them and still same issue and its even an issue if I set my Idle Timeout to one minute and it counts down to shut down and nothing happens at the end.  I even tried a separate TP-link plug and same thing it didn't shut off. So with that I decided to reinstall Octoprint and just the TP-Link Plugin and still same issue exists.

[image]
[image]
[image]

jneilliii, 2021-02-21:
I think this is a duplicate of #250. If you disable the gcode trigger options the idle timer will work as it's supposed to. You don't need both enabled in your use case anyway.

oxicottin, 2021-02-21:
@jneilliii the images are what worked for me for at least 4 months and several prints. I tried your suggestion with the settings below and still after 1 minute it wont shut down plug. I have the correct plug IP so that isn't it and like I mentioned I tried a second  TP plug and still same issue. Thanks,

[image]
[image]

jneilliii, 2021-02-21:
Could you enable debug logging and restart octoprint and run through another round of testing and then share your octoprint.log and plugin_tplinksmartplug_debug.log files here by dragging them into a comment? That will help me diagnose a little further.

Jack77777777, 2021-02-21:
That's really starge. For me it is working when gcode trigger is disabled. Do you have an cloud accountvat kasa or guest and have you changed anything in kasa lately?

oxicottin, 2021-02-21:
@jneilliii I also wanted to say the lightning bolt does nothing, It doesn't turn off or disconnect nor does it turn on. It just spins for a sec and stops, no color change for indication.
[octoprint.log](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/files/6018442/octoprint.log)
[plugin_tplinksmartplug_debug.log](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/files/6018443/plugin_tplinksmartplug_debug.log)

jneilliii, 2021-02-21:
Ok, that is strange. The plugin sees the plug as on the network, but the firmware is not responding to it when it sends commands. This is very similar to what was happening in #218. Like @Jack77777777 mentioned have you recently connected them to the cloud or run firmware updates on these plugs that may not be blocking local access?

oxicottin, 2021-02-21:
No I havent messed with updating the firmware of the plug or anything with it.

jneilliii, 2021-02-21:
I'm stumped since as you said it worked before and all of a sudden stopped working and doesn't work on another plug as well. The only thing I can think of is that the plug just isn't responding or doesn't have access to send the command to the plug. The HS105 is a plug that I've never tested personally but have received reports that it works (including from yourself). I have personally tested HS100, HS107, HS110, and HS300. Have you tried unplugging your HS105 from the wall and back in just to see if it's a random communication issue? For me, I have all mine that I can configured for local only mode in the kasa app.

oxicottin, 2021-02-21:
How can you configure it local only? I poked around in kasa app and dont see that. I just ordered a HS100 to see if that works.

jneilliii, 2021-02-21:
https://www.tp-link.com/us/support/faq/2707/

Jack77777777, 2021-02-22:
Removed Plugs from Kasa app, deinstalled plugin with deletion of data. Re-added devices with guest account to Kasa app.
Added devices to plugin and configured. Same behaviour as before. As soon as gcode trigger is enabled idle power off not working.
Tested with 0.9.26 and 1.0.0rc2. When i login to Octoprint the idle timer starts and printer powers off. It seems that the plugin needs a API request with my configuration. At least when gcode trigger is enabled for one device.

[config.yaml.txt](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/files/6022545/config.yaml.txt)
[octoprint.log](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/files/6022553/octoprint.log)
[plugin_tplinksmartplug_debug.log](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/files/6022554/plugin_tplinksmartplug_debug.log)

Jack77777777, 2021-02-22:
@jneilliii 
One more test: Removed @TPLINKOFF after print and it works. Light goes on with print start TPLINKON and off when idle like the printer. For me this is something i can live with at the moment.
[image]

Attaching more logs for analyzing:
[plugin_tplinksmartplug_debug.log](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/files/6022784/plugin_tplinksmartplug_debug.log)
[octoprint.log](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/files/6022785/octoprint.log)
[config.yaml.txt](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/files/6022813/config.yaml.txt)

jneilliii, 2021-02-23:
Yep, I figured it out @Jack77777777 thanks for all the details. Now I just have to figure out how to adjust the code for it.

jneilliii, 2021-02-23:
Just released 1.0.0rc3 version that I think will resolve this issue.

abrash9, 2021-03-13:
Sorry to keep this thread open..
I have a DLINK router and using the TPLINK HS103
my IP address is set to reserve the Plugs IP address.
i notice other peoples IP addresses are shorter. is this why mine doesn't work? 
IP ADDRESS: 192.168.100.194

jneilliii, 2021-03-13:
No, the ip address shouldn't matter, as long as it is the correct IP address. Your issue sounds like a communication related problem, but I must request you to open a separate issue, after enabling debug logging in the plugin's settings and then restart OctoPrint. In your issue please make sure to include the file plugin_tplinksmartplug_debug.log from OctoPrint's logging section.

abrash9, 2021-03-15:
I have got it working now. I had to uninstall the plugs and then restart my router, install the plugs, set them to Reserve IP Address, then restart my router again. then install the plug in, do the settings then restart octoprint.

PR #264 (2021-04-14): 1.0.0
* clear _autostart_file on print done or canceled events to prevent from auto starting a file on next connect.
* fix missing poll_status in init
* resolve idle timeout issues related to plugs powering off with gcode commands, resolves #254
* change gcode trigger logic for off commands to account for printer buffer, resolves #258
* change index values for strip devices to match labels and the Kasa app, resolves #243
* add sending gcode commands before off and after on, resolves #160, resolves #170
* turn on only if we are closed or errored state with upload event, thanks to @ayufan
* adde [... cut]
closed 2021-04-06 by github-actions[bot]