jneilliii/OctoPrint-TPLinkSmartplug #255: [BUG]: Run system Command not working

ingo916, 2021-02-22:
**Describe the bug**
Added   @WS_LIGHTSON to Run System Command After On  and @WS_LIGHTSOFF to Run System Command Before Off when off on idle is triggered

**To Reproduce**
see screen shot

**Expected behavior**
When the plug is scheduled to turn on or off i want the commands @WS_LIGHTSON and @WS_LIGHTSOFF to run. I can run the command from the terminal tab in OctoPrint and the led lights will turn on or off.

**Screenshots**
[image]

**Desktop (please complete the following information):**
    OctoPrint 1.5.3
    Python 3.7.3
    OctoPi 0.18.0
    Plugin Version - 0.9.26

**Additional context**
I am using WS281x LED Status (0.6.0) plugin  which reacts to @ Commands, i want to send the following commands @WS_LIGHTSON and @WS_LIGHTSOFF to either turn on or off the led status light when the TPLinkSmartplug plugin detects the printer as idle and turns off the power to the selected plug.

jneilliii, 2021-02-22:
Those aren't system commands, those are action commands which are sent via gcode. System commands are like `sudo service shutdown` or `/path/to/scripts/my_script.sh`. The plugin will not do what you are attempting, but the same commands could be put into the gcode scripts section of OctoPrint for before print starts and after print complete/canceled.

jneilliii, 2021-02-22:
The option to run other gcode commands has been requested though in #160, which would work the way you are attempting, I just haven't implemented that feature yet.

closed 2021-02-22 by jneilliii

ingo916, 2021-02-22:
OK great thanks, I hope you implement the option to run gcode commands.