jneilliii/OctoPrint-TPLinkSmartplug #256: I just got the plug so I am not sure If I am doing sth else wrong.

whooooosh1, 2021-02-24:
I just got the plug so I am not sure If I am doing sth else wrong. 
This shows up in the log:

> [2020-07-16 23:10:32,481] DEBUG: ImmutableMultiDict([('checkStatus', u'192.168.1.102')])
> [2020-07-16 23:10:32,482] DEBUG: Checking status of 192.168.1.102.
> [2020-07-16 23:10:32,482] DEBUG: {'system': {'get_sysinfo': {}}}
> [2020-07-16 23:10:32,483] DEBUG: IP 192.168.1.102 is valid.
> [2020-07-16 23:10:32,483] DEBUG: Sending command {'system': {'get_sysinfo': {}}} to 192.168.1.102
> [2020-07-16 23:10:35,611] DEBUG: Could not connect to 192.168.1.102.
> [2020-07-16 23:10:35,612] DEBUG: 
> [2020-07-16 23:10:35,612] DEBUG: {'emeter': {'err_code': True}, 'system': {'get_sysinfo': {'relay_state': 3}}}

_Originally posted by @PhriZ in https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/issues/180#issuecomment-659703340_

jneilliii, 2021-02-24:
There have been a couple of reports of plugs not talking. There was #163 from the UK devices that blocked access, and there have been reports of plugs that are connected to the cloud not being able to be controlled by the plugin in local only mode. Last but not least, there is also the chance that you're not entering the correct IP of the plug like you think you are. Enable debug logging, restart OctoPrint, try a couple of times to get a status by clicking the button in the navbar and then share both your plugin_tplinksmartplug_debug.log and octoprint.log for further analysis.

jneilliii, 2021-02-24:
Closing since I noticed in the other issue you posted you have an incompatible device.

closed 2021-02-24 by jneilliii