jneilliii/OctoPrint-TPLinkSmartplug #259: Need help getting it to work

abrash9, 2021-03-13:
Sorry to keep this thread open..
I have a DLINK router and using the TPLINK HS103
my IP address is set to reserve the Plugs IP address.
I notice other peoples IP addresses are shorter. is this why mine doesn't work?
My router settings are setup as Dynamic IP (DHCP)
IP ADDRESS: 192.168.100.194

I've tried several times to use the OctoPi Plugin and even uninstalled and re installed a few times hoping that would fix it.
I am able to control it with my google home assistant and by my phone, but can't get the plug in to work.

after seeing your answer:
"No, the ip address shouldn't matter, as long as it is the correct IP address. Your issue sounds like a communication related problem, but I must request you to open a separate issue, after enabling debug logging in the plugin's settings and then restart OctoPrint. In your issue please make sure to include the file plugin_tplinksmartplug_debug.log from OctoPrint's logging section."

I did these steps.. I think and restarted the application and the lightning bolt still just rotates when I click it

jneilliii, 2021-03-13:
Go into OctoPrint's logging section in settings and download the file plugin_tplinksmartplug_debug.log file and drag it here into a comment so I can see what the plugin sees.

abrash9, 2021-03-13:
[plugin_tplinksmartplug_debug (1).log](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/files/6135508/plugin_tplinksmartplug_debug.1.log)

jneilliii, 2021-03-13:
ok your log indicates it was working with 192.168.100.159 and 192.168.100.120, but not with ip 192.168.100.192. You sure that you setup the ip and/or DHCP reservation correctly? From what the log shows the plug is not responding to the commands that are being sent to it. I've only seen this in a couple of other issues. One was attributed to a firmware update that blocked local communication from outside tools (#218). The other was related to the device being connected to the cloud and not running in local only mode. Some newer firmware/devices cause local communication to be blocked if you connect it to the cloud IIRC.

abrash9, 2021-03-13:
Yes sorry, I switched the printer to a different TP plug same model. wasn't sure if that was the issue.
that new plug IP is 192.168.100.159
I'm not sure what the IP 192.168.100.120 is, I don't see that in my connected clients list on my router. 
is it because i have the plug set up into the google home app to control over voice?

I do see in the Firmware version for the plug that it is on version 1.1.3 so this is probably why?

jneilliii, 2021-03-13:
I'm not 100% sure if the google home app integration causes the tplink to go into cloud mode or not, but there is probably a good chance of that. From the information that I know it was only UK models that were effected by that firmware update, and there was a way to get in contact with TPLink and get the update reverted. It's linked in that issue. For the other issue there was a process in which you could reset your tplink to factory defaults and during initial setup bypass the cloud connection stuff, documented [here](https://www.tp-link.com/us/support/faq/2707/).

abrash9, 2021-03-13:
Thanks for your help, once this print finishes I will give this a try and report back. I am in Canada and didn't know if the plug would work or not.

abrash9, 2021-03-18:
I believe I got this working. 
after some messing around with the app and contacting TP LINK. The plug I bought, (TPLINK HS103) on firmware (version 1.1.3) does not have the same issues as the HS100 was having. Rather looking further into my router settings, somehow the smart plug connected to my 5g network and was stuck there. I had to do a factory reset and then connected to my 2.4 network and everything seems to be working fine. Make sure your smart plugs and your Pi are on the same network configuration. I believe this could also make it not work if the Pi was on 5g and plugs were on 2.4. 

this issue is now closed. 

Thanks for the help @jneilliii

closed 2021-03-18 by abrash9

jneilliii, 2021-03-18:
I've actually been having a slightly different but similar issue with my newest pi 4. Sometimes it just drops off the network and is unreachable by devices on the same SSID, but then switching my laptop from 5G to 2.4G I can reach it again even though the Pi is on the 5G network.

abrash9, 2021-03-18:
yes, same. I had it working for 5 prints now it won't find it. and I cant even turn it off or on from the Kasa app. 

My Pi was also on 5g network and couldn't get it to work. I switched it to 2.4 but still having connection issues, but now my End GCode wont turn it off using the
G4 P30000 ; Wait
M81 192.168.100.157 ; Turn off printer plug

it just keeps on on and won't actually send the command, not sure if it's octoprint, the Pi, the plug in, or the damn troublesome TpLink plug.