jneilliii/OctoPrint-TPLinkSmartplug #261: [Feature Request]: Listen to OnClientConnected Event to start when Cura connects

lukaskurz, 2021-03-22:
**Is your feature request related to a problem? Please describe.**
I usually slice my stuff with Cura and then directly upload it to my octoprint from within Cura, using [this Plugin](https://marketplace.ultimaker.com/app/cura/plugins/fieldofview/OctoPrintPlugin)  from their marketplace.
After Cura connects to octoprint, it warns me that no printer is connected (since the printer is powered off) and that i can only upload the .stl, and not start the print. This kinda bums me out, since this was an otherwise streamlined process of slicing,uploading and printing, which would be greatly enhanced if I don't have to go into the octoprint dashboard or kasa to turn on the printer.

**Describe the solution you'd like**
You already have an option of turning on the plug, by listening to UploadEvent.
I took a look at their [docs](https://docs.octoprint.org/en/master/events/index.html#server) and it seems, that they have a "OnClientConnect" type of event. I imagine that this should in theory be the kind of event that gets triggered when i connect with the cura plugin, which could then be used to trigger the plug for the printer. There might be some slightly odd behaviour if octoprint takes too long to connect to the printer, but IMO that is negligible.

**Describe alternatives you've considered**
Maybe there is some way to send a on/off command from within the cura plugin ? dunno, but i guess that's not the right repository to ask about the cura plugin :sweat_smile: 

**Additional context**
BTW. I love you for creating this plugin, I was already thinking of hacking together a little script that engages with the smart plugs, but this is way better than what I had thought of.

Also, if you could tell me wether you think this is doable or not, I'd be willing to write the PR myself, shouldn't be too hard.

jneilliii, 2021-03-22:
That plugin actually has options to configure auto power on from within it on the Cura side. It will detect this plugin installed and allow you to select it from a list where you setup the connection to OctoPrint.

lukaskurz, 2021-03-22:
It was a bit hidden, but you're totally right. This plugin just became even more cool 😎 Probably should've checked beforehand if something like that existed.

Thanks for the help and sorry for the inconvenience.

closed 2021-03-22 by lukaskurz