jneilliii/OctoPrint-TPLinkSmartplug #263: [Feature Request]: Handle OctoPrint Shutdown

jbienz, 2021-04-08:
**Is your feature request related to a problem? Please describe.**

This is a feature request.

**Describe the solution you'd like**

I would like an option to power off outlets when OctoPrint shuts down. This is a similar request to #233, but 233 was denied because the user was using a script to detect GPIO. I am not using a script or GPIO. I want to be able to use this menu item in OctoPrint UI:

[image]

And when I click that item, I would like to automatically shut off the printer and the lights.

It appears there is a way to respond to the OctoPrint shutdown event as documented here:

https://docs.octoprint.org/en/master/plugins/mixins.html?highlight=shutdown#octoprint.plugin.ShutdownPlugin.on_shutdown

**Describe alternatives you've considered**

I have not been able to come up with any alternatives, but I'm open to suggestions. Right now I need to manually click one icon for the printer and another for the lights before issuing the shutdown command.

**Additional context**

None that I can think of. Thank you for considering.

jneilliii, 2021-04-08:
I can definitely add that shutdown event monitor as an option similar to the startup event that was added recently.

jneilliii, 2021-04-08:
Added with the above commit. If you'd like to test and confirm you can change the Release Channel for the plugin in OctoPrint's Software Update settings to Release Candidate and upgrade when prompted.

jbienz, 2021-04-12:
Wow, that was _incredibly_ fast! Thank you @jneilliii. I'm in the middle of a print but I'll update and test as soon as that's done.

jbienz, 2021-04-12:
I had a chance to test this morning and it worked _beautifully_. Stellar work. Also, thanks for supporting the release candidate feature, it made testing incredibly simple. I sent a donation to your PayPal. Thanks for the awesome work! Please feel free to close this one now or when you release, whichever you prefer.

michaelmargolis, 2021-04-13:
Just tested 1.0.0rc8 and the feature to turn off the smartplag at system at shutdown is working as expected.  However the plugin does not automatically turn on the smartplug at startup.
Do I have the setup configured correctly?
[image]

jneilliii, 2021-04-13:
yes, those settings look correct to power on during startup. to clarify what startup is, that is when octoprint first starts on the server side (pi). So if you reboot your pi or restart octoprint is when that should trigger. I don't think the addition of the shutdown event would cause this, but I could have broke something.

michaelmargolis, 2021-04-13:
The test workflow was:
Boot octoprint, (should turn on smarplug but doesn’t)
Turn on smartplug by clicking flash icon (does turn on smartplug)
Shut down  Pi (either from octoprint  or using dtoverlay=gpio-shutdown) – this does turn off smartplug.

My intended use involves having  pushbutton that when connecting GPIO3 to ground will start up the Pi (and hopefully turn on the printer via the smartplug) . The same pushbutton will shut down the pi and the printer.  The pi boot/shutdown works well, as does the smartplug shutdown.  Hopefully I can get the startup working as well.

jneilliii, 2021-04-13:
I see you still had debug logging enabled. Can you please share your plugin_tplinksmartplug_debug.log file?

michaelmargolis, 2021-04-13:
here is the latter part of the log, let me know if you need more 

[2021-04-13 20:34:39,074] DEBUG: abortTimeout: 30
[2021-04-13 20:34:39,084] DEBUG: powerOffWhenIdle: False
[2021-04-13 20:34:39,086] DEBUG: idleTimeout: 30
[2021-04-13 20:34:39,086] DEBUG: idleIgnoreCommands: M105
[2021-04-13 20:34:39,087] DEBUG: idleTimeoutWaitTemp: 50
[2021-04-13 20:34:39,088] DEBUG: powering on due to startup.
[2021-04-13 20:34:39,088] DEBUG: powering on 192.168.1.31 due to startup.
[2021-04-13 20:34:39,089] DEBUG: Turning on 192.168.1.31.
[2021-04-13 20:34:39,089] DEBUG: {'autoConnect': True, 'gcodeRunCmdOn': '', 'gcodeCmdOn': False, 'ip': '192.168.1.31', 'btnColor': '#808080', 'autoDisconnectDelay': 0, 'sysCmdOff': False, 'gcodeEnabled': False, 'gcodeOnDelay': 0, 'event_on_shutdown': True, 'emeter': {'get_realtime': {}}, 'countdownOnDelay': 1, 'label': 'Ender5', 'warnPrinting': False, 'thermal_runaway': False, 'gcodeCmdOff': False, 'gcodeOffDelay': 0, 'sysCmdOnDelay': 0, 'event_on_error': False, 'countdownOffDelay': 1, 'sysRunCmdOff': '', 'sysRunCmdOn': '', 'autoDisconnect': True, 'automaticShutdownEnabled': False, 'event_on_upload': False, 'useCountdownRules': False, 'event_on_startup': True, 'icon': 'icon-bolt', 'displayWarning': True, 'autoConnectDelay': 10, 'event_on_disconnect': False, 'sysCmdOn': False, 'gcodeRunCmdOff': '', 'currentState': 'off', 'sysCmdOffDelay': 0}
[2021-04-13 20:34:39,090] DEBUG: IP 192.168.1.31 is valid.
[2021-04-13 20:34:39,090] DEBUG: Sending command {'system': {'set_relay_state': {'state': 1}}} to 192.168.1.31
[2021-04-13 20:34:39,091] DEBUG: Could not connect to 192.168.1.31.
[2021-04-13 20:34:39,092] DEBUG: None
[2021-04-13 20:34:52,600] DEBUG: ImmutableMultiDict([('checkStatus', u'192.168.1.31')])
[2021-04-13 20:34:52,601] DEBUG: Checking status of 192.168.1.31.
[2021-04-13 20:34:52,608] DEBUG: {'system': {'get_sysinfo': {}}}
[2021-04-13 20:34:52,608] DEBUG: IP 192.168.1.31 is valid.
[2021-04-13 20:34:52,609] DEBUG: Sending command {'system': {'get_sysinfo': {}}} to 192.168.1.31
[2021-04-13 20:34:52,683] DEBUG: « ?í"system":{"get_sysinfo":{"sw_ver":"1.0.3 Build 191111 Rel.144915","hw_ver":"4.1","model":"HS100(UK)","deviceId":"800680793892C6E47BE40118213A40421C79F9DA","oemId":"BDE8C5BFBD2235EFFDD4DC1257333B68","hwId":"00105B47562710D8AA66EF95D3FDCE49","rssi":-45,"longitude_i":-1468,"latitude_i":515960,"alias":"Ender5","status":"new","mic_type":"IOT.SMARTPLUGSWITCH","feature":"TIM","mac":"CC:32:E5:D7:7B:75","updating":0,"led_off":0,"relay_state":0,"on_time":0,"active_mode":"none","icon_hash":"","dev_name":"Smart Wi-Fi Plug","next_action":{"type":-1},"ntc_state":0,"err_code":0}}}
[2021-04-13 20:34:52,734] DEBUG: TIM
[2021-04-13 20:34:57,698] DEBUG: Turning on 192.168.1.31.
[2021-04-13 20:34:57,700] DEBUG: {'autoConnect': True, 'gcodeRunCmdOn': '', 'gcodeCmdOn': False, 'ip': '192.168.1.31', 'btnColor': '#808080', 'autoDisconnectDelay': 0, 'sysCmdOff': False, 'gcodeEnabled': False, 'gcodeOnDelay': 0, 'event_on_shutdown': True, 'emeter': {'get_realtime': {}}, 'countdownOnDelay': 1, 'label': 'Ender5', 'warnPrinting': False, 'thermal_runaway': False, 'gcodeCmdOff': False, 'gcodeOffDelay': 0, 'sysCmdOnDelay': 0, 'event_on_error': False, 'countdownOffDelay': 1, 'sysRunCmdOff': '', 'sysRunCmdOn': '', 'autoDisconnect': True, 'automaticShutdownEnabled': False, 'event_on_upload': False, 'useCountdownRules': False, 'event_on_startup': True, 'icon': 'icon-bolt', 'displayWarning': True, 'autoConnectDelay': 10, 'event_on_disconnect': False, 'sysCmdOn': False, 'gcodeRunCmdOff': '', 'currentState': 'off', 'sysCmdOffDelay': 0}
[2021-04-13 20:34:57,704] DEBUG: IP 192.168.1.31 is valid.
[2021-04-13 20:34:57,704] DEBUG: Sending command {'system': {'set_relay_state': {'state': 1}}} to 192.168.1.31
[2021-04-13 20:34:57,728] DEBUG: «  -ý"system":{"set_relay_state":{"err_code":0}}}
[2021-04-13 20:34:57,735] DEBUG: 0
[2021-04-13 20:34:57,736] DEBUG: Checking status of 192.168.1.31.
[2021-04-13 20:34:57,737] DEBUG: {'system': {'get_sysinfo': {}}}
[2021-04-13 20:34:57,737] DEBUG: IP 192.168.1.31 is valid.
[2021-04-13 20:34:57,738] DEBUG: Sending command {'system': {'get_sysinfo': {}}} to 192.168.1.31
[2021-04-13 20:34:57,801] DEBUG: « ?í"system":{"get_sysinfo":{"sw_ver":"1.0.3 Build 191111 Rel.144915","hw_ver":"4.1","model":"HS100(UK)","deviceId":"800680793892C6E47BE40118213A40421C79F9DA","oemId":"BDE8C5BFBD2235EFFDD4DC1257333B68","hwId":"00105B47562710D8AA66EF95D3FDCE49","rssi":-48,"longitude_i":-1468,"latitude_i":515960,"alias":"Ender5","status":"new","mic_type":"IOT.SMARTPLUGSWITCH","feature":"TIM","mac":"CC:32:E5:D7:7B:75","updating":0,"led_off":0,"relay_state":1,"on_time":0,"active_mode":"none","icon_hash":"","dev_name":"Smart Wi-Fi Plug","next_action":{"type":-1},"ntc_state":0,"err_code":0}}}
[2021-04-13 20:34:57,832] DEBUG: TIM
[2021-04-13 20:36:13,975] DEBUG: powering off 192.168.1.31 due to shutdown event.
[2021-04-13 20:36:13,976] DEBUG: Turning off 192.168.1.31.
[2021-04-13 20:36:13,976] INFO: Turning off 192.168.1.31 at 2021-04-13 20:36:13.975932
[2021-04-13 20:36:13,977] DEBUG: {'autoConnect': True, 'gcodeRunCmdOn': '', 'gcodeCmdOn': False, 'ip': '192.168.1.31', 'btnColor': '#808080', 'autoDisconnectDelay': 0, 'sysCmdOff': False, 'gcodeEnabled': False, 'gcodeOnDelay': 0, 'event_on_shutdown': True, 'emeter': {'get_realtime': {}}, 'countdownOnDelay': 1, 'label': 'Ender5', 'warnPrinting': False, 'thermal_runaway': False, 'gcodeCmdOff': False, 'gcodeOffDelay': 0, 'sysCmdOnDelay': 0, 'event_on_err [... cut ...] Rules': False, 'event_on_startup': True, 'icon': 'icon-bolt', 'displayWarning': True, 'autoConnectDelay': 10, 'event_on_disconnect': False, 'sysCmdOn': False, 'gcodeRunCmdOff': '', 'currentState': 'off', 'sysCmdOffDelay': 0}
[2021-04-13 20:36:28,167] DEBUG: IP 192.168.1.31 is valid.
[2021-04-13 20:36:28,167] DEBUG: Sending command {'system': {'set_relay_state': {'state': 1}}} to 192.168.1.31
[2021-04-13 20:36:28,167] DEBUG: Could not connect to 192.168.1.31.
[2021-04-13 20:36:28,167] DEBUG: None

jneilliii, 2021-04-13:
```
[2021-04-13 20:36:28,164] DEBUG: powering on due to startup.
[2021-04-13 20:36:28,165] DEBUG: powering on 192.168.1.31 due to startup.
[2021-04-13 20:36:28,165] DEBUG: Turning on 192.168.1.31.
[2021-04-13 20:36:28,166] DEBUG: {'autoConnect': True, 'gcodeRunCmdOn': '', 'gcodeCmdOn': False, 'ip': '192.168.1.31', 'btnColor': '#808080', 'autoDisconnectDelay': 0, 'sysCmdOff': False, 'gcodeEnabled': False, 'gcodeOnDelay': 0, 'event_on_shutdown': True, 'emeter': {'get_realtime': {}}, 'countdownOnDelay': 1, 'label': 'Ender5', 'warnPrinting': False, 'thermal_runaway': False, 'gcodeCmdOff': False, 'gcodeOffDelay': 0, 'sysCmdOnDelay': 0, 'event_on_error': False, 'countdownOffDelay': 1, 'sysRunCmdOff': '', 'sysRunCmdOn': '', 'autoDisconnect': True, 'automaticShutdownEnabled': False, 'event_on_upload': False, 'useCountdownRules': False, 'event_on_startup': True, 'icon': 'icon-bolt', 'displayWarning': True, 'autoConnectDelay': 10, 'event_on_disconnect': False, 'sysCmdOn': False, 'gcodeRunCmdOff': '', 'currentState': 'off', 'sysCmdOffDelay': 0}
[2021-04-13 20:36:28,167] DEBUG: IP 192.168.1.31 is valid.
[2021-04-13 20:36:28,167] DEBUG: Sending command {'system': {'set_relay_state': {'state': 1}}} to 192.168.1.31
[2021-04-13 20:36:28,167] DEBUG: Could not connect to 192.168.1.31.
[2021-04-13 20:36:28,167] DEBUG: None
```

Interesting, it seems to trigger the on command but then doesn't get the expected response back from the plug, if at all. I wonder if this is a timing thing where it's not completely on the network at initial octoprint startup possibly.

michaelmargolis, 2021-04-14:
I tested with a fresh install on a spare Raspberry pi (also a Pi 4) and the plugin works as expected.
I will do some more testing to see why my main pi has the problem.

michaelmargolis, 2021-04-14:
I updated the Pi OS (apt update and full-upgrade) and now have the plugin functioning as expected.  I had thought that an update would not be needed following an install of the latest version of octopi.  
I had installed the OS and octoprint manually on the spare Pi so perhaps that is whey the problem didn't exist.

jneilliii, 2021-04-14:
Interesting, and good to know. I'll keep that in mind if I see anyone else reporting issues that the plug isn't responding.

PR #264 (2021-04-14): 1.0.0
* clear _autostart_file on print done or canceled events to prevent from auto starting a file on next connect.
* fix missing poll_status in init
* resolve idle timeout issues related to plugs powering off with gcode commands, resolves #254
* change gcode trigger logic for off commands to account for printer buffer, resolves #258
* change index values for strip devices to match labels and the Kasa app, resolves #243
* add sending gcode commands before off and after on, resolves #160, resolves #170
* turn on only if we are closed or errored state with upload event, thanks to @ayufan
* adde [... cut]
closed 2021-04-14 by jneilliii