jneilliii/OctoPrint-TPLinkSmartplug #265: Can't connect back after auto shutdown

guigro, 2021-04-20:
Hello !
I'm not sure it's a plugin issue but it happens since I use the plugin.

I use a Raspberry Pi 4 on a Prusa MK3s.

After a print and cooling down, I ask the printer to be powered off. This works.
The octoprint interface switch works too (everything works !! yeeha)

But, when the printer was working, then went off after cool down, if I turn it back on (manually or on upload), then Octoprint won't be able to connect to it.

In the terminal I have this error : 
`Changing monitoring state from "Offline" to "Opening serial connection"
Connecting to port /dev/ttyACM0, baudrate 250000
Changing monitoring state from "Opening serial connection" to "Error: Connection error, see Terminal tab"
Changing monitoring state from "Error: Connection error, see Terminal tab" to "Offline (Error: Connection error, see Terminal tab)"
Unexpected error while connecting to serial port /dev/ttyACM0, baudrate 250000 from hook default: SerialException: '[Errno 11] Could not exclusively lock port /dev/ttyACM0: [Errno 11] Resource temporarily unavailable' @ comm.py:_open_serial:3670`

Maybe it's a process that is still trying to connect to the printer and then it's busy ?

To solve it I have to manually restart Octoprint each time.

Let me know if there is more I can test !
Guigro

jneilliii, 2021-04-20:
This sounds like maybe a timing issue. Have you tried increasing your auto connect time to something larger than the default 10 seconds? There also is a chance that there is an issue with backpowering, does your printer's LCD stay powered on when the printer is off but the Pi is connected via USB?

guigro, 2021-04-20:
I tried with a longer auto connect, only change is that the error message comes later.

The LCD is off when the power of the printer is off. While the Octopi is still on of course and connected through USB.
But as the power of the printer is off, the LCD is off too.

I have an additionnal error message to provide, coming from the Dashboard plugin (that gives me another message than the log : 
`Offline (Error: No more candidates to test, and no working port/baudrate combination detected.)`

This is why it looks like there is a process using the port/baudrate combination to try to stay connected or something like that.

Again, let me know if I can provide more info or logs or make any test ! Thanks :)

jneilliii, 2021-04-20:
Do you have any other apps or plugins that may be trying to auto connect, like octodash or octoeverywhere? Please try enabling debug logging in the plugin's settings and restart OctoPrint and try the process again. If it's still giving you the same issue grab the plugin_tplinksmartplug_debug.log and octoprint.log files and share them here by dragging them into a comment to attach.

guigro, 2021-04-20:
OK well I tried to disable plugins one buy one to find if one was faulty, and I found which one !
It's "WS281x LED Status (0.7.2)" that controls my led strip and change colors regarding statuses.
I guess it's keep checking if the printer is online and then blocks the port and baudrate.

I'll post this issue also in the plugin's Github : https://github.com/cp2004/OctoPrint-WS281x_LED_Status/issues

Maybe there's a workaround :)

jneilliii, 2021-04-20:
Interesting bug @cp2004 seems to be already investigating based on his response to your issue. Thanks for reporting back.

closed 2021-04-20 by jneilliii

guigro, 2021-04-20:
Yes he answered very fast and accurately !
Thanks for answering fast too, as I answered him the second tip made it work, so it has something to do with his plugin and power management.

thanks for your amazing plugin !