jneilliii/OctoPrint-TPLinkSmartplug #266: [BUG]: Plug turns of mid-print (on rare occasions)

nickbuhling, 2021-04-23:
Dear developer,
I found that on some rare occasions, say 1 in 20 prints or so, I find the printer shut off during the print. This is of course very unfortunate. I was wondering whether this is a bug, whether there is a way to improve reliability or whether you think this has to do with the TP link internal software or something instead of this plugin?

Would love to know.

Thanks!
Kind regards
Nick

jneilliii, 2021-04-23:
Hard to say without any information in regard to your settings. Depending on your settings, could be something as simple as disconnect event monitoring. If you want to help solve the problem, which I haven't had since plugin inception, enable debug logging in the plugin's settings, restart octoprint, and operate like normal. When it does something unexpected grab the plugin_tplinksmartplug_debug.log and octoprint.log and share in a reply here.

nickbuhling, 2021-04-23:
Thank you for your fast reply! Good to know there is no known issue. I will enable debug logging

closed 2021-05-15 by github-actions[bot]